coolsnowwolf/lede #9881: mt7621 /proc/net/dev 统计异常

jshensh, 2022-08-01:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：/proc/net/dev 中数据统计异常，编译时默认使用 MediaTek HWNAT

```
root@OpenWrt:~# cat /proc/net/dev
Inter-|   Receive                                                |  Transmit
 face |bytes    packets errs drop fifo frame compressed multicast|bytes    packets errs drop fifo colls carrier compressed
    lo: 53444959  301861    0    0    0     0          0         0 53444959  301861    0    0    0     0       0          0
  lan2: 1794093386 6992723    0 1045565    0     0          0         0 20392244561 21426245    0    0    0     0       0          0
pppoe-telecom: 1697729580 5943153    0    1    0     0          0         0 19920833891 21422229    0    0    0     0       0          0
 wlan0: 18791011  206231    0    0    0     0          0         0 54345630  556948    0    0    0     0       0          0
   wan: 65646527334 62178933    0 5222    0     0          0         0 3245854992 11139907    0    0    0     0       0          0
 wlan1: 1747152275 2951020    0 4053    0     0          0         0 4679167960 3620996    0    0    0     0       0          0
pppoe-wan: 1708229379 4862476    0   17    0     0          0         0 43327659809 40302946    0    0    0     0       0          0
  lan1: 1751257559 4872946    0    0    0     0          0         0 44214639932 40313416    0    0    0     0       0          0
br-lan: 62732088149 52868355    0    0    0     0          0     63128 3418202692 10260736    0    0    0     0       0          0
  eth0: 98549493469 103590214 1176 2432    0     0          0         0 96351092081 102453211    0    0    0     0       0          0
ztrfyghjp3:  291003     400    0    0    0     0          0         0    39122     401    0    0    0     0       0          0
```

### (2) 路由器型号和固件版本

A：
型号：JDCloud RE-SP-01B
源码版本：https://github.com/coolsnowwolf/lede/commit/005c491b27a63e673e158834ed48462b7af304bc
Packages 版本：https://github.com/coolsnowwolf/packages/commit/5d695554f1ba2f3272891ff4a87d3bcb2e086498

编译时使用的 DTS 等配置文件：

<details><summary>target/linux/ramips/dts/mt7621_jdcloud_re-sp-01b.dts</summary>

```
// SPDX-License-Identifier: GPL-2.0-or-later OR MIT
/dts-v1/;

#include "mt7621.dtsi"

#include <dt-bindings/gpio/gpio.h>
#include <dt-bindings/input/input.h>

/ {
        compatible = "jdcloud,re-sp-01b", "mediatek,mt7621-soc";
        model = "JDCloud RE-SP-01B";

        aliases {
                led-boot = &led_red;
                led-failsafe = &led_red;
                led-running = &led_green;
                led-upgrade = &led_blue;
                label-mac-device = &gmac0;
        };

        chosen {
                bootargs = "console=ttyS0,115200";
        };

        keys {
                compatible = "gpio-keys";

                reset {
[... 549 lines cut ...]
        option proto 'pppoe'
        option keepalive '0'
        option username '021********'
        option password '******'
        option metric '41'
        option ipv6 'auto'

config interface 'rndis'
        option proto 'dhcp'
        option macaddr '2C:B2:1A:5D:8A:00'
        option _orig_ifname 'usb0'
        option _orig_bridge 'false'
        option metric '40'
        option ifname 'eth1'

config interface 'telecom'
        option proto 'pppoe'
        option ifname 'lan2'
        option username 'ad********'
        option password '********'
        option keepalive '0'
        option ipv6 'auto'
        option metric '42'
```
</details>

Cathgao, 2022-08-24:
硬件加速的流量无法统计，二选一

coolsnowwolf, 2026-09-05:
报告来自 2022 年 mt7621 私有 HWNAT 配置，/proc/net/dev 在硬件卸载时不必包含所有被硬件转发的字节，且没有关闭 HWNAT 后的对照数据。没有可复现的现行驱动缺陷，关闭为“不计划处理”。

closed 2026-09-05 by coolsnowwolf