coolsnowwolf/lede #9882: openwrt 5.19内核编译不成功

Eay5, 2022-08-02:
[image]

mitsukileung, 2022-08-02:
我也是一样的错误

WYC-2020, 2022-08-02:
我去都5.19了 。这个补丁需要更新

hezhiming1995, 2022-08-02:
linux 5.19内核据说性能提升非常明显。可以查看IT之家关于Linux发布5.19内核的资讯。
https://www.ithome.com/0/631/204.htm（在WebXPRT的测试下，带来了50%性能提升，相较于5.15）
期待L大可以加固5.19的内核！

xiangfeidexiaohuo, 2022-08-02:
952  953 补丁都要修，还有drm相关、cifs、ntfs3等一些地方也要修，等待吧。

WYC-2020, 2022-08-02:
提升要看是哪方面的提示，反正也就那样

GreatMichaelLee, 2022-08-02:
> 提升要看是哪方面的提升，反正也就那样

我不要提升，我只要稳定，目前看来还是5.10最稳

hcym, 2022-08-02:
默认够用，不折腾

ricksuzade-maker, 2022-08-02:
> > 提升要看是哪方面的提升，反正也就那样
> 
> 我不要提升，我只要稳定，目前看来还是5.10最稳

目前lean最稳定的是5.15内核，远比5.10更好用

firkerword, 2022-08-02:
> 

我不信

firkerword, 2022-08-02:
昨天就遇到这个问题了看了下没人反馈还以为是自己的问题😂

GreatMichaelLee, 2022-08-02:
Applying /home/builder/lede_x86/target/linux/generic/hack-5.19/952-add-net-conntrack-events-support-multiple-registrant.patch using plaintext:
patching file include/net/netfilter/nf_conntrack_ecache.h
Hunk #1 succeeded at 65 (offset -16 lines).
Hunk #2 succeeded at 103 (offset -11 lines).
Hunk #3 FAILED at 135.
Hunk #4 FAILED at 150.
2 out of 4 hunks FAILED -- saving rejects to file include/net/netfilter/nf_conntrack_ecache.h.rej
patching file include/net/netns/conntrack.h
Hunk #1 succeeded at 107 with fuzz 2 (offset -5 lines).
patching file net/netfilter/Kconfig
patching file net/netfilter/nf_conntrack_core.c
Hunk #1 succeeded at 2802 (offset -35 lines).
patching file net/netfilter/nf_conntrack_ecache.c
patching file net/netfilter/nf_conntrack_netlink.c
Hunk #2 succeeded at 3798 (offset -34 lines).
Hunk #3 succeeded at 3907 (offset -34 lines).
Patch failed!  Please fix /home/builder/lede_x86/target/linux/generic/hack-5.19/952-add-net-conntrack-events-support-multiple-registrant.patch!
make[4]: *** [Makefile:36: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/linux-5.19/.prepared_e6e5cbd56e2dcc6c62440cd61f644fb7] Error 1
make[4]: Leaving directory '/home/builder/lede_x86/target/linux/x86'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/home/builder/lede_x86/target/linux'
time: target/linux/compile#5.11#23.46#56.99
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:25: target/linux/compile] Error 1
make[2]: Leaving directory '/home/builder/lede_x86'
make[1]: *** [target/Makefile:18: /home/builder/lede_x86/staging_dir/target-x86_64_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/builder/lede_x86'
make: *** [/home/builder/lede_x86/include/toplevel.mk:230：world] 错误 2
builder@Build-Server:~$

WYC-2020, 2022-08-02:
[image]
已上车 也就那样

Zaphkito, 2022-08-02:
没必要那么着急赶新
有一说一，OpenWrt你是拿来当路由器系统用的，你在乎那性能提升干什么
5w的分和9w的分在日常上网有什么区别吗，它主要是当路由器系统用的，没那性能提升拿来干网络重活也足够了（比如跑满带宽的科学上网）
一天天别那么多伪需求

unknown10777, 2022-08-02:
请问楼主的openwrt是路由器呢还是nas呢？

ricksuzade-maker, 2022-08-02:
> 

测试内核，愿意更新就更新，没强迫谁，是吧？有的人就喜欢追新，因为他有新硬件，比如3588，比如qat加速卡。。。

GreatMichaelLee, 2022-08-02:
> 没必要那么着急赶新 有一说一，OpenWrt你是拿来当路由器系统用的，你在乎那性能提升干什么 5w的分和9w的分在日常上网有什么区别吗，它主要是当路由器系统用的，没那性能提升拿来干网络重活也足够了（比如跑满带宽的科学上网） 一天天别那么多伪需求

> 没必要那么着急赶新 有一说一，OpenWrt你是拿来当路由器系统用的，你在乎那性能提升干什么 5w的分和9w的分在日常上网有什么区别吗，它主要是当路由器系统用的，没那性能提升拿来干网络重活也足够了（比如跑满带宽的科学上网） 一天天别那么多伪需求

我不需要，所以大家都不需要…哈哈

FMJYY23, 2022-08-02:
选的测试版但是编译出来的还是5.18

xiangfeidexiaohuo, 2022-08-02:
> 选的测试版但是编译出来的还是5.18

多看看提交，测试改回5.18了，5.19还有些bug，在说不一定每个人都能编译过，要玩，自己改5.19呗

WYC-2020, 2022-08-02:
玩还是可以玩的，不过目前很多东西没适配好，要自己修复一下，如果追求稳定或者作为你家路由器主系统用的话，建议还是用稳定版系统，爱折腾喜欢尝鲜的可以玩下，至于性能没跑分具体不清楚，哈哈

Yukari316, 2022-08-03:
也是一样的报错，在WSL2里也不能编译

WYC-2020, 2022-08-03:
> 也是一样的报错，在WSL2里也不能编译

要自己修才能编译过

TonyTeng22, 2022-08-04:
希望大神能出个简短教程, 至少能编译通过 5.19, 大家玩起来后问题解决也快些

ricksuzade-maker, 2022-08-04:
mkdir -p /home/zhiliang/scripts/lede/staging_dir/target-x86_64_musl/root-x86/stamp

SHELL= flock /home/zhiliang/scripts/lede/tmp/.root-copy.flock -c 'cp -fpR /home/zhiliang/scripts/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/.pkgdir/kernel/. /home/zhiliang/scripts/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/zhiliang/scripts/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.kernel_installed
ERROR: module '/home/zhiliang/scripts/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.19/drivers/dma-buf/dma-shared-buffer.ko' is missing.
make[3]: *** [modules/other.mk:136: /home/zhiliang/scripts/lede/bin/targets/x86/64/packages/kmod-dma-buf_5.19-1_x86_64.ipk] Error 1
make[3]: Leaving directory '/home/zhiliang/scripts/lede/package/kernel/linux'
time: package/kernel/linux/compile#1.88#0.17#2.04
    ERROR: package/kernel/linux failed to build.
make[2]: *** [package/Makefile:116: package/kernel/linux/compile] Error 1
make[2]: Leaving directory '/home/zhiliang/scripts/lede'
make[1]: *** [package/Makefile:110: /home/zhiliang/scripts/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/zhiliang/scripts/lede'
make: *** [/home/zhiliang/scripts/lede/include/toplevel.mk:230: world] Error 2

今天的版本还是有问题

coolsnowwolf, 2022-08-04:
已修复

ricksuzade-maker, 2022-08-04:
SHELL= flock /home/zhiliang/scripts/lede/tmp/.root-copy.flock -c 'cp -fpR /home/zhiliang/scripts/lede/build_dir/target-x86_64_musl/linux-firmware-20220708/.pkgdir/r8169-firmware/. /home/zhiliang/scripts/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/zhiliang/scripts/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.r8169-firmware_installed
echo "r8169-firmware" >> /home/zhiliang/scripts/lede/staging_dir/target-x86_64_musl/pkginfo/linux-firmware.default.install
make[3]: Leaving directory '/home/zhiliang/scripts/lede/package/firmware/linux-firmware'
time: package/firmware/linux-firmware/compile#1.40#0.63#1.78
make[3]: Entering directory '/home/zhiliang/scripts/lede/package/kernel/linux'
mkdir -p /home/zhiliang/scripts/lede/staging_dir/target-x86_64_musl/root-x86/stamp

SHELL= flock /home/zhiliang/scripts/lede/tmp/.root-copy.flock -c 'cp -fpR /home/zhiliang/scripts/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/.pkgdir/kernel/. /home/zhiliang/scripts/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/zhiliang/scripts/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.kernel_installed
ERROR: module '/home/zhiliang/scripts/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.19/drivers/gpu/drm/i915/gvt/kvmgt.ko' is missing.
make[3]: *** [modules/virt.mk:116: /home/zhiliang/scripts/lede/bin/targets/x86/64/packages/kmod-i915-gvt_5.19-1_x86_64.ipk] Error 1
make[3]: Leaving directory '/home/zhiliang/scripts/lede/package/kernel/linux'
time: package/kernel/linux/compile#1.78#0.20#1.96
    ERROR: package/kernel/linux failed to build.
make[2]: *** [package/Makefile:116: package/kernel/linux/compile] Error 1
make[2]: Leaving directory '/home/zhiliang/scripts/lede'
make[1]: *** [package/Makefile:110: /home/zhiliang/scripts/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/zhiliang/scripts/lede'
make: *** [/home/zhiliang/scripts/lede/include/toplevel.mk:230: world] Error 2

只好一关关过了

GreatMichaelLee, 2022-08-04:
NOTICE: module '/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/linux-5.19/drivers/video/fbdev/core/fb.ko' is built-in.
NOTICE: module '/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/linux-5.19/lib/fonts/font.ko' is built-in.
NOTICE: module '/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/linux-5.19/arch/x86/video/fbdev.ko' is built-in.
Packaged contents of /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-fb into /home/builder/lede_x86/bin/targets/x86/64/packages/kmod-fb_5.19-1_x86_64.ipk
NOTICE: module '/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/linux-5.19/drivers/video/fbdev/core/cfbfillrect.ko' is built-in.
Packaged contents of /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-fb-cfb-fillrect into /home/builder/lede_x86/bin/targets/x86/64/packages/kmod-fb-cfb-fillrect_5.19-1_x86_64.ipk
NOTICE: module '/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/linux-5.19/drivers/video/fbdev/core/cfbcopyarea.ko' is built-in.
Packaged contents of /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-fb-cfb-copyarea into /home/builder/lede_x86/bin/targets/x86/64/packages/kmod-fb-cfb-copyarea_5.19-1_x86_64.ipk
NOTICE: module '/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/linux-5.19/drivers/video/fbdev/core/cfbimgblt.ko' is built-in.
Packaged contents of /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-fb-cfb-imgblt into /home/builder/lede_x86/bin/targets/x86/64/packages/kmod-fb-cfb-imgblt_5.19-1_x86_64.ipk
NOTICE: module '/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/linux-5.19/drivers/video/fbdev/core/fb_sys_fops.ko' is built-in.
Packaged contents of /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-fb-sys-fops into /home/builder/lede_x86/bin/targets/x86/64/packages/kmod-fb-sys-fops_5.19-1_x86_64.ipk
NOTICE: module '/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/linux-5.19/drivers/video/fbdev/core/syscopyarea.ko' is built-in.
NOTICE: module '/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/linux-5.19/drivers/video/fbdev/core/sysfillrect.ko' is built-in.
NOTICE: module '/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/linux-5.19/drivers/video/fbdev/core/sysimgblt.ko' is built-in.
Packaged contents of /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-fb-sys-ram into /home/builder/lede_x86/bin/targets/x86/64/packages/kmod-fb-sys-ram_5.19-1_x86_64.ipk
NOTICE: module '/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/linux-5.19/drivers/gpu/drm/drm.ko' is built-in.
NOTICE: module '/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/linux-5.19/drivers/gpu/drm/drm_panel_orientation_quirks.ko' is built-in.
Packaged contents of /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-drm into /home/builder/lede_x86/bin/targets/x86/64/packages/kmod-drm_5.19-1_x86_64.ipk
NOTICE: module '/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/linux-5.19/drivers/gpu/drm/ttm/ttm.ko' is built-in.
Packaged contents of /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-drm-ttm into /home/builder/lede_x86/bin/targets/x86/64/packages/kmod-drm-ttm_5.19-1_x86_64.ipk
NOTICE: module '/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/linux-5.19/drivers/gpu/drm/drm_kms_helper.ko' is built-in.
Packaged contents of /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-drm-kms-helper into /home/builder/lede_x86/bin/targets/x86/64/packages/kmod-drm-kms-helper_5.19-1_x86_64.ipk
NOTICE: module '/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/linux-5.19/drivers/gpu/drm/i915/i915.ko' is built-in.
Packaged contents of /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-drm-i915 into /home/builder/lede_x86/bin/targets/x86/64/packages/kmod-drm-i915_5.19-1_x86_64.ipk
rstrip.sh: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-amazon-ena/lib/modules/5.19/ena.ko: relocatable
Packaged contents of /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-amazon-ena into /home/builder/lede_x86/bin/targets/x86/64/packages/kmod-amazon-ena_5.19-1_x86_64.ipk
Package kmod-amd-xgbe is missing dependencies for the following libraries:
mdio_devres.ko
make[3]: *** [/home/builder/lede_x86/target/linux/x86/modules.mk:35: /home/builder/lede_x86/bin/targets/x86/64/packages/kmod-amd-xgbe_5.19-1_x86_64.ipk] Error 1
make[3]: Leaving directory '/home/builder/lede_x86/package/kernel/linux'
time: package/kernel/linux/compile#25.60#10.08#27.69
    ERROR: package/kernel/linux failed to build.
make[2]: *** [package/Makefile:116: package/kernel/linux/compile] Error 1
make[2]: Leaving directory '/home/builder/lede_x86'
make[1]: *** [package/Makefile:110: /home/builder/lede_x86/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/builder/lede_x86'
make: *** [/home/builder/lede_x86/include/toplevel.mk:230：world] 错误 2
builder@Build-Server:/home/builder/lede_x86$

learnsometech, 2022-08-04:
make[7]: Entering directory '/home/core/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-atm-2.5.2/src/maint'
/bin/bash ../../libtool  --tag=CC   --mode=link x86_64-openwrt-linux-musl-gcc  -I../../src/include -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/core/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-atm-2.5.2=linux-atm-2.5.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -Wall -Wshadow -Wpointer-arith -Wwrite-strings -Wstrict-prototypes  -L/home/core/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/home/core/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib -znow -zrelro  -o atmdiag atmdiag.o ../../src/lib/libatm.la 
OpenWrt-libtool: link: x86_64-openwrt-linux-musl-gcc -I../../src/include -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/core/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-atm-2.5.2=linux-atm-2.5.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z -Wl,now -Wl,-z -Wl,relro -Wall -Wshadow -Wpointer-arith -Wwrite-strings -Wstrict-prototypes -znow -zrelro -o .libs/atmdiag atmdiag.o  -L/home/core/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/home/core/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib ../../src/lib/.libs/libatm.so -lresolv
/bin/bash ../../libtool  --tag=CC   --mode=link x86_64-openwrt-linux-musl-gcc  -I../../src/include -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/core/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-atm-2.5.2=linux-atm-2.5.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -Wall -Wshadow -Wpointer-arith -Wwrite-strings -Wstrict-prototypes  -L/home/core/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/home/core/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib -znow -zrelro  -o atmdump atmdump.o ../../src/lib/libatm.la 
OpenWrt-libtool: link: x86_64-openwrt-linux-musl-gcc -I../../src/include -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/core/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-atm-2.5.2=linux-atm-2.5.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z -Wl,now -Wl,-z -Wl,relro -Wall -Wshadow -Wpointer-arith -Wwrite-strings -Wstrict-prototypes -znow -zrelro -o .libs/atmdump atmdump.o  -L/home/core/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/home/core/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib ../../src/lib/.libs/libatm.so -lresolv
/bin/bash ../../libtool  --tag=CC   --mode=link x86_64-openwrt-linux-musl-gcc  -I../../src/include -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/core/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-atm-2.5.2=linux-atm-2.5.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -Wall -Wshadow -Wpointer-arith -Wwrite-strings -Wstrict-prototypes  -L/home/core/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/home/core/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib -znow -zrelro  -o sonetdiag sonetdiag.o ../../src/lib/libatm.la 
OpenWrt-libtool: link: x86_64-openwrt-linux-musl-gcc -I../../src/include -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/core/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-atm-2.5.2=linux-atm-2.5.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z -Wl,now -Wl,-z -Wl,relro -Wall -Wshadow -Wpointer-arith -Wwrite-strings -Wstrict-prototypes -znow -zrelro -o .libs/sonetdiag sonetdiag.o  -L/home/core/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/home/core/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib ../../src/lib/.libs/libatm.so -lresolv
/bin/bash ../../libtool  --tag=CC   --mode=link x86_64-openwrt-linux-musl-gcc  -I../../src/include -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/core/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-atm-2.5.2=linux-atm-2.5.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -Wall -Wshadow -Wpointer-arith -Wwrite-strings -Wstrict-prototypes  -L/home/core/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/home/core/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib -znow -zrelro  -o saaldump saaldump.o ../../src/q2931/qd.dump.o ../../src/saal/libsaal.a ../../src/lib/libatm.la 
OpenWrt-libtool: link: x86_64-openwrt-linux-musl-gcc -I../../src/include -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/core/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-atm-2.5.2=linux-atm-2.5.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z -Wl,now -Wl,-z -Wl,relro -Wall -Wshadow -Wpointer-arith -Wwrite-strings -Wstrict-prototypes -znow -zrelro -o .libs/saaldump saaldump.o ../../src/q2931/qd.dump.o  -L/home/core/lede/stagin [... cut ...]
make[3]: Leaving directory '/home/core/lede/package/network/utils/linux-atm'
time: package/network/utils/linux-atm/compile#0.85#0.15#0.92
    ERROR: package/network/utils/linux-atm failed to build.
make[2]: *** [package/Makefile:116: package/network/utils/linux-atm/compile] Error 1
make[2]: Leaving directory '/home/core/lede'
make[1]: *** [package/Makefile:110: /home/core/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/core/lede'
make: *** [/home/core/lede/include/toplevel.mk:230：world] 错误 2

xiexiaosheng, 2022-08-05:
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency kmod-fs-ntfs3 for ntfs3-mount
 * pkg_hash_fetch_best_installation_candidate: Packages for ntfs3-mount found, but incompatible with the architectures configured
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for automount:
 *      kmod-fs-ntfs3
 * opkg_install_cmd: Cannot install package automount.
make[2]: *** [package/Makefile:70: package/install] Error 255
make[2]: Leaving directory '/home/xiexiaosheng/openwrt'
make[1]: *** [package/Makefile:111: /home/xiexiaosheng/openwrt/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/xiexiaosheng/openwrt'
make: *** [/home/xiexiaosheng/openwrt/include/toplevel.mk:230: world] Error 2

coolsnowwolf, 2022-08-05:
[image]

coolsnowwolf, 2022-08-05:
git pull 后已经可以正常编译使用 5.19 内核

ricksuzade-maker, 2022-08-05:
mkdir -p /home/zhiliang/scripts/lede/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/zhiliang/scripts/lede/tmp/.root-copy.flock -c 'cp -fpR /home/zhiliang/scripts/lede/build_dir/target-x86_64_musl/linux-firmware-20220708/.pkgdir/r8169-firmware/. /home/zhiliang/scripts/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/zhiliang/scripts/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.r8169-firmware_installed
echo "r8169-firmware" >> /home/zhiliang/scripts/lede/staging_dir/target-x86_64_musl/pkginfo/linux-firmware.default.install
make[3]: Leaving directory '/home/zhiliang/scripts/lede/package/firmware/linux-firmware'
time: package/firmware/linux-firmware/compile#1.39#0.60#1.77
make[3]: Entering directory '/home/zhiliang/scripts/lede/package/kernel/linux'
mkdir -p /home/zhiliang/scripts/lede/staging_dir/target-x86_64_musl/root-x86/stamp

SHELL= flock /home/zhiliang/scripts/lede/tmp/.root-copy.flock -c 'cp -fpR /home/zhiliang/scripts/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/.pkgdir/kernel/. /home/zhiliang/scripts/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/zhiliang/scripts/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.kernel_installed
ERROR: module '/home/zhiliang/scripts/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.19/drivers/gpu/drm/i915/gvt/kvmgt.ko' is missing.
make[3]: *** [modules/virt.mk:116: /home/zhiliang/scripts/lede/bin/targets/x86/64/packages/kmod-i915-gvt_5.19-1_x86_64.ipk] Error 1
make[3]: Leaving directory '/home/zhiliang/scripts/lede/package/kernel/linux'
time: package/kernel/linux/compile#1.84#0.13#1.96
    ERROR: package/kernel/linux failed to build.
make[2]: *** [package/Makefile:116: package/kernel/linux/compile] Error 1
make[2]: Leaving directory '/home/zhiliang/scripts/lede'
make[1]: *** [package/Makefile:110: /home/zhiliang/scripts/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/zhiliang/scripts/lede'
make: *** [/home/zhiliang/scripts/lede/include/toplevel.mk:230: world] Error 2

xiangfeidexiaohuo, 2022-08-06:
[image]
没体验出，提升多少。

xiangfeidexiaohuo, 2022-08-06:
> 

kmod-i915-gvt设置为# CONFIG_PACKAGE_kmod-i915-gvt is not set

915-gvt这个是还有bug

xiangfeidexiaohuo, 2022-08-06:
> mkdir -p /home/zhiliang/scripts/lede/staging_dir/target-x86_64_musl/root-x86/stamp SHELL= flock /home/zhiliang/scripts/lede/tmp/.root-copy.flock -c 'cp -fpR [...]

kmod-i915-gvt设置为# CONFIG_PACKAGE_kmod-i915-gvt is not set

915-gvt这个是还有bug

WYC-2020, 2022-08-06:
> [image] 没体验出，提升多少。

我跑过分 基本没差距 和5.4

Zaphkito, 2022-08-06:
> > [image] 没体验出，提升多少。
> 
> 我跑过分 基本没差距 和5.4

那个所谓的50%性能提升就是测试的什么浏览器测试和jpeg渲染什么的，openwrt作为路由器系统来说真的没必要

xiangfeidexiaohuo, 2022-08-06:
> > [image] 没体验出，提升多少。
> 
> 我跑过分 基本没差距 和5.4

跑分确实没差，这是的。就是内核版本高，看着舒服，哈哈。

xiangfeidexiaohuo, 2022-08-06:
> > > [image] 没体验出，提升多少。
> > 
> > 
> > 我跑过分 基本没差距 和5.4
> 
> 那个所谓的50%性能提升就是测试的什么浏览器测试和jpeg渲染什么的，openwrt作为路由器系统来说真的没必要

纯粹就是内核版本看着高，舒服而已。其他没啥。

eightsheep67, 2022-08-10:
所以5.18就要这么死了？命太短

closed 2022-09-14 by aiamadeus