coolsnowwolf/lede #9884: 系统日志乱码是怎么回事？

keeno1982, 2022-08-02:
主机是R4S 4G，拉的最新的代码编译
固件运行一段时间后，就会开始刷出大量乱码日志，咋回事呢？

以下是部分日志内容：
Tue Aug  2 19:16:07 2022 kern.info kernel: [64322.863995]  ?w??_?+0xe4/0x198
Tue Aug  2 19:16:07 2022 kern.info kernel: [64322.864282]  ?ߢu?+0x1dc/0x4a8
Tue Aug  2 19:16:07 2022 kern.info kernel: [64322.864567]  ߢu?+0x50/0xc8
Tue Aug  2 19:16:07 2022 kern.info kernel: [64322.864831]  wse?d?n_`??wp?h+0x384/0x568
Tue Aug  2 19:16:07 2022 kern.info kernel: [64322.865194]  ?n_`e+0x34/0x50
Tue Aug  2 19:16:07 2022 kern.info kernel: [64322.865467]  Ԯm+0x2c/0x108
Tue Aug  2 19:16:07 2022 kern.info kernel: [64322.865741]  ??t+0x60/0x128
Tue Aug  2 19:16:07 2022 kern.info kernel: [64322.866008]  o_??+0x248/0x8a8
Tue Aug  2 19:16:07 2022 kern.info kernel: [64322.866286]  ?k?tݫ??+0x70/0xf0
Tue Aug  2 19:16:07 2022 kern.info kernel: [64322.866578]  ie+0x200/0x230
Tue Aug  2 19:16:07 2022 kern.info kernel: [64322.866839]  i?k?n??a?t+0x380/0x390
Tue Aug  2 19:16:07 2022 kern.info kernel: [64322.867159]  ?o_k?n??a?t+0xf4/0x178
Tue Aug  2 19:16:07 2022 kern.info kernel: [64322.867486]  o?a?a?a+0x5c/0x68
Tue Aug  2 19:16:07 2022 kern.info kernel: [64322.867774]  o_??sl?i??a?t+0x40/0x58
Tue Aug  2 19:16:07 2022 kern.info kernel: [64322.868100]  o?e???t+0x3c/0x88
Tue Aug  2 19:16:07 2022 kern.info kernel: [64322.868381]  ?1_??t+0x40/0x60
Tue Aug  2 19:16:07 2022 kern.info kernel: [64322.868657]  ?1h_??n?hl?+0x5c/0xd0
Tue Aug  2 19:16:07 2022 kern.info kernel: [64322.868980]  ?1h_??nc+0x64/0x68
Tue Aug  2 19:16:07 2022 kern.info kernel: [64322.869269]  ????d?p?e+0x4c/0x2b8
Tue Aug  2 19:16:07 2022 kern.info kernel: [64322.869579]  i?#??+0x3dc/0x4a0
Tue Aug  2 19:16:07 2022 kern.info kernel: [64322.869880]  ?ƣ?a?t+0xa50/0xb78
Tue Aug  2 19:16:07 2022 kern.info kernel: [64322.870189]  hƣ?a?t+0x14c/0x1d8
Tue Aug  2 19:16:07 2022 kern.info kernel: [64322.870498]  o??fa?t+0x1c4/0x380
Tue Aug  2 19:16:07 2022 kern.info kernel: [64322.870808]  o_??sl?i??a?t+0x50/0x58
Tue Aug  2 19:16:07 2022 kern.info kernel: [64322.871134]  o?e???t+0x3c/0x88
Tue Aug  2 19:16:07 2022 kern.info kernel: [64322.871415]  ?0_ia+0x64/0xc0
Tue Aug  2 19:16:07 2022 kern.info kernel: [64322.871685]  ?0???n?hl?+0x70/0xb0
Tue Aug  2 19:16:07 2022 kern.info kernel: [64322.872001]  ?0???nc+0x15c/0x160
Tue Aug  2 19:17:10 2022 kern.err kernel: [64385.877697] rcu: INFO: rcu_preempt detected stalls on CPUs/tasks:
Tue Aug  2 19:17:10 2022 kern.err kernel: [64385.878244] rcu: 	Tasks blocked on level-0 rcu_node (CPUs 0-5): P9413/1:b..l P9416/1:b..l P9414/1:b..l P9415/1:b..l
Tue Aug  2 19:17:10 2022 kern.warn kernel: [64385.879173] 	(detected by 5, t=5284530 jiffies, g=1493841, q=3087608)
Tue Aug  2 19:17:10 2022 kern.info kernel: [64385.879739] task:grep            state:D stack:    0 pid: 9415 ppid:     1 flags:0x00000000
Tue Aug  2 19:17:10 2022 kern.info kernel: [64385.880472] Call trace:
Tue Aug  2 19:17:10 2022 kern.info kernel: [64385.880688]  ?w??_?+0xe4/0x198
Tue Aug  2 19:17:10 2022 kern.info kernel: [64385.880971]  ?ߢu?+0x1dc/0x4a8
Tue Aug  2 19:17:10 2022 kern.info kernel: [64385.881243]  ߢu?+0x50/0xc8
Tue Aug  2 19:17:10 2022 kern.info kernel: [64385.881492]  wse?d?n_`??wp?h+0x384/0x568
Tue Aug  2 19:17:10 2022 kern.info kernel: [64385.881840]  ?n_`e+0x34/0x50
Tue Aug  2 19:17:10 2022 kern.info kernel: [64385.882097]  Ԯm+0x2c/0x108
Tue Aug  2 19:17:10 2022 kern.info kernel: [64385.882358]  ??t+0x60/0x128
Tue Aug  2 19:17:10 2022 kern.info kernel: [64385.882609]  o_??+0x248/0x8a8
Tue Aug  2 19:17:10 2022 kern.info kernel: [64385.882873]  ?k?tݫ??+0x70/0xf0
Tue Aug  2 19:17:10 2022 kern.info kernel: [64385.883150]  ie+0x200/0x230
Tue Aug  2 19:17:10 2022 kern.info kernel: [64385.883400]  i?k?n??a?t+0x380/0x390
Tue Aug  2 19:17:10 2022 kern.info kernel: [64385.883707]  ?o_k?n??a?t+0xf4/0x178
Tue Aug  2 19:17:10 2022 kern.info kernel: [64385.884017]  o?a?a?a+0x5c/0x68
Tue Aug  2 19:17:10 2022 kern.info kernel: [64385.884289]  o_??sl?i??a?t+0x40/0x58
Tue Aug  2 19:17:10 2022 kern.info kernel: [64385.884604]  o?e???t+0x3c/0x88
Tue Aug  2 19:17:10 2022 kern.info kernel: [64385.884874]  ?1_??t+0x40/0x60
Tue Aug  2 19:17:10 2022 kern.info kernel: [64385.885139]  ?1h_??n?hl?+0x5c/0xd0
Tue Aug  2 19:17:10 2022 kern.info kernel: [64385.885448]  ?1h_??nc+0x64/0x68
Tue Aug  2 19:17:10 2022 kern.info kernel: [64385.885725]  ????d?p?e+0x4c/0x2b8
Tue Aug  2 19:17:10 2022 kern.info kernel: [64385.886021]  i?#??+0x3dc/0x4a0
Tue Aug  2 19:17:10 2022 kern.info kernel: [64385.886308]  ?ƣ?a?t+0xa50/0xb78
Tue Aug  2 19:17:10 2022 kern.info kernel: [64385.886605]  hƣ?a?t+0x14c/0x1d8
Tue Aug  2 19:17:10 2022 kern.info kernel: [64385.886899]  o??fa?t+0x1c4/0x380
Tue Aug  2 19:17:10 2022 kern.info kernel: [64385.887194]  o_??sl?i??a?t+0x50/0x58
Tue Aug  2 19:17:10 2022 kern.info kernel: [64385.887510]  o?e???t+0x3c/0x88
Tue Aug  2 19:17:10 2022 kern.info kernel: [64385.887780]  ?0?a+0x20/0x50
Tue Aug  2 19:17:10 2022 kern.info kernel: [64385.888027]  ?0???n?hl?+0x60/0xb0
Tue Aug  2 19:17:10 2022 kern.info kernel: [64385.888329]  ?0???nc+0x15c/0x160
Tue Aug  2 19:17:10 2022 kern.info kernel: [64385.888615] task:grep            state:D stack:    0 pid: 9414 ppid:     1 flags:0x00000008
Tue Aug  2 19:17:10 2022 kern.info kernel: [64385.889347] Call trace:
Tue Aug  2 19:17:10 2022 kern.info kernel: [64385.889563]  ?w??_?+0xe4/0x198
Tue Aug  2 19:17:10 2022 kern.info kernel: [64385.889835]  ?ߢu?+0x1dc/0x4a8
Tue Aug  2 19:17:10 2022 kern.info kernel: [64385.890107]  ߢu?+0x50/0xc8
[... 60 lines cut ...]
Tue Aug  2 19:17:10 2022 kern.info kernel: [64385.908039]  wse?d?n_`??wp?h+0x384/0x568
Tue Aug  2 19:17:10 2022 kern.info kernel: [64385.908386]  ?n_`e+0x34/0x50
Tue Aug  2 19:17:10 2022 kern.info kernel: [64385.908644]  Ԯm+0x2c/0x108
Tue Aug  2 19:17:10 2022 kern.info kernel: [64385.908900]  ??t+0x60/0x128
Tue Aug  2 19:17:10 2022 kern.info kernel: [64385.909149]  o_??+0x248/0x8a8
Tue Aug  2 19:17:10 2022 kern.info kernel: [64385.909412]  ?k?tݫ??+0x70/0xf0
Tue Aug  2 19:17:10 2022 kern.info kernel: [64385.909691]  ie+0x200/0x230
Tue Aug  2 19:17:10 2022 kern.info kernel: [64385.909939]  i?k?n??a?t+0x380/0x390
Tue Aug  2 19:17:10 2022 kern.info kernel: [64385.910247]  ?o_k?n??a?t+0xf4/0x178
Tue Aug  2 19:17:10 2022 kern.info kernel: [64385.910557]  o?a?a?a+0x5c/0x68
Tue Aug  2 19:17:10 2022 kern.info kernel: [64385.910829]  o_??sl?i??a?t+0x40/0x58
Tue Aug  2 19:17:10 2022 kern.info kernel: [64385.911145]  o?e???t+0x3c/0x88
Tue Aug  2 19:17:10 2022 kern.info kernel: [64385.911415]  ?1_??t+0x40/0x60
Tue Aug  2 19:17:10 2022 kern.info kernel: [64385.911678]  ?1h_??n?hl?+0x5c/0xd0
Tue Aug  2 19:17:10 2022 kern.info kernel: [64385.911987]  ?1h_??nc+0x64/0x68
Tue Aug  2 19:17:10 2022 kern.info kernel: [64385.912265]  ????d?p?e+0x4c/0x2b8
Tue Aug  2 19:17:10 2022 kern.info kernel: [64385.912558]  i?#??+0x3dc/0x4a0
Tue Aug  2 19:17:10 2022 kern.info kernel: [64385.912845]  ?ƣ?a?t+0xa50/0xb78
Tue Aug  2 19:17:10 2022 kern.info kernel: [64385.913138]  hƣ?a?t+0x14c/0x1d8
Tue Aug  2 19:17:10 2022 kern.info kernel: [64385.913432]  o??fa?t+0x1c4/0x380
Tue Aug  2 19:17:10 2022 kern.info kernel: [64385.913727]  o_??sl?i??a?t+0x50/0x58
Tue Aug  2 19:17:10 2022 kern.info kernel: [64385.914041]  o?e???t+0x3c/0x88
Tue Aug  2 19:17:10 2022 kern.info kernel: [64385.914311]  ?0_ia+0x64/0xc0
Tue Aug  2 19:17:10 2022 kern.info kernel: [64385.914567]  ?0???n?hl?+0x70/0xb0
Tue Aug  2 19:17:10 2022 kern.info kernel: [64385.914867]  ?0???nc+0x15c/0x160

moonerick, 2022-08-02:
换个终端登录试试

keeno1982, 2022-08-03:
@moonerick 跟终端没关系，这是登录路由后台看的，一开始是好的，运行一段时间后才变成这样

coolsnowwolf, 2022-08-03:
serverchan 崩了

keeno1982, 2022-08-03:
谢谢老大回复，我关掉serverchan观察一下

keeno1982, 2022-08-05:
关掉观察了两天，确实正常了

closed 2022-08-05 by keeno1982