coolsnowwolf/lede #9886: r68s,重启openclash导致机器掉线，具体请看日志；掉线表现为 公网无法通过预先配置的端口映射远程访问

bigguyong, 2022-08-03:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
重启openclash导致机器掉线，具体请看日志；掉线表现为 公网无法通过预先配置的端口映射远程访问
A：

### (2) 路由器型号和固件版本
r68s 4g
A：

### (3) 详细日志
Fri Aug  4 17:00:30 2017 daemon.notice procd: /etc/rc.d/S99openclash: OpenClash Already Stop!
Fri Aug  4 17:00:30 2017 daemon.info qbittorrent: Instance "main" disabled.
Fri Aug  4 17:00:30 2017 daemon.notice procd: /etc/rc.d/S99qbittorrent: Command failed: Invalid argument
Fri Aug  4 17:00:31 2017 daemon.err ttyd[6050]: [2017/08/04 17:00:31:0609] NOTICE: ttyd 1.5.2-815bb96c (libwebsockets 3.1.0)
Fri Aug  4 17:00:31 2017 daemon.err ttyd[6050]: [2017/08/04 17:00:31:0609] NOTICE: tty configuration:
Fri Aug  4 17:00:31 2017 daemon.err ttyd[6050]: [2017/08/04 17:00:31:0610] NOTICE:   start command: /bin/login
Fri Aug  4 17:00:31 2017 daemon.err ttyd[6050]: [2017/08/04 17:00:31:0610] NOTICE:   close signal: SIGHUP (1)
Fri Aug  4 17:00:31 2017 daemon.err ttyd[6050]: [2017/08/04 17:00:31:0610] NOTICE:   terminal type: xterm-256color
Fri Aug  4 17:00:31 2017 daemon.err ttyd[6050]: [2017/08/04 17:00:31:0612] NOTICE: Creating Vhost 'default' port 7681, 2 protocols, IPv6 off
Fri Aug  4 17:00:36 2017 daemon.info procd: - init complete -
Fri Aug  4 17:00:39 2017 user.info mwan3rtmon[5451]: Detect rtchange event.
Fri Aug  4 17:00:39 2017 user.notice firewall: Reloading firewall due to ifup of lan (br-lan)
Fri Aug  4 17:00:39 2017 cron.info crond[6863]: crond (busybox 1.30.1) started, log level 8
Fri Aug  4 17:00:40 2017 daemon.info dnsmasq[6947]: Connected to system UBus
Fri Aug  4 17:00:40 2017 daemon.info dnsmasq[7094]: Connected to system UBus
Fri Aug  4 17:00:40 2017 user.notice nlbwmon: Reloading nlbwmon due to ifup of lan (br-lan)
Fri Aug  4 17:00:51 2017 daemon.info procd: Instance nginx::instance1 s in a crash loop 6 crashes, 2 seconds since last crash
Fri Aug  4 17:00:54 2017 user.notice mwan3[7432]: Execute ifup event on interface wan2 (pppoe-wan2)
Fri Aug  4 17:00:54 2017 user.notice mwan3[7432]: Starting tracker on interface wan2 (pppoe-wan2)
Fri Aug  4 17:00:55 2017 user.info mwan3rtmon[5451]: Detect rtchange event.
Fri Aug  4 17:00:55 2017 user.notice firewall: Reloading firewall due to ifup of wan2 (pppoe-wan2)
Fri Aug  4 17:00:55 2017 cron.info crond[8540]: crond (busybox 1.30.1) started, log level 8
Fri Aug  4 17:00:55 2017 daemon.info dnsmasq[8598]: Connected to system UBus
Fri Aug  4 17:00:56 2017 daemon.info dnsmasq[8711]: Connected to system UBus
Fri Aug  4 17:00:56 2017 user.notice nlbwmon: Reloading nlbwmon due to ifup of wan2 (pppoe-wan2)
A：

WYC-2020, 2022-08-03:
隔壁村

d975xbx, 2022-08-04:
openclash不是这里的。

closed 2022-08-07 by coolsnowwolf