coolsnowwolf/lede #9894: 最新源码的x86 64的i210网卡驱动有问题？只能开一队列了

a904055262, 2022-08-06:
新固件：
[    9.681729] igb 0000:06:10.0: added PHC on eth0
[    9.683126] igb 0000:06:10.0: Intel(R) Gigabit Ethernet Linux Driver
[    9.684743] igb 0000:06:10.0: eth0: (PCIe:unknown:unknown)
[    9.688927] igb 0000:06:10.0 eth0: MAC: c4:83:4f:16:2a:00
[    9.690398] igb 0000:06:10.0: eth0: PBA No: Unknown
[    9.691800] igb 0000:06:10.0: LRO is disabled
[    9.693128] igb 0000:06:10.0: Using M**SI-X interrupts. 1 rx queue(s), 1 tx queue(s)**
[    9.861440] igb 0000:06:11.0: added PHC on eth1
[    9.862807] igb 0000:06:11.0: Intel(R) Gigabit Ethernet Linux Driver
[    9.864323] igb 0000:06:11.0: eth1: (PCIe:unknown:unknown)
[    9.865947] igb 0000:06:11.0 eth1: MAC: c4:83:4f:16:2a:01
[    9.867362] igb 0000:06:11.0: eth1: PBA No: Unknown
[    9.868733] igb 0000:06:11.0: LRO is disabled
[    9.870014] igb 0000:06:11.0: Using MSI-X interrupts. 1 rx queue(s), 1 tx queue(s)
[    9.877850] igbvf: Intel(R) Gigabit Virtual Function Network Driver
[    9.879329] igbvf: Copyright (c) 2009 - 2012 Intel Corporation.
[   18.749734] igb 0000:06:10.0 eth0: igb: eth0 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX/TX
[   20.153730] igb 0000:06:11.0 **eth1: igb: eth1 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: None**

6月份的固件：

[    3.176678] igb: Intel(R) Gigabit Ethernet Network Driver
[    3.178137] igb: Copyright (c) 2007-2014 Intel Corporation.
[    3.229740] igb 0000:06:10.0: added PHC on eth0
[    3.231077] igb 0000:06:10.0: Intel(R) Gigabit Ethernet Network Connection
[    3.232655] igb 0000:06:10.0: eth0: (PCIe:unknown:unknown) c4:83:4f:16:2a:00
[    3.234244] igb 0000:06:10.0: eth0: PBA No: FFFFFF-0FF
[    3.235603] igb 0000:06:10.0: **Using MSI-X interrupts. 4 rx queue(s), 4 tx queue(s)**
[    3.290976] igb 0000:06:11.0: added PHC on eth1
[    3.292359] igb 0000:06:11.0: Intel(R) Gigabit Ethernet Network Connection
[    3.294142] igb 0000:06:11.0: eth1: (PCIe:unknown:unknown) c4:83:4f:16:2a:01
[    3.295810] igb 0000:06:11.0: eth1: PBA No: FFFFFF-0FF
[    3.297213] igb 0000:06:11.0: Using MSI-X interrupts. 4 rx queue(s), 4 tx queue(s)
[    6.671140] igb 0000:06:10.0 eth0: igb: eth0 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX/TX
[    9.972966] igbvf: Intel(R) Gigabit Virtual Function Network Driver
[    9.974562] igbvf: Copyright (c) 2009 - 2012 Intel Corporation.
[   19.358786] igb 0000:06:10.0 eth0: igb: eth0 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX/TX
[   20.923209] igb 0000:06:11.0 **eth1: igb: eth1 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX**

WYC-2020, 2022-08-06:
igb 切换到了inter 你切回来编译看下

a904055262, 2022-08-06:
> igb 切换到了inter 你切回来编译看下

我不会。。。。。。

coolsnowwolf, 2022-08-07:
src/igb_param.c 

闭源驱动默认的 RSS 数值是 1

[image]

你可以打个补丁改为 4 试试，印象中 I211 只能支持  2 ？

a904055262, 2022-08-07:
> src/igb_param.c
> 
> 闭源驱动默认的 RSS 数值是 1
> 
> [image]
> 
> 你可以打个补丁改为 4 试试，印象中 I211 只能支持 2 ？

210支持4队列，211是2队列。

KscapeDx, 2022-08-07:
所以有啥修改的办法吗

a904055262, 2022-08-09:
> src/igb_param.c
> 
> 闭源驱动默认的 RSS 数值是 1
> 
> [image]
> 
> 你可以打个补丁改为 4 试试，印象中 I211 只能支持 2 ？

修改后dmesg的确可以显示4队列了，不过flowcontol还是默认none，而且我简单复制文件跑满千M的内网CPU占用要10%左右，而原来是驱动只有5-7之间跳，大佬能不能切回去原来的驱动。

ShoukakuChuuCC, 2022-08-10:
> > src/igb_param.c
> > 闭源驱动默认的 RSS 数值是 1
> > [image]
> > 你可以打个补丁改为 4 试试，印象中 I211 只能支持 2 ？
> 
> 修改后dmesg的确可以显示4队列了，不过flowcontol还是默认none，而且我简单复制文件跑满千M的内网CPU占用要10%左右，而原来是驱动只有5-7之间跳，大佬能不能切回去原来的驱动。

自己切换下就行 make menuconfig-kernel modules-Network Device-取消kmod-igb-intel 选择kmod-igb

a904055262, 2022-08-10:
> 

谢谢大佬

baodp, 2022-08-13:
I350 也是：
 Using MSI-X interrupts. 1 rx queue(s), 1 tx queue(s)

closed 2022-08-16 by coolsnowwolf