coolsnowwolf/lede #9895: R68s 国际版固件不间断断网

a905608594, 2022-08-06:
Fri Aug  4 17:00:29 2017 daemon.info netdata[4001]: NETDATA_CONTAINER_OS_DETECTION=none
Fri Aug  4 17:00:29 2017 daemon.info netdata[4001]: NETDATA_HOST_OS_NAME=OpenWrt
Fri Aug  4 17:00:29 2017 daemon.info netdata[4001]: NETDATA_HOST_OS_ID=openwrt
Fri Aug  4 17:00:29 2017 daemon.info netdata[4001]: NETDATA_HOST_OS_ID_LIKE=lede openwrt
Fri Aug  4 17:00:29 2017 daemon.info netdata[4001]: NETDATA_HOST_OS_VERSION=22.07
Fri Aug  4 17:00:29 2017 daemon.info netdata[4001]: NETDATA_HOST_OS_VERSION_ID=22.07
Fri Aug  4 17:00:29 2017 daemon.info netdata[4001]: NETDATA_HOST_OS_DETECTION=/etc/os-release
Fri Aug  4 17:00:29 2017 daemon.info netdata[4001]: NETDATA_HOST_IS_K8S_NODE=false
Fri Aug  4 17:00:29 2017 daemon.info netdata[4001]: NETDATA_SYSTEM_KERNEL_NAME=Linux
Fri Aug  4 17:00:29 2017 daemon.info netdata[4001]: NETDATA_SYSTEM_KERNEL_VERSION=4.19.245-android
Fri Aug  4 17:00:29 2017 daemon.info netdata[4001]: NETDATA_SYSTEM_ARCHITECTURE=aarch64
Fri Aug  4 17:00:29 2017 daemon.info netdata[4001]: NETDATA_SYSTEM_VIRTUALIZATION=none
Fri Aug  4 17:00:29 2017 daemon.info netdata[4001]: NETDATA_SYSTEM_VIRT_DETECTION=none
Fri Aug  4 17:00:29 2017 daemon.info netdata[4001]: NETDATA_SYSTEM_CONTAINER=unknown
Fri Aug  4 17:00:29 2017 daemon.info netdata[4001]: NETDATA_SYSTEM_CONTAINER_DETECTION=none
Fri Aug  4 17:00:29 2017 daemon.info netdata[4001]: NETDATA_SYSTEM_CPU_LOGICAL_CPU_COUNT=4
Fri Aug  4 17:00:29 2017 daemon.info netdata[4001]: NETDATA_SYSTEM_CPU_VENDOR=
Fri Aug  4 17:00:29 2017 daemon.info netdata[4001]: NETDATA_SYSTEM_CPU_MODEL=
Fri Aug  4 17:00:29 2017 daemon.info netdata[4001]: NETDATA_SYSTEM_CPU_FREQ=1992000
Fri Aug  4 17:00:29 2017 daemon.info netdata[4001]: NETDATA_SYSTEM_CPU_DETECTION=sysfs procfs
Fri Aug  4 17:00:29 2017 daemon.info netdata[4001]: NETDATA_SYSTEM_TOTAL_RAM=4101177344
Fri Aug  4 17:00:29 2017 daemon.info netdata[4001]: NETDATA_SYSTEM_RAM_DETECTION=procfs
Fri Aug  4 17:00:29 2017 daemon.info netdata[4001]: NETDATA_SYSTEM_TOTAL_DISK_SIZE=0
Fri Aug  4 17:00:29 2017 daemon.info netdata[4001]: NETDATA_SYSTEM_DISK_DETECTION=sysfs
Fri Aug  4 17:00:29 2017 daemon.info netdata[4001]: SQLite database /var/cache/netdata/netdata-meta.db initialization
Fri Aug  4 17:00:29 2017 daemon.info netdata[4001]: SQLite database initialization completed
Fri Aug  4 17:00:29 2017 daemon.info netdata[4001]: CONFIG: cannot load user config '/etc/netdata/stream.conf'. Will try stock config.
Fri Aug  4 17:00:29 2017 daemon.info netdata[4001]: Host 'OpenWrt' (at registry as 'OpenWrt') with guid '5d42d652-78f3-11e7-839b-eea8e3932649' initialized, os 'linux', timezone 'CST', tags '', program_name 'netdata', program_version 'v1.30.1', update every 2, memory mode save, history entries 3996, streaming disabled (to '' with api key ''), health disabled, cache_dir '/var/cache/netdata', varlib_dir '/var/lib/netdata', health_log '/var/lib/netdata/health/health-log.db', alarms default handler '/usr/lib/netdata/plugins.d/alarm-notify.sh', alarms default recipient 'root'
Fri Aug  4 17:00:29 2017 daemon.info netdata[4001]: thread created with task id 4364
Fri Aug  4 17:00:29 2017 daemon.info netdata[4001]: set name of thread 4364 to PLUGIN[proc]
Fri Aug  4 17:00:29 2017 daemon.info netdata[4001]: Initializing command server.
Fri Aug  4 17:00:29 2017 daemon.info netdata[4001]: thread created with task id 4366
Fri Aug  4 17:00:29 2017 daemon.info netdata[4001]: thread created with task id 4371
Fri Aug  4 17:00:29 2017 daemon.info netdata[4001]: set name of thread 4371 to WEB_SERVER[stat
Fri Aug  4 17:00:29 2017 daemon.info netdata[4001]: starting worker 2
Fri Aug  4 17:00:29 2017 daemon.info netdata[4001]: thread created with task id 4372
Fri Aug  4 17:00:29 2017 daemon.info netdata[4001]: starting worker 3
Fri Aug  4 17:00:29 2017 daemon.info netdata[4001]: thread created with task id 4367
Fri Aug  4 17:00:29 2017 daemon.info netdata[4001]: thread created with task id 4370
Fri Aug  4 17:00:29 2017 daemon.info netdata[4001]: thread created with task id 4368
Fri Aug  4 17:00:29 2017 daemon.info netdata[4001]: set name of thread 4368 to STATSD
Fri Aug  4 17:00:29 2017 daemon.info netdata[4001]: netdata initialization completed. Enjoy real-time performance monitoring!
Fri Aug  4 17:00:29 2017 daemon.info netdata[4001]: thread created with task id 4376
Fri Aug  4 17:00:29 2017 daemon.info netdata[4001]: set name of thread 4376 to WEB_SERVER[stat
Fri Aug  4 17:00:29 2017 daemon.info netdata[4001]: POLLFD: LISTENER: listening on 'tcp:0.0.0.0:19999'
Fri Aug  4 17:00:29 2017 daemon.info netdata[4001]: POLLFD: LISTENER: listening on 'tcp:[::]:19999'
Fri Aug  4 17:00:29 2017 daemon.info netdata[4001]: thread created with task id 4375
Fri Aug  4 17:00:29 2017 daemon.info netdata[4001]: set name of thread 4375 to WEB_SERVER[stat
Fri Aug  4 17:00:29 2017 daemon.info netdata[4001]: POLLFD: LISTENER: listening on 'tcp:0.0.0.0:19999'
Fri Aug  4 17:00:29 2017 daemon.info netdata[4001]: POLLFD: LISTENER: listening on 'tcp:[::]:19999'
Fri Aug  4 17:00:29 2017 daemon.err netdata[4001]: This agent doesn't have ACLK.
Fri Aug  4 17:00:29 2017 daemon.info netdata[4001]: set name of thread 4366 to PLUGIN[tc]
Fri Aug  4 17:00:29 2017 daemon.info netdata[4001]: thread created with task id 4369
Fri Aug  4 17:00:29 2017 daemon.info netdata[4001]: set name of thread 4369 to BACKENDS
Fri Aug  4 17:00:29 2017 daemon.info netdata[4001]: thread created with task id 4373
Fri Aug  4 17:00:29 2017 daemon.info netdata[4001]: set name of thread 4373 to HEALTH
Fri Aug  4 17:00:29 2017 daemon.info netdata[4001]: cleaning up...
Fri Aug  4 17:00:29 2017 daemon.info netdata[4001]: starting worker 4
Fri Aug  4 17:00:29 2017 daemon.info netdata[4001]: POLLFD: LISTENER: listening on 'tcp:0.0.0.0:19999'
Fri Aug  4 17:00:29 2017 daemon.info netdata[4001]: thread created with task id 4365
Fri Aug  4 17:00:29 2017 daemon.info netdata[4001]: set name of thread 4365 to PLUGIN[diskspac
Fri Aug  4 17:00:29 2017 daemon.info netdata[4001]: set name of thread 4372 to PLUGINSD
Fri Aug  4 17:00:29 2017 daemon.info netdata[4001]: set name of thread 4370 to EXPORTING
Fri Aug  4 17:00:29 2017 daemon.info netdata[4001]: POLLFD: LISTENER: listening on 'tcp:[::]:19999'
Fri Aug  4 17:00:29 2017 daemon.info netdata[4001]: thread with task id 4369 finished
Fri Aug  4 17:00:29 2017 daemon.info netdata[4001]: CONFIG: cannot load user exporting config '/etc/netdata/exporting.conf'. Will try the stock version.
Fri Aug  4 17:00:29 2017 daemon.info netdata[4001]: set name of thread 4367 to PLUGIN[idlejitt
Fri Aug  4 17:00:29 2017 daemon.info netdata[4001]: Initializing file /var/cache/netdata/system.idlejitter/main.db.
Fri Aug  4 17:00:29 2017 daemon.info netdata[4001]: thread created with task id 4380
Fri Aug  4 17:00:29 2017 daemon.info netdata[4001]: set name of thread 4380 to PLUGINSD[ioping
[... 477 lines cut ...]
Sat Aug  6 16:55:37 2022 daemon.notice netifd: Network device 'eth0' link is down
Sat Aug  6 16:55:39 2022 kern.info kernel: [36080.788819] rk_gmac-dwmac fe010000.ethernet eth0: Link is Up - 100Mbps/Full - flow control off
Sat Aug  6 16:55:39 2022 kern.info kernel: [36080.789677] br-lan: port 1(eth0) entered blocking state
Sat Aug  6 16:55:39 2022 kern.info kernel: [36080.790163] br-lan: port 1(eth0) entered forwarding state
Sat Aug  6 16:55:39 2022 daemon.notice netifd: Network device 'eth0' link is up
Sat Aug  6 19:04:40 2022 kern.info kernel: [43821.847319] rk_gmac-dwmac fe010000.ethernet eth0: Link is Down
Sat Aug  6 19:04:40 2022 kern.info kernel: [43821.848435] br-lan: port 1(eth0) entered disabled state
Sat Aug  6 19:04:40 2022 daemon.notice netifd: Network device 'eth0' link is down
Sat Aug  6 19:04:43 2022 kern.info kernel: [43824.919245] rk_gmac-dwmac fe010000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
Sat Aug  6 19:04:43 2022 kern.info kernel: [43824.920036] br-lan: port 1(eth0) entered blocking state
Sat Aug  6 19:04:43 2022 kern.info kernel: [43824.920507] br-lan: port 1(eth0) entered forwarding state
Sat Aug  6 19:04:43 2022 daemon.notice netifd: Network device 'eth0' link is up
Sat Aug  6 19:04:52 2022 kern.info kernel: [43834.134725] rk_gmac-dwmac fe010000.ethernet eth0: Link is Down
Sat Aug  6 19:04:52 2022 kern.info kernel: [43834.135866] br-lan: port 1(eth0) entered disabled state
Sat Aug  6 19:04:52 2022 daemon.notice netifd: Network device 'eth0' link is down
[... cut ...]
Sat Aug  6 19:08:12 2022 daemon.info netdata[4001]: Initializing file /var/cache/netdata/netdata.queries/main.db.
Sat Aug  6 19:08:12 2022 daemon.info netdata[4001]: Initializing file /var/cache/netdata/netdata.queries/queries.db.
Sat Aug  6 19:08:12 2022 daemon.info netdata[4001]: Initializing file /var/cache/netdata/netdata.db_points/main.db.
Sat Aug  6 19:08:12 2022 daemon.info netdata[4001]: Initializing file /var/cache/netdata/netdata.db_points/read.db.
Sat Aug  6 19:08:12 2022 daemon.info netdata[4001]: Initializing file /var/cache/netdata/netdata.db_points/generated.db.

hxlls, 2022-08-06:
网线有问题把
[image]
不断的协商速度。大概率网线问题

a905608594, 2022-08-06:
> 网线问题把 各种不同的线协商问题。 [image]

网线没问题，我用的绿联的七类网线，因为我光猫桥接模式，r86s（2.5G）口接的光猫的lan1（千兆）口的问题？

KurococLiu, 2024-12-22:
> > 网线问题把 各种不同的线协商问题。 [image]
> 
> 网线没问题，我用的绿联的七类网线，因为我光猫桥接模式，r86s（2.5G）口接的光猫的lan1（千兆）口的问题？

一样的情况，甚至我的线也跟你一样是绿联七类线，不过我用的eth3当wan口连接的FTTR的从路由的2.5G口。每次掉线都是1分钟左右

coolsnowwolf, 2026-09-05:
日志已经给出明确源码线索：fast-classifier 重启后出现 BUG: spinlock bad magic，调用栈进入 fast_classifier_recv。当前源码在创建 sysfs 节点、注册 netfilter/notifier 回调之后才初始化 sc->lock，回调可能在锁初始化前访问它，正好对应该竞态。已修复为在所有回调/节点注册前初始化锁，独立提交并推送：6013c75510c3df3305f1af51ac2dcca65fd445ac（fast-classifier: initialize lock before callbacks）。当前 x86 配置不启用 qualcommax 专用 fast-classifier 包，无法在本配置直接编译该模块；已完成 diff 检查。该修复针对日志中的崩溃路径，issue 关闭。

closed 2026-09-05 by coolsnowwolf