coolsnowwolf/lede #9901: Docker下ubuntu无法使用arm-linux-gcc交叉编译器

zhou-daxia, 2022-08-08:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
可能与这个项目无关，实在找不到什么原因才上来请教各位大佬。
我想在openwrt下通过docker部署一个开发环境，但出现docker->ubuntu下无法使用arm gcc交叉编译器问题。

### (1) 具体问题

A：
Docker部署了ubuntu16(x86_64)，已经安装了相关的32位支持库(libc6-i386 libc6-dev-i386等)，但还是无法执行32位的arm-linux-gcc.提示:arm-linux-gcc: cannot execute binary file: Exec format error。整个交叉编译器也给了777权限，编译器在虚拟机下可正常使用。

### (2) 路由器型号和固件版本

A：
J4125  
git-21.335.48743-5f363d9

### (3) 详细日志

A：
Docker：
[image]
主机：
[image]

a393310872, 2022-08-10:
也太难为J4125了,ESXI加OPENWRT加DOCKER加UBUNTU....

zhou-daxia, 2022-08-11:
> 也太难为J4125了,ESXI加OPENWRT加DOCKER加UBUNTU....

哈哈哈，虚拟机在其它x86主机上试的，不在J4125

closed 2022-08-12 by coolsnowwolf