coolsnowwolf/lede #9905: drm_display_helper.ko is missing

moonerick, 2022-08-09:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：仅修改lede\target\linux\x86\Makefile中的KERNEL_TESTING_PATCHVER:=5.19改成了5.18或者修改KERNEL_PATCHVER:=5.15改成5.18进行编译，均无法通过编译，已经执行过make clean 和缓存删除操作，错误日志都是下面这个一样的

### (2) 路由器型号和固件版本

A：X86

### (3) 详细日志

A：
Packaged contents of /home/sp/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-drm into /home/sp/lede/bin/targets/x86/64/packages/kmod-drm_5.18.16-1_x86_64.ipk
NOTICE: module '/home/sp/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.18.16/drivers/gpu/drm/ttm/ttm.ko' is built-in.
Packaged contents of /home/sp/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-drm-ttm into /home/sp/lede/bin/targets/x86/64/packages/kmod-drm-ttm_5.18.16-1_x86_64.ipk
NOTICE: module '/home/sp/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.18.16/drivers/gpu/drm/drm_kms_helper.ko' is built-in.
Packaged contents of /home/sp/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-drm-kms-helper into /home/sp/lede/bin/targets/x86/64/packages/kmod-drm-kms-helper_5.18.16-1_x86_64.ipk
NOTICE: module '/home/sp/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.18.16/drivers/gpu/drm/i915/i915.ko' is built-in.
NOTICE: module '/home/sp/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.18.16/drivers/gpu/drm/drm_buddy.ko' is built-in.
ERROR: module '/home/sp/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.18.16/drivers/gpu/drm/display/drm_display_helper.ko' is missing.
make[3]: *** [modules/video.mk:1111: /home/sp/lede/bin/targets/x86/64/packages/kmod-drm-i915_5.18.16-1_x86_64.ipk] Error 1
make[3]: Leaving directory '/home/sp/lede/package/kernel/linux'
time: package/kernel/linux/compile#6.85#0.68#9.99
    ERROR: package/kernel/linux failed to build.
make[2]: *** [package/Makefile:116: package/kernel/linux/compile] Error 1

xiangfeidexiaohuo, 2022-08-09:
重新同步源码，看提交修复了呀

moonerick, 2022-08-09:
> 
:~/lede$ git pull
Already up to date.
:~/lede$ git pull --force
Already up to date.

不知道什么原因，等晚上在来搞，不行多半只有删除重新pull

Wilson-coder, 2022-08-09:
一样的情况。

后来想着路由器也不用来做视频解码，就把AMDGPU和i915两个驱动都去除了，然后就成功编译了5.19，现在用了快一天，除了看数字差异之外，真实没啥感觉。

xiangfeidexiaohuo, 2022-08-09:
> 一样的情况。
> 
> 后来想着路由器也不用来做视频解码，就把AMDGPU和i915两个驱动都去除了，然后就成功编译了5.19，现在用了快一天，除了看数字差异之外，真实没啥感觉。

只i915可以编译的，amd这些不可以，会提示上面那个日志。

moonerick, 2022-08-09:
> > 一样的情况。
> > 后来想着路由器也不用来做视频解码，就把AMDGPU和i915两个驱动都去除了，然后就成功编译了5.19，现在用了快一天，除了看数字差异之外，真实没啥感觉。
> 
> 只i915可以编译的，amd这些不可以，会提示上面那个日志。

昨晚上已经将LEDE整个文件夹进行了删除，重新clone的源码，编译依然是这个问题

closed 2022-08-10 by coolsnowwolf