coolsnowwolf/lede #9916: mac80211 编译错误

1054009064, 2022-08-11:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：mac80211 编译错误

### (2) 路由器型号和固件版本

A：ipq807x / f66ea169c0ec229f46c32d886f7a0411f909ad2a

### (3) 详细日志

A：
<details><summary>Log</summary>

```
make[3]: Entering directory '/home/build/lede/package/kernel/mac80211'
mkdir -p /home/build/lede/dl
SHELL= flock /home/build/lede/tmp/.backports-5.15.33-1.tar.xz.flock -c '         /home/build/lede/scripts/download.pl "/home/build/lede/dl" "backports-5.15.33-1.tar.xz" "1b6b3bded4c81814ebebe2d194c2f8966d2399005b85ebb0557285b6e73f5422" "" "@KERNEL/linux/kernel/projects/backports/stable/v5.15.33/"    '
rm -f /home/build/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/backports-5.15.33-1/.built
touch /home/build/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/backports-5.15.33-1/.built_check
. /home/build/lede/include/shell.sh; var2file "V_mac80211_config" /home/build/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/backports-5.15.33-1/.config
make -C "/home/build/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/backports-5.15.33-1" KCFLAGS="-fmacro-prefix-map=/home/build/lede/build_dir/target-aarch64_cortex-a53_musl=target-aarch64_cortex-a53_musl" HOSTCFLAGS="-O2 -I/home/build/lede/staging_dir/host/include -I/home/build/lede/staging_dir/hostpkg/include -I/home/build/lede/staging_dir/target-aarch64_cortex-a53_musl/host/include -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="aarch64-openwrt-linux-musl-" ARCH="arm64" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Thu Aug 11 01:01:23 2022" KBUILD_BUILD_VERSION="0" KBUILD_HOSTLDFLAGS="-L/home/build/lede/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls= KBUILD_EXTRA_SYMBOLS="/home/build/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/symvers/cryptodev-linux.symvers /home/build/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/symvers/fullconenat.symvers /home/build/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/symvers/gpio-button-hotplug.symvers /home/build/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/symvers/qca-nss-clients-64.symvers /home/build/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/symvers/qca-nss-dp.symvers /home/build/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/symvers/qca-nss-drv-64.symvers /home/build/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/symvers/qca-nss-ecm-64.symvers /home/build/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/symvers/qca-ssdk.symvers /home/build/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/symvers/shortcut-fe-simulated-driver.symvers /home/build/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/symvers/shortcut-fe.symvers" KERNELRELEASE=5.10.135 EXTRA_CFLAGS="-I/home/build/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/backports-5.15.33-1/include -fmacro-prefix-map=/home/build/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/backports-5.15.33-1=backports-5.15.33-1 " KLIB_BUILD="/home/build/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.10.135" MODPROBE=true KLIB=/lib/modules/5.10.135 KERNEL_SUBLEVEL=10 KBUILD_LDFLAGS_MODULE_PREREQ= allnoconfig
make[4]: Entering directory '/home/build/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/backports-5.15.33-1'
make[5]: 'Kconfig.versions' is up to date.
make[6]: 'conf' is up to date.
boolean symbol CRYPTO_LIB_ARC4 tested for 'm'? test forced to 'n'
#
# configuration written to .config
#
make[4]: Leaving directory '/home/build/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/backports-5.15.33-1'
rm -rf /home/build/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/backports-5.15.33-1/modules
make  -C "/home/build/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/backports-5.15.33-1" KCFLAGS="-fmacro-prefix-map=/home/build/lede/build_dir/target-aarch64_cortex-a53_musl=target-aarch64_cortex-a53_musl" HOSTCFLAGS="-O2 -I/home/build/lede/staging_dir/host/include -I/home/build/lede/staging_dir/hostpkg/include -I/home/build/lede/staging_dir/target-aarch64_cortex-a53_musl/host/include -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="aarch64-openwrt-linux-musl-" ARCH="arm64" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Thu Aug 11 01:01:23 2022" KBUILD_BUILD_VERSION="0" KBUILD_HOSTLDFLAGS="-L/home/build/lede/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls= KBUILD_EXTRA_SYMBOLS="/home/build/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/symvers/cryptodev-linux.symvers /home/build/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/symvers/fullconenat.symvers /home/build/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/symvers/gpio-button-hotplug.symvers /home/build/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/symvers/qca-nss-clients-64.symvers /home/build/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/symvers/qca-nss-dp.symvers /home/build/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/symvers/qca-nss-drv-64.symvers /home/build/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/symvers/qca-nss-ecm-64.symvers /home/build/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/symvers/qca-ssdk.symvers /home/build/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/symvers/shortcut-fe-simulated-driver.symvers /home/build/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/symvers/shortcut-fe.symvers" KERNELRELEASE=5.10.135 EXTRA_CFLAGS="-I/home/build/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/backports-5.15.33-1/include -fmacro-prefix-map=/home/build/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/backports-5.15.33-1=backports-5.15.33-1 " KLIB_BUILD="/home/build/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.10.135" MODPROBE=true KLIB=/lib/modules/5.10.135 KERNEL_SUBLEVEL=10 KBUILD_LDFLAGS_MODULE_PREREQ= modules
make[4]: Entering directory '/home/build/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/backports-5.15.33-1'
make[5]: 'Kconfig.versions' is up to date.
make[7]: 'Kconfig.versions' is up to date.
make[8]: 'conf' is up to date.
boolean symbol CRYPTO_LIB_ARC4 tested for 'm'? test forced to 'n'
#
# configuration written to .config
#
Building backport-include/backport/autoconf.h ... done.
  CC [M]  /home/build/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/backports-5.15.33-1/net/mac80211/iface.o
/home/build/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/backports-5.15.33-1/net/mac80211/iface.c: In function 'ieee80211_do_open':
/home/build/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/backports-5.15.33-1/net/mac80211/iface.c:1277:3: error: implicit declaration of function 'eth_hw_addr_set'; did you mean 'eth_hw_addr_crc'? [-Werror=implicit-function-declaration]
   eth_hw_addr_set(dev, local->hw.wiphy->perm_addr);
   ^~~~~~~~~~~~~~~
   eth_hw_addr_crc
cc1: some warnings being treated as errors
make[9]: *** [scripts/Makefile.build:286: /home/build/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/backports-5.15.33-1/net/mac80211/iface.o] Error 1
make[8]: *** [scripts/Makefile.build:503: /home/build/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/backports-5.15.33-1/net/mac80211] Error 2
make[7]: *** [Makefile:1831: /home/build/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/backports-5.15.33-1] Error 2
make[6]: *** [Makefile.build:13: modules] Error 2
make[5]: *** [Makefile.real:93: modules] Error 2
make[4]: *** [Makefile:121: modules] Error 2
make[4]: Leaving directory '/home/build/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/backports-5.15.33-1'
make[3]: *** [Makefile:559: /home/build/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/backports-5.15.33-1/.built] Error 2
make[3]: Leaving directory '/home/build/lede/package/kernel/mac80211'
time: package/kernel/mac80211/regular/compile#1.23#0.35#1.53
    ERROR: package/kernel/mac80211 failed to build (build variant: regular).
make[2]: *** [package/Makefile:116: package/kernel/mac80211/compile] Error 1
make[2]: Leaving directory '/home/build/lede'
make[1]: *** [package/Makefile:110: /home/build/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/build/lede'
make: *** [/home/build/lede/include/toplevel.mk:230: world] Error 2
```

</details>

zhangguanzhang, 2022-08-11:
https://github.com/coolsnowwolf/lede/issues/9803

coolsnowwolf, 2022-08-11:
已修复

closed 2022-08-11 by 1054009064

1054009064, 2022-08-11:
#9917

yuzhe1129, 2022-08-11:
我靠，刚才我也是，前几天 编译成功的，刚编译出错了，