coolsnowwolf/lede #9919: 编译进程Terminated自动终止

Yvan86, 2022-08-11:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

ubuntu22.04 云编译，编译失败。报错信息显示很多进程终止，各位大佬帮看下

### (2) 路由器型号和固件版本

newifi d2

### (3) 详细日志

2022-08-11T22:34:33.5715149Z checking for mipsel-openwrt-linux-strip... (cached) mipsel-openwrt-linux-musl-strip
2022-08-11T22:34:33.5717132Z checking for mipsel-openwrt-linux-ranlib... mipsel-openwrt-linux-musl-gcc-ranlib
2022-08-11T22:34:33.8119411Z > gmake[4]: Entering directory '/workdir/lede/build_dir/target-mipsel_24kc_musl/qtbase-everywhere-src-5.15.4/config.tests/ipv6ifname'
2022-08-11T22:34:33.8122275Z > mipsel-openwrt-linux-musl-g++ -c -pipe -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -fmacro-prefix-map=/workdir/lede/build_dir/target-mipsel_24kc_musl/qtbase-everywhere-src-5.15.4=qtbase-everywhere-src-5.15.4 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DPIC -fpic -ffunction-sections -fdata-sections -flto -I/workdir/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/usr/include -I/workdir/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/include/fortify -I/workdir/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/include --sysroot=/workdir/lede/staging_dir/target-mipsel_24kc_musl --sysroot=/workdir/lede/staging_dir/target-mipsel_24kc_musl -O2 -w -fPIC  -I. -I/workdir/lede/build_dir/target-mipsel_24kc_musl/qtbase-everywhere-src-5.15.4/mkspecs/devices/linux-generic-g++ -o main.o main.cpp
2022-08-11T22:34:33.8124780Z > mipsel-openwrt-linux-musl-g++ -L/workdir/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/usr/lib -L/workdir/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/lib -znow -zrelro -Wl,--gc-sections,--as-needed --sysroot=/workdir/lede/staging_dir/target-mipsel_24kc_musl --sysroot=/workdir/lede/staging_dir/target-mipsel_24kc_musl -Wl,-O1 -o ipv6ifname main.o      
2022-08-11T22:34:33.8125775Z > gmake[4]: Leaving directory '/workdir/lede/build_dir/target-mipsel_24kc_musl/qtbase-everywhere-src-5.15.4/config.tests/ipv6ifname'
2022-08-11T22:34:33.8126355Z test config.qtbase_network.tests.ipv6ifname succeeded
2022-08-11T22:34:33.8126989Z Checking for Linux AF_NETLINK sockets... 
2022-08-11T22:34:33.8129148Z + cd /workdir/lede/build_dir/target-mipsel_24kc_musl/qtbase-everywhere-src-5.15.4/config.tests/linux-netlink && PKG_CONFIG_SYSROOT_DIR=/workdir/lede/staging_dir/target-mipsel_24kc_musl PKG_CONFIG_LIBDIR=/workdir/lede/staging_dir/target-mipsel_24kc_musl/usr/lib/pkgconfig:/workdir/lede/staging_dir/target-mipsel_24kc_musl/usr/share/pkgconfig /workdir/lede/build_dir/target-mipsel_24kc_musl/qtbase-everywhere-src-5.15.4/bin/qmake "CONFIG -= qt debug_and_release app_bundle lib_bundle" "CONFIG += shared warn_off console single_arch" "QMAKE_CFLAGS += --sysroot=/workdir/lede/staging_dir/target-mipsel_24kc_musl" "QMAKE_CXXFLAGS += --sysroot=/workdir/lede/staging_dir/target-mipsel_24kc_musl" "QMAKE_LFLAGS += --sysroot=/workdir/lede/staging_dir/target-mipsel_24kc_musl" -early "CONFIG += cross_compile" /workdir/lede/build_dir/target-mipsel_24kc_musl/qtbase-everywhere-src-5.15.4/config.tests/linux-netlink
2022-08-11T22:34:33.8587811Z checking command to parse mipsel-openwrt-linux-musl-gcc-nm output from mipsel-openwrt-linux-musl-gcc object... libtool: compile:  mipsel-openwrt-linux-musl-g++ -DHAVE_CONFIG_H -I. -I.. -I../lib -I../intl -I./includes -I./includes -DLOCALEDIR=\"/usr/share/locale\" -DHAVE_CONFIG_H -I/workdir/lede/staging_dir/target-mipsel_24kc_musl/usr/include -I/workdir/lede/staging_dir/target-mipsel_24kc_musl/usr/include -I../deps/wslay/lib/includes -I../deps/wslay/lib/includes -I/workdir/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/usr/include -I/workdir/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/include/fortify -I/workdir/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/include -I/workdir/lede/staging_dir/target-mipsel_24kc_musl/usr/lib/libiconv-stub/include -I/workdir/lede/staging_dir/target-mipsel_24kc_musl/usr/lib/libintl-stub/include -pipe -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -fmacro-prefix-map=/workdir/lede/build_dir/target-mipsel_24kc_musl/aria2-1.36.0=aria2-1.36.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/workdir/lede/staging_dir/target-mipsel_24kc_musl/usr/lib/libiconv-stub/include -I/workdir/lede/staging_dir/target-mipsel_24kc_musl/usr/lib/libintl-stub/include -ffunction-sections -fdata-sections -flto -MT RpcRequest.lo -MD -MP -MF .deps/RpcRequest.Tpo -c RpcRequest.cc  -fPIC -DPIC -o .libs/RpcRequest.o
2022-08-11T22:34:33.9076907Z + cd /workdir/lede/build_dir/target-mipsel_24kc_musl/qtbase-everywhere-src-5.15.4/config.tests/linux-netlink && MAKEFLAGS= /usr/bin/gmake
2022-08-11T22:34:34.1456407Z make[6]: *** wait: No child processes.  Stop.
2022-08-11T22:34:34.1479282Z ninja: build stopped: interrupted by user.
2022-08-11T22:34:34.1535197Z make[6]: *** Waiting for unfinished jobs....
2022-08-11T22:34:34.1535549Z make[6]: *** wait: No child processes.  Stop.
2022-08-11T22:34:34.1538023Z make[6]: *** Deleting file 'debugfs.o'
2022-08-11T22:34:34.1538575Z make[7]: *** Deleting file 'device.lo'
2022-08-11T22:34:34.1539176Z Terminated
2022-08-11T22:34:34.1539657Z make[5]: *** [Makefile:1621: all-recursive] Terminated
2022-08-11T22:34:34.1540344Z make[5]: *** [Makefile:558: all-recursive] Terminated
2022-08-11T22:34:34.1541134Z make[6]: *** [Makefile:1749: all-recursive] Terminated
2022-08-11T22:34:34.1541398Z make[4]: *** [Makefile:469: all] Terminated
2022-08-11T22:34:34.1541763Z make[5]: *** [Makefile:443: all-progs-recursive] Terminated
2022-08-11T22:34:34.1542032Z make[4]: *** [Makefile:371: all] Terminated
2022-08-11T22:34:34.1542478Z make[3]: *** [Makefile:107: /workdir/lede/build_dir/target-mipsel_24kc_musl/aria2-1.36.0/.built] Terminated
2022-08-11T22:34:34.1542832Z make[7]: *** [Makefile:2638: RpcRequest.lo] Terminated
2022-08-11T22:34:34.1543064Z Terminated
2022-08-11T22:34:34.1543534Z make[3]: *** [Makefile:269: /workdir/lede/build_dir/target-mipsel_24kc_musl/gnutls-3.7.6/.configured_e4b3ef16620d322ec8fba766d9028a1d] Terminated
2022-08-11T22:34:34.1544201Z make[3]: *** [Makefile:204: /workdir/lede/build_dir/target-mipsel_24kc_musl/qtbase-everywhere-src-5.15.4/.configured_05469b1aeae0f6f21907996995746f99] Terminated
2022-08-11T22:34:34.1544794Z make[3]: *** [Makefile:65: /workdir/lede/build_dir/target-mipsel_24kc_musl/rblibtorrent-1.2.16/.built] Error 2
2022-08-11T22:34:34.1545122Z make[4]: *** [Makefile:1543: all] Terminated
2022-08-11T22:34:34.1545597Z make[3]: *** [Makefile:80: /workdir/lede/build_dir/target-mipsel_24kc_musl/parted-3.4/.built] Terminated
2022-08-11T22:34:34.1546117Z make[3]: *** [Makefile:322: /workdir/lede/build_dir/target-mipsel_24kc_musl/e2fsprogs-1.46.5/.built] Terminated
2022-08-11T22:34:34.1546457Z make[7]: *** [Makefile:1727: device.lo] Terminated
2022-08-11T22:34:34.1547033Z make[3]: *** [Makefile:181: /workdir/lede/build_dir/target-mipsel_24kc_musl/transmission-openssl/transmission-2.94/.configured_68b329da9893e34099c7d8ad5cb9c940] Terminated
2022-08-11T22:34:34.1547483Z make[6]: *** [Makefile:391: debugfs.o] Terminated
2022-08-11T22:34:34.1547773Z make[7]: *** [Makefile:2638: RpcMethodImpl.lo] Terminated
2022-08-11T22:34:34.1548140Z make[2]: *** [package/Makefile:116: package/feeds/packages/aria2/compile] Terminated
2022-08-11T22:34:34.1548510Z make[2]: *** [package/Makefile:116: package/feeds/packages/qtbase/compile] Terminated
2022-08-11T22:34:34.1548893Z make[2]: *** [package/Makefile:116: package/feeds/packages/rblibtorrent/compile] Terminated
2022-08-11T22:34:34.1549273Z make[2]: *** [package/Makefile:116: package/feeds/packages/parted/compile] Terminated
2022-08-11T22:34:34.1549629Z make[2]: *** [package/Makefile:116: package/feeds/packages/gnutls/compile] Terminated
2022-08-11T22:34:34.1549980Z make[2]: *** [package/Makefile:116: package/utils/e2fsprogs/compile] Terminated
2022-08-11T22:34:34.1550351Z make[2]: *** [package/Makefile:116: package/feeds/packages/transmission/compile] Terminated
2022-08-11T22:34:34.1595090Z make: *** [/workdir/lede/include/toplevel.mk:230: world] Terminated
2022-08-11T22:34:34.1619424Z make[1]: *** [package/Makefile:110: /workdir/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Terminated
2022-08-11T22:34:34.8888775Z ##[error]The operation was canceled.

EnnawYang, 2022-08-12:
请勿使用ssh

Yvan86, 2022-08-12:
> 请勿使用ssh

ok,我试一下

coolsnowwolf, 2022-08-12:
内存太小了

zhangguanzhang, 2022-08-12:
你ssh到云端运行的命令在你退出ssh，例如你笔记本休眠断网了，这个ssh的会话的进程都会停止的。你可以用screen避免这种情况来执行一个很久才结束的命令

closed 2022-08-12 by aiamadeus