coolsnowwolf/lede #9921: R4S 最近编译的新版随机死机

BenjaminX, 2022-08-12:
怎么去抓到日志呢？略头痛

coolsnowwolf, 2022-08-12:
编译今天的版本

BenjaminX, 2022-08-12:
> 编译今天的版本

测试了一下最新的代码，不死机了。随机重启

coolsnowwolf, 2022-08-13:
对的

BenjaminX, 2022-08-13:
> 对的

换了TF一样重启

coolsnowwolf, 2022-08-13:
接上ttl 看看

BenjaminX, 2022-08-14:
> 接上ttl 看看

总算抓到异常了，看起来自己解决不了。求老大指导要一下

[ 6131.496625] Unable to handle kernel NULL pointer dereference at virtual address 0000000000000936
[ 6131.497426] Mem abort info:
[ 6131.497681]   ESR = 0x0000000096000006
[ 6131.498070]   EC = 0x25: DABT (current EL), IL = 32 bits
[ 6131.498551]   SET = 0, FnV = 0
[ 6131.498829]   EA = 0, S1PTW = 0
[ 6131.499114]   FSC = 0x06: level 2 translation fault
[ 6131.499552] Data abort info:
[ 6131.499813]   ISV = 0, ISS = 0x00000006
[ 6131.500158]   CM = 0, WnR = 0
[ 6131.500429] user pgtable: 4k pages, 48-bit VAs, pgdp=000000000c980000
[ 6131.501006] [0000000000000936] pgd=0800000009e5a003, p4d=0800000009e5a003, pud=080000000c414003, pmd=0000000000000000
[ 6131.501989] Internal error: Oops: 96000006 [#1] PREEMPT SMP
[ 6131.502492] Modules linked in: fast_classifier xt_FULLCONENAT pppoe ppp_async uvcvideo pppox ppp_mppe ppp_generic ipt_REJECT zstd xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_socket xt_recent xt_quota xt_pkttype xt_owner xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_esp xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_cgroup xt_addrtype xt_TPROXY xt_TCPMSS xt_REDIRECT xt_NETMAP xt_MASQUERADE xt_LOG xt_HL xt_DSCP xt_CT xt_CLASSIFY videobuf2_v4l2 ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda ts_fsm ts_bm tcp_bbr slhc r8168 nf_tproxy_ipv6 nf_tproxy_ipv4 nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp
[ 6131.503030]  nf_conntrack_broadcast ts_kmp nf_conntrack_amanda nf_conncount macvlan lzo iptable_raw iptable_nat iptable_mangle iptable_filter ipt_ah ipt_ECN ip_tables crc_ccitt asn1_decoder videobuf2_vmalloc videobuf2_memops videobuf2_common videodev cryptodev xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip6t_NPT ip6t_rt ip6t_mh ip6t_ipv6header ip6t_hbh ip6t_frag ip6t_eui64 ip6t_ah ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 bonding sit tunnel4 ip_tunnel tun vfat fat ntfs3 raid10 raid1 raid0 md_mod nls_utf8 nls_cp437 shortcut_fe_ipv6 shortcut_fe crypto_user algif_skcipher algif_rng algif_hash algif_aead af_alg sha1_generic seqiv

BenjaminX, 2022-08-14:
[image]

BenjaminX, 2022-08-14:
看起来像是
CONFIG_PACKAGE_kmod-fast-classifier
包引起的

BenjaminX, 2022-08-14:
关闭 Shortcut-FE flow offloading 选项，问题解决。重新打包踢出 shortcutfe包，问题解决。

closed 2022-08-14 by BenjaminX