coolsnowwolf/lede #9922: r66s的改动影响了r2s的编译失败

zhangguanzhang, 2022-08-12:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：
```
aarch64-openwrt-linux-musl-cpp: error: /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.203/arch/arm64/boot/dts/rockchip/rk3568-r66s.dts: No such file or directory
```

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

```
make[4]: Entering directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.203'
  CALL    scripts/checksyscalls.sh
  CALL    scripts/atomic/check-atomics.sh
  CHK     include/generated/compile.h
  Building modules, stage 2.
  MODPOST 390 modules
WARNING: "nf_ct_tcp_be_liberal" [net/netfilter/nf_conntrack] is a static EXPORT_SYMBOL_GPL
WARNING: "nf_ct_tcp_no_window_check" [net/netfilter/nf_conntrack] is a static EXPORT_SYMBOL_GPL
make[4]: Leaving directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.203'
cmp -s /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.203/vmlinux /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/vmlinux.debug || { aarch64-openwrt-linux-musl-objcopy -O binary -R .reginfo -R .notes -R .note -R .comment -R .mdebug -R .note.gnu.build-id -S /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.203/vmlinux /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/vmlinux; aarch64-openwrt-linux-musl-objcopy -R .reginfo -R .notes -R .note -R .comment -R .mdebug -R .note.gnu.build-id -S /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.203/vmlinux /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/vmlinux.elf; cp -fpR /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.203/vmlinux /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/vmlinux.debug;  cp -fpR /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.203/arch/arm64/boot//Image /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/Image;  }
touch /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.203/.image
make -C image compile install TARGET_BUILD=
make[4]: Entering directory '/workdir/openwrt/target/linux/rockchip/image'
cp -fpR /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.203/COPYING /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/COPYING.linux
aarch64-openwrt-linux-musl-cpp -nostdinc -x assembler-with-cpp -I/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.203/arch/arm64/boot/dts -I/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.203/arch/arm64/boot/dts/include -I/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.203/include/ -undef -D__DTS__  -o /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3328-doornet1.dtb.tmp /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.203/arch/arm64/boot/dts/rockchip/rk3328-doornet1.dts
/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.203/scripts/dtc/dtc -O dtb -i/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.203/arch/arm64/boot/dts/rockchip/ -Wno-unit_address_vs_reg -Wno-simple_bus_reg -Wno-unit_address_format -Wno-pci_bridge -Wno-pci_device_bus_num -Wno-pci_device_reg -Wno-avoid_unnecessary_addr_size -Wno-alias_paths -Wno-graph_child_address -Wno-graph_port -Wno-unique_unit_address   -o /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3328-doornet1.dtb /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3328-doornet1.dtb.tmp
/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.203/arch/arm64/boot/dts/rockchip/rk3328-doornet1.dts:488.3-13: Warning (reg_format): /usb@ff600000/dwc3@ff600000/usb-eth@2:reg: property has invalid length (4 bytes) (#address-cells == 2, #size-cells == 1)
/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3328-doornet1.dtb: Warning (i2c_bus_reg): Failed prerequisite 'reg_format'
/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3328-doornet1.dtb: Warning (spi_bus_reg): Failed prerequisite 'reg_format'
/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.203/arch/arm64/boot/dts/rockchip/rk3328-doornet1.dts:486.12-491.4: Warning (avoid_default_addr_size): /usb@ff600000/dwc3@ff600000/usb-eth@2: Relying on default #address-cells value
/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.203/arch/arm64/boot/dts/rockchip/rk3328-doornet1.dts:486.12-491.4: Warning (avoid_default_addr_size): /usb@ff600000/dwc3@ff600000/usb-eth@2: Relying on default #size-cells value
rm -f /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3328-doornet1.dtb.tmp
aarch64-openwrt-linux-musl-cpp -nostdinc -x assembler-with-cpp -I/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.203/arch/arm64/boot/dts -I/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.203/arch/arm64/boot/dts/include -I/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.203/include/ -undef -D__DTS__  -o /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3399-doornet2.dtb.tmp /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.203/arch/arm64/boot/dts/rockchip/rk3399-doornet2.dts
/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.203/scripts/dtc/dtc -O dtb -i/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.203/arch/arm64/boot/dts/rockchip/ -Wno-unit_address_vs_reg -Wno-simple_bus_reg -Wno-unit_address_format -Wno-pci_bridge -Wno-pci_device_bus_num -Wno-pci_device_reg -Wno-avoid_unnecessary_addr_size -Wno-alias_paths -Wno-graph_child_address -Wno-graph_port -Wno-unique_unit_address   -o /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3399-doornet2.dtb /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3399-doornet2.dtb.tmp
rm -f /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3399-doornet2.dtb.tmp
aarch64-openwrt-linux-musl-cpp -nostdinc -x assembler-with-cpp -I/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.203/arch/arm64/boot/dts -I/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.203/arch/arm64/boot/dts/include -I/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.203/include/ -undef -D__DTS__  -o /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3568-r66s.dtb.tmp /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.203/arch/arm64/boot/dts/rockchip/rk3568-r66s.dts
aarch64-openwrt-linux-musl-cpp: error: /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.203/arch/arm64/boot/dts/rockchip/rk3568-r66s.dts: No such file or directory
aarch64-openwrt-linux-musl-cpp: warning: '-x assembler-with-cpp' after last input file has no effect
aarch64-openwrt-linux-musl-cpp: fatal error: no input files
compilation terminated.
make[4]: *** [Makefile:78: /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3568-r66s.dtb] Error 1
make[4]: Leaving directory '/workdir/openwrt/target/linux/rockchip/image'
make[3]: *** [Makefile:25: install] Error 2
make[3]: Leaving directory '/workdir/openwrt/target/linux/rockchip'
make[2]: *** [Makefile:11: install] Error 2
make[2]: Leaving directory '/workdir/openwrt/target/linux'
time: target/linux/install#15.11#6.02#20.84
    ERROR: target/linux failed to build.
make[1]: *** [target/Makefile:25: target/linux/install] Error 1
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:230: target/install] Error 2
Error: Process completed with exit code 2.
```

DHDAXCW, 2022-08-12:
发下配置来？你用5.4就不要在config里面选r66s

zhangguanzhang, 2022-08-12:
我是构建的r2s和doornet2都这样，没勾选r66s，我再构建下看看，貌似10.38的提交修复了这个问题

DHDAXCW, 2022-08-12:
上午提交那个修复和你缺少dts没关系，你直接[注释33行](https://github.com/coolsnowwolf/lede/blob/master/target/linux/rockchip/image/armv8.mk#L33)就能编译过

zhangguanzhang, 2022-08-12:
我试下

zhangguanzhang, 2022-08-12:
@coolsnowwolf 大佬，我最新代码编译还是失败

zhangguanzhang, 2022-08-12:
Fixes by: https://github.com/coolsnowwolf/lede/commit/07c6d87fb007e23bfccf18ea58f14c3dfcd86cba

closed 2022-08-12 by zhangguanzhang

zhangguanzhang, 2022-08-12:
```
        sed -ri '/^\s*TARGET_DEVICES\s.+?(fastrhino_r66s|firefly_station-p2)/d' target/linux/rockchip/image/armv8.mk
```