coolsnowwolf/lede #9924: R2S/R4S无法编译

bi4tms, 2022-08-12:
拉取最新源码编译5.10内核的R2S、R4S均失败

日志如下：
Makefile:78: recipe for target '/home/r2s.510/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3568-station-p2.dtb' failed
make[5]: *** [/home/r2s.510/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3568-station-p2.dtb] Error 1
make[5]: Leaving directory '/home/r2s.510/lede/target/linux/rockchip/image'
Makefile:25: recipe for target 'install' failed
make[4]: *** [install] Error 2
make[4]: Leaving directory '/home/r2s.510/lede/target/linux/rockchip'
Makefile:11: recipe for target 'install' failed
make[3]: *** [install] Error 2
make[3]: Leaving directory '/home/r2s.510/lede/target/linux'
time: target/linux/install#9.78#46.90#59.67
    ERROR: target/linux failed to build.
target/Makefile:23: recipe for target 'target/linux/install' failed
make[2]: *** [target/linux/install] Error 1
make[2]: Leaving directory '/home/r2s.510/lede'
target/Makefile:19: recipe for target '/home/r2s.510/lede/staging_dir/target-aarch64_generic_musl/stamp/.target_install' failed
make[1]: *** [/home/r2s.510/lede/staging_dir/target-aarch64_generic_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/home/r2s.510/lede'
/home/r2s.510/lede/include/toplevel.mk:228: recipe for target 'world' failed
make: *** [world] Error 2

DHDAXCW, 2022-08-12:
注释r66s和stationp2就能过
修改target/linux/rockchip/image/armv8.mk两个前面加#，重新编译即可
```mk
#TARGET_DEVICES += fastrhino_r66s
#TARGET_DEVICES += firefly_station-p2
```

zhangguanzhang, 2022-08-12:
https://github.com/coolsnowwolf/lede/issues/9922
参照这个

closed 2022-08-12 by bi4tms