coolsnowwolf/lede #9928: 已拉取最新源码，卡在rk3568-r66s.dtb上

bi4tms, 2022-08-12:
刚刚拉取最新源码，编译R2S/R4S的5.10的固件仍然失败，日志如下：
Makefile:78: recipe for target '/home/r4s.510/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3568-r66s.dtb' failed
make[5]: *** [/home/r4s.510/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3568-r66s.dtb] Error 1
make[5]: Leaving directory '/home/r4s.510/lede/target/linux/rockchip/image'
Makefile:25: recipe for target 'install' failed
make[4]: *** [install] Error 2
make[4]: Leaving directory '/home/r4s.510/lede/target/linux/rockchip'
Makefile:11: recipe for target 'install' failed
make[3]: *** [install] Error 2
make[3]: Leaving directory '/home/r4s.510/lede/target/linux'
time: target/linux/install#9.43#41.67#53.44
    ERROR: target/linux failed to build.
target/Makefile:23: recipe for target 'target/linux/install' failed
make[2]: *** [target/linux/install] Error 1
make[2]: Leaving directory '/home/r4s.510/lede'
target/Makefile:19: recipe for target '/home/r4s.510/lede/staging_dir/target-aarch64_generic_musl/stamp/.target_install' failed
make[1]: *** [/home/r4s.510/lede/staging_dir/target-aarch64_generic_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/home/r4s.510/lede'
/home/r4s.510/lede/include/toplevel.mk:228: recipe for target 'world' failed
make: *** [world] Error 2

coolsnowwolf, 2022-08-13:
不建议用 5.10 内核了

bi4tms, 2022-08-13:
> 不建议用 5.10 内核了

好的，那就不再编译5.10内核了

closed 2022-08-13 by bi4tms