coolsnowwolf/lede #9934: 就想做个调查，8125B的同学你们使用有问题吗？有问题的说说自己网络环境

laaArmy, 2022-08-13:


hcym, 2022-08-13:
实体安装，没有问题，千兆以下

EnnawYang, 2022-08-14:
你是指这个问题吗，没注意到是怎么发生的，实体安装，千兆+2.5G
```
[306003.705286] ------------[ cut here ]------------
[306003.705663] NETDEV WATCHDOG: eth0 (r8125): transmit queue 0 timed out
[306003.706161] WARNING: CPU: 2 PID: 0 at dev_watchdog+0x29b/0x2a0
[306003.706587] Modules linked in: pppoe ppp_async rndis_host pppox ppp_mppe ppp_generic ipt_REJECT cdc_ether ax88179_178a asix zstd xt_time xt_tcpudp xt_state xt_socket xt_quota xt_pkttype xt_owner xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack xt_comment xt_cgroup xt_addrtype xt_TPROXY xt_TCPMSS xt_REDIRECT xt_MASQUERADE xt_LOG xt_CT via_velocity usbnet ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda tulip ts_fsm ts_bm slhc selftests rtl8150 r8168 r8152 r8125 pcnet32 nf_tproxy_ipv6 nf_tproxy_ipv4 nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_syslog nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast ts_kmp nf_conntrack_amanda mlx5_core mlx4_en mlx4_core macvlan lzo iptable_raw iptable_nat
[306003.706801]  iptable_mangle iptable_filter ip_tables igc iavf i915 i40e hso forcedeth e1000e crc_ccitt br_netfilter bnx2x bnx2 alx 8139too 8139cp fuse configs evdev i2c_dev cryptodev xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip6t_NPT ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 bonding ixgbe igbvf igb e1000 amd_xgbe mdio_devres dummy sit mdio tunnel4 ip_tunnel veth tun xfrm_user xfrm_ipcomp af_key xfrm_algo ntfs3 cifs cifs_md4 cifs_arc4 dns_resolver dm_mirror dm_region_hash dm_log dm_crypt dm_mod dax nls_utf8 vxlan udp_tunnel ip6_udp_tunnel ena crypto_user algif_skcipher algif_rng algif_hash algif_aead af_alg sha512_ssse3
[306003.712900]  sha512_generic sha256_ssse3 sha1_ssse3 sha1_generic seqiv jitterentropy_rng drbg md5 hmac echainiv des_generic libdes deflate cmac authenc arc4 crypto_acompress nls_iso8859_1 nls_cp437 vfat fat uas sdhci_pltfm xhci_plat_hcd fsl_mph_dr_of ehci_platform ehci_fsl xfs exfat btrfs zstd_decompress zstd_compress xxhash xor raid6_pq lzo_decompress lzo_compress libcrc32c button_hotplug ptp pps_core mii libphy cbc encrypted_keys trusted tpm oid_registry asn1_encoder asn1_decoder
[306003.721799] CPU: 2 PID: 0 Comm: swapper/2 Not tainted 5.15.58 #0
[306003.722234] Hardware name: Default string Default string/Default string, BIOS 1.010 09/27/2021
[306003.722842] RIP: 0010:dev_watchdog+0x29b/0x2a0
[306003.723170] Code: ff ff ff 4c 8b 7d d0 c6 05 45 3e b8 00 01 4c 89 ff e8 f9 37 fb ff 89 d9 4c 89 fe 48 c7 c7 a8 c5 06 82 48 89 c2 e8 16 2b 15 00 <0f> 0b eb b3 90 55 48 89 e5 41 57 49 89 d7 41 56 4d 89 c6 41 55 49
[306003.724458] RSP: 0018:ffffc90000128e88 EFLAGS: 00010292
[306003.724832] RAX: 0000000000000039 RBX: 0000000000000000 RCX: 0000000000000027
[306003.725344] RDX: 0000000000000027 RSI: ffffc90000128ce0 RDI: ffff888277d1c588
[306003.725859] RBP: ffffc90000128ec0 R08: 0000000000000000 R09: c0000000ffffefff
[306003.726364] R10: 0000000000000001 R11: ffffc90000128c88 R12: ffff888107ed0c40
[306003.726862] R13: ffff88810a5e0478 R14: 0000000000000004 R15: ffff88810a5e0000
[306003.727369] FS:  0000000000000000(0000) GS:ffff888277d00000(0000) knlGS:0000000000000000
[306003.727952] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[306003.728371] CR2: 00007fe06b64aafc CR3: 000000000320a000 CR4: 0000000000350ee0
[306003.728873] Call Trace:
[306003.729072]  <IRQ>
[306003.729236]  ? pfifo_fast_enqueue+0x150/0x150
[306003.729561]  call_timer_fn.isra.39+0x13/0x80
[306003.729884]  run_timer_softirq+0x2a3/0x2c0
[306003.730190]  ? clockevents_program_event+0x8c/0xf0
[306003.730541]  ? tick_program_event+0x3f/0x70
[306003.730854]  __do_softirq+0xc6/0x209
[306003.731129]  irq_exit_rcu+0x52/0x70
[306003.731394]  sysvec_apic_timer_interrupt+0x74/0xa0
[306003.731751]  </IRQ>
[306003.731914]  <TASK>
[306003.732075]  asm_sysvec_apic_timer_interrupt+0x1a/0x20
[306003.732442] RIP: 0010:cpuidle_enter_state+0xb7/0x2e0
[306003.732807] Code: c3 a2 ff 65 8b 3d 81 71 95 7e e8 44 c1 a2 ff 31 ff 48 89 c3 e8 2a cd a2 ff 80 7d d7 00 0f 85 9e 01 00 00 fb 66 0f 1f 44 00 00 <45> 85 ff 0f 88 c8 00 00 00 49 63 d7 48 2b 5d c8 48 8d 04 52 48 8d
[306003.734115] RSP: 0018:ffffc900000bfe78 EFLAGS: 00000246
[306003.734493] RAX: ffff888277d21c80 RBX: 0001164f0a266532 RCX: 000000000000001f
[306003.735005] RDX: 0001164f0a266532 RSI: 00000000401a41a4 RDI: 0000000000000000
[306003.735521] RBP: ffffc900000bfeb0 R08: 0000000000000000 R09: 00000000000214c0
[306003.736030] R10: 00022bbe25060212 R11: 000000000000000f R12: ffff8881010ed000
[306003.736530] R13: 0000000000000003 R14: ffffffff822d0580 R15: 0000000000000003
[306003.737046]  ? cpuidle_enter_state+0xa6/0x2e0
[306003.737371]  cpuidle_enter+0x2f/0x50
[306003.737644]  do_idle+0xfb/0x1e0
[306003.737888]  cpu_startup_entry+0x18/0x20
[306003.738178]  start_secondary+0xf9/0x100
[306003.738464]  secondary_startup_64_no_verify+0xb0/0xbb
[306003.738834]  </TASK>
[306003.739005] ---[ end trace aecebc09cc768035 ]---
```

csrutil, 2022-08-19:
官方OpenWrt，自己编译的驱动，千兆宽带。

[image]

[image]

helianling, 2022-09-14:
实体跑OP，没有问题，连2.5G和千兆设备都正常

coolsnowwolf, 2026-09-05:
这是一个社区调查帖，没有具体故障、设备版本或可验证日志；不属于可修复的 LEDE issue，关闭为“不计划处理”。

closed 2026-09-05 by coolsnowwolf