coolsnowwolf/lede #9941: R2S 内核版本从5.18开始到5.19 带宽跑不满

SilentSing, 2022-08-14:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述
R2S 内核版本从5.18开始到5.19 带宽跑不满
### (1) 具体问题

A：通过二次编译的固件版本升到5.19.1 千兆带宽只能跑到700左右

### (2) 路由器型号和固件版本

A：R2S  5.18-5.19.1

### (3) 详细日志

A：

DHDAXCW, 2022-08-14:
配置打开`CONFIG_PACKAGE_kmod-usb-dwc2=y`看看有没有效果？

coolsnowwolf, 2022-08-14:
因为加了 oaf 了，无法走加速

SilentSing, 2022-08-15:
> 因为加了 oaf 了，无法走加速

那怎么弄才能行呢 还是说新的固件没办法走加速    这个oaf是app过滤吗?

SilentSing, 2022-08-15:
> 因为加了 oaf 了，无法走加速
我把过滤的插件删除了 重新编译以后还是不行  编译出来的固件之前还能跑700左右 删除以后只能跑300左右了

BenjaminX, 2022-08-15:
也是 CCC 加速的 shortcut-FE 导致的，观察一下这。

我这里加上会随机重启，关闭了也出现速度慢很多的问题。  R4S 硬件

SilentSing, 2022-08-15:
> 也是 CCC 加速的 shortcut-FE 导致的，观察一下这。
> 
> 我这里加上会随机重启，关闭了也出现速度慢很多的问题。 R4S 硬件
我R2S 开启加速不会随机重启   但关闭以后速度慢应该是正常的吧 本来就是加速包转发性能的

czw1259, 2022-08-15:
我也是从5.15更新到5.18 5.19 发现跑不满带宽！L大佬能否帮忙测试一下是啥原因啊？

DHDAXCW, 2022-08-15:
看样子patchs-5.19里面少了一个dmc补丁吧...

SilentSing, 2022-08-15:
> 看样子patchs-5.19里面少了一个dmc补丁吧...

那要怎么弄才能把这个问题解决呢 骷髅头大佬

DHDAXCW, 2022-08-15:
> > 看样子patchs-5.19里面少了一个dmc补丁吧...
> 
> 那要怎么弄才能把这个问题解决呢 骷髅头大佬

明天我验证下再pr吧

DHDAXCW, 2022-08-16:
> > 看样子patchs-5.19里面少了一个dmc补丁吧...
> 
> 那要怎么弄才能把这个问题解决呢 骷髅头大佬

帮忙执行一下 cat /proc/interrupts  复制发这里来

SilentSing, 2022-08-16:
> > > 看样子patchs-5.19里面少了一个dmc补丁吧...
> > 
> > 
> > 那要怎么弄才能把这个问题解决呢 骷髅头大佬
> 
> 帮忙执行一下 cat /proc/interrupts 复制发这里来

           CPU0       CPU1       CPU2       CPU3
 11:      46065      24805      26494      28902     GICv2  30 Level     arch_ti                                                                                                                                                             mer
 18:          0          0          0          0     GICv2  94 Level     rockchi                                                                                                                                                             p_usb2phy
 19:          0          0          0          0     GICv2  32 Level     ff1f000                                                                                                                                                             0.dma-controller
 20:          0          0          0          0     GICv2  33 Level     ff1f000                                                                                                                                                             0.dma-controller
 21:          6          0          0          0     GICv2  89 Level     ttyS2
 22:          0          0          0          0     GICv2  43 Level     ff35080                                                                                                                                                             0.iommu
 23:          0          0          0          0     GICv2 106 Level     ff36048                                                                                                                                                             0.iommu
 24:      84067          0          0          0     GICv2  99 Level     xhci-hc                                                                                                                                                             d:usb1
 25:          0          0          0          0     GICv2  48 Level     ehci_hc                                                                                                                                                             d:usb3
 26:          0          0          0          0     GICv2  49 Level     ohci_hc                                                                                                                                                             d:usb2
 27:       1614          0          0          0     GICv2  69 Level     ff16000                                                                                                                                                             0.i2c
 28:          0          0          0          0  rockchip_gpio_irq  24 Level                                                                                                                                                                  rk805
 29:          0          0          0          0     rk805   0 Edge      rk805_p                                                                                                                                                             wrkey_fall
 34:          0          0          0          0     rk805   5 Edge      RTC ala                                                                                                                                                             rm
 36:          0          0          0          0     rk805   7 Edge      rk805_p                                                                                                                                                             wrkey_rise
 37:          0          0          0          0     GICv2  90 Level     rockchi                                                                                                                                                             p_thermal
 38:      34500          0          0          0     GICv2  56 Level     eth0
 39:       9101          0          0          0     GICv2  44 Level     dw-mci
 40:          0          0          0          0  rockchip_gpio_irq   0 Edge                                                                                                                                                                   keys
 41:          0          0          0          0     GICv2  55 Level     ff58000                                                                                                                                                             0.usb, dwc2_hsotg:usb5
IPI0:     21060      28945      24283      26548       Rescheduling interrupts
IPI1:      3762       2483       1587       1099       Function call interrupts
IPI2:         0          0          0          0       CPU stop interrupts
IPI3:         0          0          0          0       CPU stop (for crash dump)                                                                                                                                                              interrupts
IPI4:      4498       7115       6535       8197       Timer broadcast interrupt                                                                                                                                                             s
IPI5:        93        140         98        124       IRQ work interrupts
IPI6:         0          0          0          0       CPU wake-up interrupts

SilentSing, 2022-08-16:
> > > 看外观patchs-5.19里面少了一个dmc吧...
> > 
> > 
> > 那要怎么弄才能解决这个问题呢？骷髅头大佬
> 
> 帮忙执行一下 cat /proc/interrupts 复制发这里来
[image]

DHDAXCW, 2022-08-16:
执行一下
```bash
echo 2 > /proc/irq/38/smp_affinity
echo 4 > /proc/irq/24/smp_affinity
```
然后跑下宽带

SilentSing, 2022-08-16:
> 执行一下
> 
> ```shell
> echo 2 > /proc/irq/38/smp_affinity
> echo 4 > /proc/irq/24/smp_affinity
> ```
> 
> 然后跑下宽带

执行了以后带宽从300左右能跑到700+  但是还是不能达到满速  我是千兆的网  之前5.15的固件可以跑到940左
[image]
右

DHDAXCW, 2022-08-16:
那估计是这个了、、、、  [image]

SilentSing, 2022-08-16:
> 

> 那估计是这个了、、、、 [image]

L大说的这个oaf是APP过滤吗   能屏蔽或者不用吗   还是说内核就是这样已经没办法走加速了?  还是只能等L大优化了?

SilentSing, 2022-08-16:
> > 
> 
> > 那估计是这个了、、、、 [image]
> 
> L大说的这个oaf是APP过滤吗 能屏蔽或者不用吗 还是说内核就是这样已经没办法走加速了? 还是只能等L大优化了?

而且这个内核版本 smartDNS也有问题  能返回2个或以上地址 好像没有起到作用

czw1259, 2022-08-16:
> 那估计是这个了、、、、 [image]

也就是说从5.18以后的版本R2S就不支持优化加速了？是吗？

yimozi, 2022-08-21:
同样的问题，还以为是路由器坏了。已经退回5.15内核了。

closed 2022-09-14 by aiamadeus