coolsnowwolf/lede #9942: 更换5.19内核刷入后CPU频率一直满速

Rakau, 2022-08-14:
## 1.关于你要提交的问题

RT，请问如何解决？

Q：是否搜索了issue (使用 "x" 选择)
* [ X] 没有类似的issue

DHDAXCW, 2022-08-14:
[image]
不知道你说的是哪个 设备型号和log日志都没有，算不出来命

Rakau, 2022-08-14:
BROUNION R86S/R86S - Intel(R) Celeron(R) N5105 @ 2.00GHz : 4 Core 4 Thread

> [image] 不知道你说的是哪个 设备型号和log日志都没有，算不出来命

Rakau, 2022-08-14:
[    0.000000] Linux version 5.19.1 (debian@debian) (x86_64-openwrt-linux-musl-gcc (OpenWrt GCC 8.4.0 r4790-f861ce8a6) 8.4.0, GNU ld (GNU Binutils) 2.34) #0 SMP PREEMPT_DYNAMIC Sun Aug 14 08:37:26 2022
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz root=PARTUUID=a1e64fb5-020c-a300-d615-62d168fa1302 rootwait console=tty0 console=ttyS0,115200n8 noinitrd
[    0.000000] x86/split lock detection: #AC: crashing the kernel on kernel split_locks and warning on user-space split_locks
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Enabled xstate features 0x3, context size is 576 bytes, using 'compacted' format.
[    0.000000] signal: max sigframe size: 1104
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009efff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009f000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x0000000071f5bfff] usable
[    0.000000] BIOS-e820: [mem 0x0000000071f5c000-0x000000007445bfff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007445c000-0x00000000746dbfff] ACPI data
[    0.000000] BIOS-e820: [mem 0x00000000746dc000-0x00000000747dbfff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000747dc000-0x0000000074dfefff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000074dff000-0x0000000074efefff] type 20
[    0.000000] BIOS-e820: [mem 0x0000000074eff000-0x0000000074efffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000074f00000-0x000000007fbfffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000c0000000-0x00000000cfffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fe000000-0x00000000fe010fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed00000-0x00000000fed00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed20000-0x00000000fed7ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x00000002803fffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] efi: EFI v2.70 by American Megatrends
[    0.000000] efi: ACPI=0x74786000 ACPI 2.0=0x74786014 TPMFinalLog=0x74789000 SMBIOS=0x74c7e000 SMBIOS 3.0=0x74c7d000 MEMATTR=0x6e618018 ESRT=0x70e09118 
[    0.000000] SMBIOS 3.3.0 present.
[    0.000000] DMI: BROUNION R86S/R86S, BIOS 5.19 06/19/2022
[    0.000000] tsc: Detected 2000.000 MHz processor
[    0.000000] tsc: Detected 1996.800 MHz TSC
[    0.000310] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000313] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000321] last_pfn = 0x280400 max_arch_pfn = 0x400000000
[    0.000370] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.000584] last_pfn = 0x74f00 max_arch_pfn = 0x400000000
[    0.009070] esrt: Reserving ESRT space from 0x0000000070e09118 to 0x0000000070e09178.
[    0.009075] e820: update [mem 0x70e09000-0x70e09fff] usable ==> reserved
[    0.009608] Secure boot disabled
[    0.009610] ACPI: Early table checksum verification disabled
[    0.009613] ACPI: RSDP 0x0000000074786014 000024 (v02 ALASKA)
[    0.009617] ACPI: XSDT 0x0000000074785728 0000DC (v01 ALASKA A M I    01072009 AMI  01000013)
[    0.009622] ACPI: FACP 0x00000000746D8000 000114 (v06 ALASKA A M I    01072009 AMI  01000013)
[    0.009627] ACPI: DSDT 0x0000000074692000 045F74 (v02 ALASKA A M I    01072009 INTL 20191018)
[    0.009630] ACPI: FACS 0x00000000747DA000 000040
[    0.009632] ACPI: MCFG 0x00000000746D9000 00003C (v01 ALASKA A M I    01072009 MSFT 00000097)
[    0.009635] ACPI: FIDT 0x0000000074691000 00009C (v01 ALASKA A M I    01072009 AMI  00010013)
[    0.009638] ACPI: SSDT 0x000000007468E000 0025EE (v02 CpuRef CpuSsdt  00003000 INTL 20191018)
[    0.009640] ACPI: SSDT 0x00000000746DA000 001CCD (v02 SaSsdt SaSsdt   00003000 INTL 20191018)
[    0.009643] ACPI: SSDT 0x000000007468A000 0032CD (v02 INTEL  IgfxSsdt 00003000 INTL 20191018)
[    0.009645] ACPI: HPET 0x0000000074689000 000038 (v01 ALASKA A M I    01072009 AMI  01000013)
[    0.009648] ACPI: APIC 0x0000000074688000 00012C (v05 ALASKA A M I    01072009 AMI  01000013)
[    0.009650] ACPI: PRAM 0x0000000074687000 000030 (v01 ALASKA A M I    01072009 AMI  01000013)
[    0.009653] ACPI: SSDT 0x0000000074686000 000EF4 (v02 ALASKA Ther_Rvp 00001000 INTL 20191018)
[    0.009655] ACPI: SSDT 0x0000000074683000 002FC0 (v02 INTEL  xh_jslv2 00000000 INTL 20191018)
[    0.009658] ACPI: NHLT 0x0000000074681000 001771 (v00 ALASKA A M I    01072009 AMI  01000013)
[    0.009660] ACPI: LPIT 0x0000000074680000 000094 (v01 ALASKA A M I    01072009 AMI  01000013)
[    0.009662] ACPI: SSDT 0x000000007467C000 002720 (v02 ALASKA PtidDevc 00001000 INTL 20191018)
[    0.009665] ACPI: SSDT 0x000000007467B000 000122 (v02 ALASKA TbtTypeC 00000000 INTL 20191018)
[    0.009667] ACPI: DBGP 0x000000007467A000 000034 (v01 ALASKA A M I    01072009 AMI  01000013)
[    0.009670] ACPI: DBG2 0x0000000074679000 000054 (v00 ALASKA A M I    01072009 AMI  01000013)
[    0.009672] ACPI: SSDT 0x0000000074677000 0015C6 (v02 ALASKA UsbCTabl 00001000 INTL 20191018)
[    0.009675] ACPI: DMAR 0x0000000074676000 000088 (v02 INTEL  EDK2     00000002      01000013)
[... cut ...]
[   51.704130] IPv6: ADDRCONF(NETDEV_CHANGE): eth3: link becomes ready
[   52.154324] pppoe-wan: renamed from ppp0
[   55.114276] igc 0000:02:00.0 eth1: NIC Link is Down
[   55.224178] br-lan: port 1(eth1) entered disabled state
[   61.634571] igc 0000:02:00.0 eth1: NIC Link is Up 2500 Mbps Full Duplex, Flow Control: RX/TX
[   61.644138] br-lan: port 1(eth1) entered blocking state
[   61.649977] br-lan: port 1(eth1) entered forwarding state
[   74.845859] blk_print_req_error: 4 callbacks suppressed

DHDAXCW, 2022-08-14:
看着没啥问题

Rakau, 2022-08-14:
> 看着没啥问题

虽然负载是正常，但是频率一直是满速

WYC-2020, 2022-08-16:
这是又想马儿跑得快，又不让马儿吃草哈哈，想要自动升降频，满频能耗多少度电 哈哈

Rakau, 2022-08-16:
> 这是又想马儿跑得快，又不让马儿吃草哈哈，想要自动升降频，满频能耗多少度电 哈哈

不懂就滚，人云亦云

HugoPresents, 2022-08-18:
我也是 N5105, 物理机刷过几个版本，也有锁频 2GHz 的情况，目前就 esir 那个固件可以睿频也可以降频，不知道是什么原因

closed 2022-09-14 by aiamadeus

fishg, 2022-09-29:
根据在ubuntu上的经验（n5105 hyper-v 跑的ubuntu 20.04.5 内核5.19.12），是intel_pstate问题，改用ACPI CPUfreq 是可以睿频的。

fishg, 2022-10-03:
hyper-v 可以看看这个： https://learn.microsoft.com/zh-cn/windows-server/administration/performance-tuning/hardware/power/power-performance-tuning
可以试试
Powercfg -setacvalueindex scheme_current sub_processor PERFBOOSTMODE 4

gameopenwrt, 2023-05-10:
> 
人间各有需求，我跑网心云的，而且电费是免费的。
就想找跑满CPU速率的固件。一点都是不想让CPU闲缓下来。