coolsnowwolf/lede #9945: 关于在编译时出现  go download xxx.mod dial up tcp: 207.xxx.xxx.xxx i/o time out 错误的解决方法

Kalansas, 2022-08-15:
部分 G 友包括我在内编译固件的时经常会遇到 go download 失败导致编译无法成功，

经过资料查找我发现该问题的原因是防火长城 BAN 掉了 Go Mod 的下载端口。

### 解决方法

如果你的电脑是 MacOS 或 Linux，
执行
`$ go env -w GO111MODULE=on
$ go env -w GOPROXY=https://goproxy.cn,direct`
或
`$ echo "export GO111MODULE=on" >> ~/.profile
$ echo "export GOPROXY=https://goproxy.cn" >> ~/.profile
$ source ~/.profile`
提示找不到命令就安装 golang-go。

如果你用的是 WSL，那么再打开 PowerShell 并执行
`C:\> $env:GO111MODULE = "on"
C:\> $env:GOPROXY = "https://goproxy.cn"`

之后你就可以享受蹿……啊不，一泻千里般的下载速度了。

P.S：该方案来自 [Goproxy.cn](https://goproxy.cn/)，完全免费且无任何带宽速率限制，请去他们的页面支持他们！

zhangguanzhang, 2022-08-15:
这种就不要提issue了。。。
```
export GOPROXY=https://goproxy.cn,https://mirrors.aliyun.com/goproxy/,https://goproxy.io,direct
```

coolsnowwolf, 2026-09-05:
这是 Go 模块下载受网络环境影响时设置 GOPROXY 的经验说明，不是 LEDE 源码或包的缺陷；当前构建系统允许通过环境变量配置代理，关闭为“不计划处理”。

closed 2026-09-05 by coolsnowwolf