coolsnowwolf/lede #9947: 最新源码

aaaol, 2022-08-15:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
重新拉取过代码一样是这个问题 各种google百度都没相关的资料
### (1) 具体问题
stamp 编译报错
A：

### (2) 路由器型号和固件版本

A：
`X86`
### (3) 详细日志

A：
`Applying ./patches-pip/001-pep517-pyc-fix.patch using plaintext: 
patching file pip/_vendor/pep517/in_process/__init__.py
touch /home/moe/lede/staging_dir/hostpkg/lib/python3.9/site-packages/.pip_installed_22.0.4-1
mkdir -p /home/moe/lede/staging_dir/hostpkg/stamp
touch /home/moe/lede/build_dir/hostpkg/Python-3.9.12/.built
touch /home/moe/lede/staging_dir/hostpkg/stamp/.python3_installed
make[3]: Leaving directory '/home/moe/lede/feeds/packages/lang/python/python3'
time: package/feeds/packages/python3/host-compile#179.05#19.45#1007.19
make[2]: Leaving directory '/home/moe/lede'
make[1]: *** [package/Makefile:110: /home/moe/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/moe/lede'
make: *** [/home/moe/lede/include/toplevel.mk:230：world] 错误 2
`

WYC-2020, 2022-08-16:
单线程日志

coolsnowwolf, 2026-09-05:
仅提供了 make 末尾的通用 Error 2，真正失败点和完整编译命令均缺失；日志中 Python3 host 编译已完成，无法验证具体源码问题。请重新提供从第一个错误开始的完整日志，现关闭为“不计划处理”。

closed 2026-09-05 by coolsnowwolf