coolsnowwolf/lede #9948: 最新源码编译报错 mt76 compile Error

keeno1982, 2022-08-15:
机型 R4S

报错日志如下：
make[3]: Entering directory '/home/dietpi/lede/package/kernel/mt76'
mkdir -p /home/dietpi/lede/dl
SHELL= flock /home/dietpi/lede/tmp/.mt76-2022-07-03-93e3fce9.tar.xz.flock -c '          /home/dietpi/lede/scripts/download.pl "/home/dietpi/lede/dl" "mt76-2022-07-03-93e3fce9.tar.xz" "ad966e7aa81301f0601ceb7aae2ecb471c452f5d8366f701add807152fd08971" "" || (  /home/dietpi/lede/scripts/dl_github_archive.py --dl-dir="/home/dietpi/lede/dl" --url="https://github.com/openwrt/mt76" --version="93e3fce916c62d06892d41bf00d0f4c2926c9a0b" --subdir="mt76-2022-07-03-93e3fce9" --source="mt76-2022-07-03-93e3fce9.tar.xz" --hash="ad966e7aa81301f0601ceb7aae2ecb471c452f5d8366f701add807152fd08971" || (      echo "Checking out files from the git repository..."; mkdir -p /home/dietpi/lede/tmp/dl && cd /home/dietpi/lede/tmp/dl && rm -rf mt76-2022-07-03-93e3fce9 && [ \! -d mt76-2022-07-03-93e3fce9 ] && git clone  https://github.com/openwrt/mt76 mt76-2022-07-03-93e3fce9 && (cd mt76-2022-07-03-93e3fce9 && git checkout 93e3fce916c62d06892d41bf00d0f4c2926c9a0b && git submodule update --init --recursive) && echo "Packing checkout..." && export TAR_TIMESTAMP=`cd mt76-2022-07-03-93e3fce9 && git log -1 --format='\''@%ct'\''` && rm -rf mt76-2022-07-03-93e3fce9/.git &&         tar --numeric-owner --owner=0 --group=0 --mode=a-s --sort=name ${TAR_TIMESTAMP:+--mtime="$TAR_TIMESTAMP"} -c mt76-2022-07-03-93e3fce9 |      xz -zc -7e > /home/dietpi/lede/tmp/dl/mt76-2022-07-03-93e3fce9.tar.xz && mv /home/dietpi/lede/tmp/dl/mt76-2022-07-03-93e3fce9.tar.xz /home/dietpi/lede/dl/ && rm -rf mt76-2022-07-03-93e3fce9; );  )    '
rm -f /home/dietpi/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/mt76-2022-07-03-93e3fce9/.configured_*
rm -f /home/dietpi/lede/staging_dir/target-aarch64_generic_musl/stamp/.mt76_installed
mkdir -p /home/dietpi/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/mt76-2022-07-03-93e3fce9/tools
(cd /home/dietpi/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/mt76-2022-07-03-93e3fce9/tools; CFLAGS="-Os -pipe -mcpu=generic -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/dietpi/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/mt76-2022-07-03-93e3fce9=mt76-2022-07-03-93e3fce9 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/home/dietpi/lede/staging_dir/target-aarch64_generic_musl/usr/include/libnl-tiny " CXXFLAGS="-Os -pipe -mcpu=generic -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/dietpi/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/mt76-2022-07-03-93e3fce9=mt76-2022-07-03-93e3fce9 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/home/dietpi/lede/staging_dir/target-aarch64_generic_musl/usr/include/libnl-tiny " LDFLAGS="-L/home/dietpi/lede/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/usr/lib -L/home/dietpi/lede/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/lib -znow -zrelro " cmake -DCMAKE_SYSTEM_NAME=Linux -DCMAKE_SYSTEM_VERSION=1 -DCMAKE_SYSTEM_PROCESSOR=aarch64 -DCMAKE_BUILD_TYPE=Release -DCMAKE_C_FLAGS_RELEASE="-DNDEBUG" -DCMAKE_CXX_FLAGS_RELEASE="-DNDEBUG" -DCMAKE_C_COMPILER_LAUNCHER="" -DCMAKE_C_COMPILER="/home/dietpi/lede/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/bin/aarch64-openwrt-linux-musl-gcc" -DCMAKE_CXX_COMPILER_LAUNCHER="" -DCMAKE_CXX_COMPILER="/home/dietpi/lede/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/bin/aarch64-openwrt-linux-musl-g++" -DCMAKE_ASM_COMPILER_LAUNCHER="" -DCMAKE_ASM_COMPILER="/home/dietpi/lede/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/bin/aarch64-openwrt-linux-musl-gcc" -DCMAKE_EXE_LINKER_FLAGS:STRING="-L/home/dietpi/lede/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/usr/lib -L/home/dietpi/lede/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/lib -znow -zrelro" -DCMAKE_MODULE_LINKER_FLAGS:STRING="-L/home/dietpi/lede/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/usr/lib -L/home/dietpi/lede/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/lib -znow -zrelro -Wl,-Bsymbolic-functions" -DCMAKE_SHARED_LINKER_FLAGS:STRING="-L/home/dietpi/lede/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/usr/lib -L/home/dietpi/lede/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/lib -znow -zrelro -Wl,-Bsymbolic-functions" -DCMAKE_AR="/home/dietpi/lede/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/bin/aarch64-openwrt-linux-musl-gcc-ar" -DCMAKE_NM="/home/dietpi/lede/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/bin/aarch64-openwrt-linux-musl-gcc-nm" -DCMAKE_RANLIB="/home/dietpi/lede/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/bin/aarch64-openwrt-linux-musl-gcc-ranlib" -DCMAKE_FIND_ROOT_PATH="/home/dietpi/lede/staging_dir/target-aarch64_generic_musl/usr;/home/dietpi/lede/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl" -DCMAKE_FIND_ROOT_PATH_MODE_PROGRAM=BOTH -DCMAKE_FIND_ROOT_PATH_MODE_LIBRARY=ONLY -DCMAKE_FIND_ROOT_PATH_MODE_INCLUDE=ONLY -DCMAKE_STRIP=: -DCMAKE_INSTALL_PREFIX=/usr -DDL_LIBRARY=/home/dietpi/lede/staging_dir/target-aarch64_generic_musl -DCMAKE_PREFIX_PATH=/home/dietpi/lede/staging_dir/target-aarch64_generic_musl -DCMAKE_SKIP_RPATH=TRUE -DCMAKE_EXPORT_PACKAGE_REGISTRY=FALSE -DCMAKE_EXPORT_NO_PACKAGE_REGISTRY=TRUE -DCMAKE_FIND_USE_PACKAGE_REGISTRY=FALSE -DCMAKE_FIND_PACKAGE_NO_PACKAGE_REGISTRY=TRUE -DCMAKE_FIND_USE_SYSTEM_PACKAGE_REGISTRY=FALSE -DCMAKE_FIND_PACKAGE_NO_SYSTEM_PACKAGE_REGISTRY=TRUE  /home/dietpi/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/mt76-2022-07-03-93e3fce9/tools )
CMake Deprecation Warning at CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Configuring done
-- Generating done
-- Build files have been written to: /home/dietpi/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/mt76-2022-07-03-93e3fce9/tools
touch /home/dietpi/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/mt76-2022-07-03-93e3fce9/.configured_5cc66a6e3942ed0e7ae9eb0368a09863
rm -f /home/dietpi/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/mt76-2022-07-03-93e3fce9/.built
touch /home/dietpi/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/mt76-2022-07-03-93e3fce9/.built_check
make --jobserver-auth=3,4  -C "/home/dietpi/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.19.1" KCFLAGS="-fmacro-prefix-map=/home/dietpi/lede/build_dir/target-aarch64_generic_musl=target-aarch64_generic_musl" HOSTCFLAGS="-O2 -I/home/dietpi/lede/staging_dir/host/include -I/home/dietpi/lede/staging_dir/hostpkg/include -I/home/dietpi/lede/staging_dir/target-aarch64_generic_musl/host/include -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="aarch64-openwrt-linux-musl-" ARCH="arm64" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Sun Aug 14 18:36:59 2022" KBUILD_BUILD_VERSION="0" KBUILD_HOSTLDFLAGS="-L/home/dietpi/lede/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls= KBUILD_EXTRA_SYMBOLS="/home/dietpi/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/symvers/cryptodev-linux.symvers /home/dietpi/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/symvers/fast-classifier.symvers /home/dietpi/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/symvers/fullconenat.symvers /home/dietpi/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/symvers/gpio-button-hotplug.symvers /home/dietpi/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/symvers/mac80211.symvers /home/dietpi/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/symvers/r8168.symvers /home/dietpi/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/symvers/shortcut-fe.symvers" KERNELRELEASE=5.19.1 CONFIG_MT76_USB=m CONFIG_MT76x02_LIB=m CONFIG_MT76x02_USB=m CONFIG_MT76x2_COMMON=m CONFIG_MT76x2U=m M="/home/dietpi/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/mt76-2022-07-03-93e3fce9" NOSTDINC_FLAGS="-nostdinc  -isystem /home/dietpi/lede/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/lib/gc [... cut ...]
make[3]: Leaving directory '/home/dietpi/lede/package/kernel/mt76'
time: package/kernel/mt76/compile#13.89#3.42#7.03
    ERROR: package/kernel/mt76 failed to build.
make[2]: *** [package/Makefile:116: package/kernel/mt76/compile] Error 1
make[2]: Leaving directory '/home/dietpi/lede'
make[1]: *** [package/Makefile:110: /home/dietpi/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/dietpi/lede'
make: *** [/home/dietpi/lede/include/toplevel.mk:230: world] Error 2

ricksuzade-maker, 2022-08-15:
mt76还不支持5.19内核，作者还在等

WYC-2020, 2022-08-16:
usb_maxpacket 函数原型变了，这个需要改下调用参数，用宏区分内核版本 就可以了

closed 2022-09-03 by aiamadeus