coolsnowwolf/lede #9951: 5.19.x 内核后 R4S 和 Shortcut-FE 存在 Kernel panic 随机重启问题

BenjaminX, 2022-08-16:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
切最新的代码分支编译固件，存在随机重启问题。

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：Shortcut-FE 引起随机重启，关闭 Shortcut-FE 问题解决；但是，关闭后会严重降速。

### (2) 路由器型号和固件版本

A：R4S + 最新 lean 直接 master 代码分支

### (3) 详细日志

A：
[ 6131.496625] Unable to handle kernel NULL pointer dereference at virtual address 0000000000000936
[ 6131.497426] Mem abort info:
[ 6131.497681] ESR = 0x0000000096000006
[ 6131.498070] EC = 0x25: DABT (current EL), IL = 32 bits
[ 6131.498551] SET = 0, FnV = 0
[ 6131.498829] EA = 0, S1PTW = 0
[ 6131.499114] FSC = 0x06: level 2 translation fault
[ 6131.499552] Data abort info:
[ 6131.499813] ISV = 0, ISS = 0x00000006
[ 6131.500158] CM = 0, WnR = 0
[ 6131.500429] user pgtable: 4k pages, 48-bit VAs, pgdp=000000000c980000
[ 6131.501006] [0000000000000936] pgd=0800000009e5a003, p4d=0800000009e5a003, pud=080000000c414003, pmd=0000000000000000
[ 6131.501989] Internal error: Oops: 96000006 [https://github.com/coolsnowwolf/lede/issues/1] PREEMPT SMP

BenjaminX, 2022-08-16:
[image]

WYC-2020, 2022-08-16:
5.19 很多玩意都没匹配好，甚至还有内存泄漏的 有的patch 直接连释放都没有，不建议在主路由上面玩这个，如果只是玩的话 到可以

BenjaminX, 2022-08-16:
回滚 5.18.15 内核，暂时解决问题。

moonerick, 2022-08-17:
> 

赞同，前面5.18.14开机20天不关机，毫无问题，现在测试了5.19，有莫名奇妙的问题，之前是遇到莫名的程序无法启动，今天又遇到了AdGuardHome莫名的崩溃，遇到问题后只能重启机器才能解决问题

kumitom, 2022-08-19:
我这几天编译了版本后也出现下载时随机重启的情况，也是关闭了sfe，但是关闭了也没感觉降速啥的，所以就还好

BenjaminX, 2022-08-23:
打算回滚 5.15 先看看了，最新的 5.19.3 还是一样随机重启

keeno1982, 2022-08-25:
这个问题解决了吗？

closed 2022-09-03 by coolsnowwolf