coolsnowwolf/lede #9952: 有没有办法把独立的ipk编译进去固件顺便把依赖装了

grdll, 2022-08-16:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：想要装的ipk直接装有些依赖会提示 kernel = required 所以想问问看怎么把独立的ipk编译进去固件顺便把依赖装了

### (2) 路由器型号和固件版本

A：x86_64

### (3) 详细日志

A：Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.134-1-ed4b70894fa69e9cb42e2a4d9550231c) for kmod-crypto-hash
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.134-1-ed4b70894fa69e9cb42e2a4d9550231c) for kmod-crypto-null
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.134-1-ed4b70894fa69e9cb42e2a4d9550231c) for kmod-crypto-aead
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.134-1-ed4b70894fa69e9cb42e2a4d9550231c) for kmod-crypto-manager
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.134-1-ed4b70894fa69e9cb42e2a4d9550231c) for kmod-crypto-authenc
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.134-1-ed4b70894fa69e9cb42e2a4d9550231c) for kmod-nf-reject
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.134-1-ed4b70894fa69e9cb42e2a4d9550231c) for kmod-nf-ipt
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.134-1-ed4b70894fa69e9cb42e2a4d9550231c) for kmod-nf-log
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-nf-log found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.134-1-ed4b70894fa69e9cb42e2a4d9550231c) for kmod-ipt-core
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.134-1-ed4b70894fa69e9cb42e2a4d9550231c) for kmod-nf-conntrack
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.134-1-ed4b70894fa69e9cb42e2a4d9550231c) for kmod-ipt-conntrack
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.134-1-ed4b70894fa69e9cb42e2a4d9550231c) for kmod-nf-conntrack6
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.134-1-ed4b70894fa69e9cb42e2a4d9550231c) for kmod-nf-nat
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.134-1-ed4b70894fa69e9cb42e2a4d9550231c) for kmod-ipt-nat
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.134-1-ed4b70894fa69e9cb42e2a4d9550231c) for kmod-nfnetlink
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.134-1-ed4b70894fa69e9cb42e2a4d9550231c) for kmod-nf-reject6
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.134-1-ed4b70894fa69e9cb42e2a4d9550231c) for kmod-nf-log6
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-nf-log6 found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.134-1-ed4b70894fa69e9cb42e2a4d9550231c) for kmod-crypto-crc32c
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.134-1-ed4b70894fa69e9cb42e2a4d9550231c) for kmod-lib-crc32c
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.134-1-ed4b70894fa69e9cb42e2a4d9550231c) for kmod-nft-core
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-nft-core found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.134-1-ed4b70894fa69e9cb42e2a4d9550231c) for kmod-nft-compat
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-nft-compat found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency wrtbwmon for luci-wrtbwmon
 * pkg_hash_fetch_best_installation_candidate: Packages for luci-wrtbwmon found, but incompatible with the architectures configured

zhangguanzhang, 2022-08-16:
用我的固件 https://github.com/zhangguanzhang/Actions-OpenWrt/tree/test#%E6%94%AF%E6%8C%81%E7%9A%84%E8%AE%BE%E5%A4%87%E5%88%97%E8%A1%A8 
[image]

Cathgao, 2022-08-24:
menuconfig选上不就编译进固件了吗

grdll, 2022-08-25:
> menuconfig选上不就编译进固件了吗

是独立的ipk文件不是feeds里那种

zhangguanzhang, 2022-08-25:
```
root@OpenWrt:~# ls -l /local_feed/luci-theme-*.ipk | wc -l
11
root@OpenWrt:~# ls -l /local_feed/luci-app-*.ipk | wc -l
149
root@OpenWrt:~# ls -l /local_feed/*.ipk | wc -l
1351
```
https://github.com/zhangguanzhang/Actions-OpenWrt/
我的固件本地源，L 大的 kernel hash对不上op的mirror的ipk，所以你用L大固件安装op源的一些kmod就会报错依赖问题，解决办法就是编译大部分 ipk，整进去做本地源

Racon94, 2022-08-29:
一般生成的package文件夹里面有一个依赖文件的吧。我都直接全部复制到nas上，用nginx代理目录，当做本地源的。用起来还行

closed 2022-09-03 by aiamadeus