coolsnowwolf/lede #9954: 请求TCP DNS 53端口时end of file

w311ang, 2022-08-16:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：请求一个未请求过UDP53的域名如facebook.com
```
~ $ dig @192.168.2.1 facebook.com +tcp
;; communications error to 192.168.2.1#53: end of file

;; communications error to 192.168.2.1#53: end of file
```
但如果通过UDP53请求了一次，再请求TCP53时则恢复正常
```
~ $ dig @192.168.2.1 facebook.com +notcp

; <<>> DiG 9.16.27 <<>> @192.168.2.1 facebook.com +notcp
; (1 server found)
;; global options: +cmd
;; Got answer:
;; ->>HEADER<<- opcode: QUERY, status: NOERROR, id: 24047
;; flags: qr rd ra; QUERY: 1, ANSWER: 1, AUTHORITY: 0, ADDITIONAL: 1

;; OPT PSEUDOSECTION:
; EDNS: version: 0, flags:; udp: 1232
;; QUESTION SECTION:
;facebook.com.                  IN      A

;; ANSWER SECTION:
facebook.com.           0       IN      A       31.13.75.35

;; Query time: 130 msec
;; SERVER: 192.168.2.1#53(192.168.2.1)
;; WHEN: Tue Aug 16 16:54:28 CST 2022
;; MSG SIZE  rcvd: 57

~ $ dig @192.168.2.1 facebook.com +tcp

; <<>> DiG 9.16.27 <<>> @192.168.2.1 facebook.com +tcp
; (1 server found)
;; global options: +cmd
;; Got answer:
;; ->>HEADER<<- opcode: QUERY, status: NOERROR, id: 18040
;; flags: qr rd ra; QUERY: 1, ANSWER: 1, AUTHORITY: 0, ADDITIONAL: 1

;; OPT PSEUDOSECTION:
; EDNS: version: 0, flags:; udp: 1232
;; QUESTION SECTION:
;facebook.com.                  IN      A

;; ANSWER SECTION:
facebook.com.           196     IN      A       31.13.75.35

;; Query time: 30 msec
;; SERVER: 192.168.2.1#53(192.168.2.1)
;; WHEN: Tue Aug 16 16:54:53 CST 2022
;; MSG SIZE  rcvd: 57
```
因为默认劫持所有TCP53到了路由器，若还使用ssrplus该问题会在国内IP的DNS的TCP53上可以复现，因为之前已经在UDP53上请求过facebook.com域名，所以没有出现问题
https://github.com/coolsnowwolf/lede/blob/7e274cbde20495634b240f566ec8b0b77da074b0/package/lean/default-settings/files/zzz-default-settings#L41
```
~ $ dig @114.114.114.114 twitter.com +tcp
;; communications error to 114.114.114.114#53: end of file

;; communications error to 114.114.114.114#53: end of file

~ $ dig @114.114.114.114 facebook.com +tcp

; <<>> DiG 9.16.27 <<>> @114.114.114.114 facebook.com +tcp
; (1 server found)
;; global options: +cmd
;; Got answer:
;; ->>HEADER<<- opcode: QUERY, status: NOERROR, id: 64967
;; flags: qr rd ra; QUERY: 1, ANSWER: 1, AUTHORITY: 0, ADDITIONAL: 1

;; OPT PSEUDOSECTION:
; EDNS: version: 0, flags:; udp: 1232
;; QUESTION SECTION:
;facebook.com.                  IN      A

;; ANSWER SECTION:
facebook.com.           5       IN      A       31.13.75.35

;; Query time: 30 msec
;; SERVER: 114.114.114.114#53(114.114.114.114)
;; WHEN: Tue Aug 16 16:58:04 CST 2022
;; MSG SIZE  rcvd: 57
```

### (2) 路由器型号和固件版本

A：Phicomm K2P OpenWrt R22.7.7 / LuCI Master (git-22.200.61437-0c3c82f)

### (3) 详细日志

A：

w311ang, 2022-08-16:
<details>
<summary>系统日志</summary>

```
Tue Aug 16 08:18:40 2022 kern.notice kernel: [    0.000000] Linux version 5.4.203 (runner@fv-az290-183) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r4750-94306d382)) #0 SMP Thu Jul 28 05:46:25 2022
Tue Aug 16 08:18:40 2022 kern.info kernel: [    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
Tue Aug 16 08:18:40 2022 kern.info kernel: [    0.000000] printk: bootconsole [early0] enabled
Tue Aug 16 08:18:40 2022 kern.info kernel: [    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
Tue Aug 16 08:18:40 2022 kern.info kernel: [    0.000000] MIPS: machine is Phicomm K2P
Tue Aug 16 08:18:40 2022 kern.info kernel: [    0.000000] Initrd not found or empty - disabling initrd
Tue Aug 16 08:18:40 2022 kern.info kernel: [    0.000000] VPE topology {2,2} total 4
Tue Aug 16 08:18:40 2022 kern.warn kernel: [    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
Tue Aug 16 08:18:40 2022 kern.warn kernel: [    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
Tue Aug 16 08:18:40 2022 kern.warn kernel: [    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
Tue Aug 16 08:18:40 2022 kern.info kernel: [    0.000000] Zone ranges:
Tue Aug 16 08:18:40 2022 kern.info kernel: [    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
Tue Aug 16 08:18:40 2022 kern.info kernel: [    0.000000]   HighMem  empty
Tue Aug 16 08:18:40 2022 kern.info kernel: [    0.000000] Movable zone start for each node
Tue Aug 16 08:18:40 2022 kern.info kernel: [    0.000000] Early memory node ranges
Tue Aug 16 08:18:40 2022 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
Tue Aug 16 08:18:40 2022 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
Tue Aug 16 08:18:40 2022 kern.debug kernel: [    0.000000] On node 0 totalpages: 32768
Tue Aug 16 08:18:40 2022 kern.debug kernel: [    0.000000]   Normal zone: 288 pages used for memmap
Tue Aug 16 08:18:40 2022 kern.debug kernel: [    0.000000]   Normal zone: 0 pages reserved
Tue Aug 16 08:18:40 2022 kern.debug kernel: [    0.000000]   Normal zone: 32768 pages, LIFO batch:7
Tue Aug 16 08:18:40 2022 kern.info kernel: [    0.000000] percpu: Embedded 14 pages/cpu s27152 r8192 d22000 u57344
Tue Aug 16 08:18:40 2022 kern.debug kernel: [    0.000000] pcpu-alloc: s27152 r8192 d22000 u57344 alloc=14*4096
Tue Aug 16 08:18:40 2022 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
Tue Aug 16 08:18:40 2022 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 32480
Tue Aug 16 08:18:40 2022 kern.notice kernel: [    0.000000] Kernel command line: console=ttyS0,57600 rootfstype=squashfs,jffs2
Tue Aug 16 08:18:40 2022 kern.info kernel: [    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
Tue Aug 16 08:18:40 2022 kern.info kernel: [    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
Tue Aug 16 08:18:40 2022 kern.info kernel: [    0.000000] Writing ErrCtl register=00005001
Tue Aug 16 08:18:40 2022 kern.info kernel: [    0.000000] Readback ErrCtl register=00005001
Tue Aug 16 08:18:40 2022 kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Tue Aug 16 08:18:40 2022 kern.info kernel: [    0.000000] Memory: 120264K/131072K available (6165K kernel code, 273K rwdata, 1296K rodata, 1276K init, 235K bss, 10808K reserved, 0K cma-reserved, 0K highmem)
Tue Aug 16 08:18:40 2022 kern.info kernel: [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Tue Aug 16 08:18:40 2022 kern.info kernel: [    0.000000] rcu: Hierarchical RCU implementation.
Tue Aug 16 08:18:40 2022 kern.info kernel: [    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
Tue Aug 16 08:18:40 2022 kern.info kernel: [    0.000000] NR_IRQS: 256
Tue Aug 16 08:18:40 2022 kern.info kernel: [    0.000000] CPU Clock: 880MHz
Tue Aug 16 08:18:40 2022 kern.info kernel: [    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
Tue Aug 16 08:18:40 2022 kern.info kernel: [    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4343773742 ns
Tue Aug 16 08:18:40 2022 kern.info kernel: [    0.000009] sched_clock: 32 bits at 440MHz, resolution 2ns, wraps every 4880645118ns
Tue Aug 16 08:18:40 2022 kern.info kernel: [    0.015575] Calibrating delay loop... 586.13 BogoMIPS (lpj=2930688)
Tue Aug 16 08:18:40 2022 kern.info kernel: [    0.087904] pid_max: default: 32768 minimum: 301
Tue Aug 16 08:18:40 2022 kern.info kernel: [    0.097244] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Tue Aug 16 08:18:40 2022 kern.info kernel: [    0.111651] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Tue Aug 16 08:18:40 2022 kern.info kernel: [    0.130188] rcu: Hierarchical SRCU implementation.
Tue Aug 16 08:18:40 2022 kern.info kernel: [    0.140474] smp: Bringing up secondary CPUs ...
Tue Aug 16 08:18:40 2022 kern.warn kernel: [    0.240019] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
Tue Aug 16 08:18:40 2022 kern.warn kernel: [    0.240030] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
Tue Aug 16 08:18:40 2022 kern.warn kernel: [    0.240043] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[... cut ...]
[26275.647658] AP SETKEYS DONE - AKMMap=WPA2PSK, PairwiseCipher=AES, GroupCipher=AES from D8:2F:E6:76:E9:C7
[26275.647658] 
[26275.748246] AP SETKEYS DONE - AKMMap=WPA2PSK, PairwiseCipher=AES, GroupCipher=AES from A8:1B:5A:AC:A2:FA
[26275.748246] 
[26275.770131] AP SETKEYS DONE - AKMMap=WPA2PSK, PairwiseCipher=AES, GroupCipher=AES from 94:0E:6B:1A:27:CA
[26275.770131] 
[27272.570833] 64d, flush one!
[27295.144061] :MtCmdPktBudgetCtrl: bssid(255),wcid(65535),type(1)
[27296.243872] :MtCmdPktBudgetCtrl: bssid(255),wcid(65535),type(0)
```

</details>

w311ang, 2022-08-19:
解决办法：编译[tcp2udp](https://github.com/w311ang/openwrt-tcp2udp)进系统，在启动项添加并到ssh运行
```bash
(tcp2udp 127.0.0.1:5335 :5335 >/dev/null 2>&1)&
```
再换一个支持tcp的上游dns

coolsnowwolf, 2026-09-05:
报告已定位到 SSR Plus 使用的 dns2tcp/上游 TCP DNS 不支持：首次 TCP 查询被上游关闭，UDP 查询缓存后才正常。这是第三方代理链路配置行为，不是 dnsmasq/LEDE 可在当前代码中修复的缺陷；请改用支持 TCP 的上游或关闭 TCP53 劫持，关闭为“不计划处理”。

closed 2026-09-05 by coolsnowwolf