coolsnowwolf/lede #9955: 未检测到 Realtek PCI 网卡

umarizulkifli, 2022-08-16:
反馈错误/问题模板，请删除建议

## 1. 关于你要提交的问题
未检测到以太网适配器

Q：问题是否被搜索（使用“x”选择）
* [x] 没有类似问题

## 2.详细说明
```
以太网控制器 [0200]：Realtek Semiconductor Co., Ltd. RTL810xE PCI Express 快速以太网控制器 [10ec:8136] (rev 02)
```
r8101 的以太网驱动程序不可用
驱动源码：https://github.com/ghostrider-reborn/realtek-r8101-linux-driver
Git 提交：https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=0e30811583e4ac97cd16d877cfc2f70887f33c6b
### (1) 具体问题

A：

### (2) 路由器型号和固件版本

答：联想 S10-3

### (3) 详细日志
```
$ lspci -v
以太网控制器 [0200]：Realtek Semiconductor Co., Ltd. RTL810xE PCI Express 快速以太网控制器 [10ec:8136] (rev 02)
$ ifconfig eth0
ifconfig: eth0: error fetching interface information: Device not found
```
A：

umarizulkifli, 2022-08-19:
Failed to build r8101

`ERROR: package/lean/r8101 failed to build.`

zhangj3, 2022-08-25:
百兆网卡....别折腾了  跟上主流设备才是王道

bihell, 2022-08-26:
> 百兆网卡....别折腾了 跟上主流设备才是王道

厉害啊，楼主节俭持家，挺好的。

closed 2022-08-27 by coolsnowwolf