coolsnowwolf/lede #9959: NanoPi R2S R2C出现找不到u-boot.img 编译失败

aaronchin, 2022-08-17:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [X] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译过程中出现找不到u-boot img导致失败

### (2) 路由器型号和固件版本

A：Rockchip RK3328 FriendlyArm NanoPi R2C＆R2S

### (3) 详细日志

A：
+ rm -f /home/aaronchin/tmp/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/tmp/openwrt-rockchip-armv8-friendlyarm_nanopi-r2s-ext4-sysupgrade.img.gz.kernel
# Copy the idbloader, uboot and trust image to the image at sector 0x40, 0x4000 and 0x6000
dd if="/home/aaronchin/tmp/lede/staging_dir/target-aarch64_generic_musl/image"/rk3328-idbloader.bin of="/home/aaronchin/tmp/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/tmp/openwrt-rockchip-armv8-friendlyarm_nanopi-r2s-ext4-sysupgrade.img.gz" seek=64 conv=notrunc
205+1 records in
205+1 records out
104964 bytes (105 kB, 103 KiB) copied, 0.166274 s, 631 kB/s
dd if="/home/aaronchin/tmp/lede/staging_dir/target-aarch64_generic_musl/image"/nanopi-r2s-rk3328-uboot.img of="/home/aaronchin/tmp/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/tmp/openwrt-rockchip-armv8-friendlyarm_nanopi-r2s-ext4-sysupgrade.img.gz" seek=16384 conv=notrunc
dd: failed to open '/home/aaronchin/tmp/lede/staging_dir/target-aarch64_generic_musl/image/nanopi-r2s-rk3328-uboot.img': No such file or directory
make[5]: *** [Makefile:80: /home/aaronchin/tmp/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/tmp/openwrt-rockchip-armv8-friendlyarm_nanopi-r2s-ext4-sysupgrade.img.gz] Error 1

zhangguanzhang, 2022-08-17:
你编译的是r2s还是r2c，是不是开了`CONFIG_TARGET_MULTI_PROFILE=y`

aaronchin, 2022-08-17:
> 你编译的是r2s还是r2c，是不是开了`CONFIG_TARGET_MULTI_PROFILE=y`

没开 multi profile 
编译r2c r2s 都一样的结果

closed 2022-08-19 by aiamadeus