coolsnowwolf/lede #9960: R5S 5.10内核编译不过

ostrichii, 2022-08-17:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ X] 没有类似的issue

## 2. 详细叙述
编译不通过，看起来是缺文件
### (1) 具体问题

A：编译不通过，看起来是缺文件

### (2) 路由器型号和固件版本

A：R5S

### (3) 详细日志

A：
Configuring uhttpd-mod-ubus.
Configuring rpcd-mod-rrdns.
Configuring luci.
Configuring default-settings.
Configuring boost.
Configuring htop.
Configuring luci-i18n-base-zh-cn.
Configuring boost-program_options.
Configuring boost-system.
Configuring bc.
Configuring fdisk.
Configuring shadowsocks-libev-ss-server.
Configuring kmod-lib-lzo.
Configuring luci-i18n-upnp-zh-cn.
Configuring libatomic1.
Configuring luci-i18n-firewall-zh-cn.
Configuring ppp-mod-pppoe.
Configuring dnsmasq-full.
Setting flags for package libc to hold.
Setting flags for package libgcc1 to hold.
Setting flags for package libatomic1 to hold.
Setting flags for package libstdcpp6 to hold.
Setting flags for package libpthread to hold.
Setting flags for package librt to hold.
cp -fpR /home/ostrich/r5s/lede/build_dir/target-aarch64_generic_musl/root-rockchip /home/ostrich/r5s/lede/build_dir/target-aarch64_generic_musl/root.orig-rockchip
sed: can't read /usr/lib/lua/luci/dispatcher.lua: No such file or directory
sed: can't read /usr/lib/lua/luci/dispatcher.lua: No such file or directory
Enabling arpbind
Enabling boot
Enabling cron
Enabling ddns
Enabling dnsmasq
Enabling done
Enabling dropbear
Enabling firewall
Enabling fstab
Enabling gpio_switch
Enabling haveged
Enabling kcptun
Enabling kms
Enabling led
Enabling log
Enabling mia
Enabling miniupnpd
Enabling netdata
Enabling network
Enabling openssl
Enabling rpcd
Enabling sqm
Enabling sysctl
[... 121 lines cut ...]
/home/ostrich/r5s/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.136/arch/arm64/boot/dts/rockchip/rk3399.dtsi:1939.25-1950.5: Warning (interrupt_provider): /pinctrl/gpio0@ff720000: Missing #address-cells in interrupt provider
/home/ostrich/r5s/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.136/arch/arm64/boot/dts/rockchip/rk3399.dtsi:1952.25-1963.5: Warning (interrupt_provider): /pinctrl/gpio1@ff730000: Missing #address-cells in interrupt provider
/home/ostrich/r5s/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.136/arch/arm64/boot/dts/rockchip/rk3399.dtsi:1965.25-1976.5: Warning (interrupt_provider): /pinctrl/gpio2@ff780000: Missing #address-cells in interrupt provider
/home/ostrich/r5s/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.136/arch/arm64/boot/dts/rockchip/rk3399.dtsi:1978.25-1989.5: Warning (interrupt_provider): /pinctrl/gpio3@ff788000: Missing #address-cells in interrupt provider
/home/ostrich/r5s/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.136/arch/arm64/boot/dts/rockchip/rk3399.dtsi:1991.25-2002.5: Warning (interrupt_provider): /pinctrl/gpio4@ff790000: Missing #address-cells in interrupt provider
rm -f /home/ostrich/r5s/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3399-nanopi-r4se.dtb.tmp
aarch64-openwrt-linux-musl-cpp -nostdinc -x assembler-with-cpp -I/home/ostrich/r5s/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.136/arch/arm64/boot/dts -I/home/ostrich/r5s/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.136/arch/arm64/boot/dts/include -I/home/ostrich/r5s/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.136/include/ -undef -D__DTS__  -o /home/ostrich/r5s/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3568-nanopi-r5s.dtb.tmp /home/ostrich/r5s/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.136/arch/arm64/boot/dts/rockchip/rk3568-nanopi-r5s.dts
In file included from /home/ostrich/r5s/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.136/arch/arm64/boot/dts/rockchip/rk3568-nanopi-r5s.dts:10:
/home/ostrich/r5s/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.136/arch/arm64/boot/dts/rockchip/rk3568.dtsi:6:10: fatal error: rk356x.dtsi: No such file or directory
 #include "rk356x.dtsi"
          ^~~~~~~~~~~~~
compilation terminated.
make[5]: *** [Makefile:78: /home/ostrich/r5s/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3568-nanopi-r5s.dtb] Error 1
make[5]: Leaving directory '/home/ostrich/r5s/lede/target/linux/rockchip/image'
make[4]: *** [Makefile:25: install] Error 2
make[4]: Leaving directory '/home/ostrich/r5s/lede/target/linux/rockchip'
make[3]: *** [Makefile:11: install] Error 2
make[3]: Leaving directory '/home/ostrich/r5s/lede/target/linux'
time: target/linux/install#6.18#1.68#7.74
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:25: target/linux/install] Error 1
make[2]: Leaving directory '/home/ostrich/r5s/lede'
make[1]: *** [target/Makefile:19: /home/ostrich/r5s/lede/staging_dir/target-aarch64_generic_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/home/ostrich/r5s/lede'
make: *** [/home/ostrich/r5s/lede/include/toplevel.mk:230: world] Error 2

DHDAXCW, 2022-08-17:
5.10内核没有rk3568，改5.19吧

ostrichii, 2022-08-17:
> 5.10内核没有rk3568，改5.19吧

好的，我试试5.19看看稳不稳

closed 2022-08-17 by ostrichii