coolsnowwolf/lede #9968: Hyper-V虚拟OpenWrt主机无法通过Hyper-V管理器关机

ronnie9211, 2022-08-18:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：想通过Hyper-V管理器关闭OpenWrt时提示“尝试关闭虚拟机时出错”，错误代码0x800710DF，只能通过强制关机选项关机。

### (2) 路由器型号和固件版本

A：X86Hyper-V虚拟主机，OpenWrt R22.8.2 / LuCI Master (git-22.227.13225-9632640)，内核版本5.15.60
### (3) 详细日志

A：

zhangguanzhang, 2022-08-18:
你安装下acpid的看看有效果没

littoy, 2022-08-19:
我加了acpid也是不行

ronnie9211, 2022-08-19:
> 

装了，还是不行

zhangguanzhang, 2022-08-19:
```
/etc/init.d/acpid status
```
看看是running的吗

ronnie9211, 2022-08-19:
> ```
> /etc/init.d/acpid status
> ```
> 
> 看看是running的吗

是的，并且我上一个固件R21.12.1没有acpid也可以通过Hyper-V管理器正常关机。

zhangguanzhang, 2022-08-19:
你切5.15内核试试看看解决没

littoy, 2022-08-20:
我试过5.19 5.15 和5.10都不行

littoy, 2022-08-28:
> 我试过5.19 5.15 和5.10都不行

纠正下，5.10可以。

fishg, 2022-12-21:
5.15内核的，从dmesg 看是缺了 hv_utils ，但是我不知道为什么，日志:
dmesg |grep hv
[    0.031517] hv_vmbus: Vmbus version:5.3
[    0.038608] hv_vmbus: registering driver hv_pci
[    0.038691] hv_vmbus: registering driver hyperv_fb
[    0.039125] hyperv_fb: Synthvid Version major 3, minor 5
[    0.040173] hv_vmbus: registering driver hv_storvsc
[    0.040412] hv_vmbus: registering driver hv_netvsc
[    0.058294] hv_vmbus: registering driver hyperv_keyboard
[    0.062053] hv_vmbus: registering driver hid_hyperv
[    0.064844] hv_vmbus: registering driver hv_balloon
[    0.065428] hv_balloon: Using Dynamic Memory protocol version 2.0
[   48.144006] hv_balloon: Max. dynamic memory size: 4096 MB

gsown, 2022-12-31:
.config里加上hv_utils然后重新变一下，6.1内核实测加上就可以了
`CONFIG_DEFAULT_kmod-hv_utils=y`

gsown, 2022-12-31:
应该是少了hv_utils.ko导致的

[image]

装ipk装不上，但是单独加载hv_utils.ko文件就没问题，加载后hyper-v关机就正常了

littoy, 2022-12-31:
这个kmod-hv_utils 的ipk怎么编译的？我手动加CONFIG_DEFAULT_kmod-hv_utils=y ,make defconfig就又无了

> 应该是少了hv_utils.ko导致的
> 
> [image]
> 
> 装ipk装不上，但是单独加载hv_utils.ko文件就没问题，加载后hyper-v关机就正常了

gsown, 2022-12-31:
> 这个kmod-hv_utils 的ipk怎么编译的？我手动加CONFIG_DEFAULT_kmod-hv_utils=y ,make defconfig就又无了
> 
> > 应该是少了hv_utils.ko导致的
> > [image]
> > 装ipk装不上，但是单独加载hv_utils.ko文件就没问题，加载后hyper-v关机就正常了

你在lede目录里find下
`find ./ -name hv_utils*`

littoy, 2023-01-01:
找到ko模块了，谢谢

> > 这个kmod-hv_utils 的ipk怎么编译的？我手动加CONFIG_DEFAULT_kmod-hv_utils=y ,make defconfig就又无了
> > > 应该是少了hv_utils.ko导致的
> > > [image]
> > > 装ipk装不上，但是单独加载hv_utils.ko文件就没问题，加载后hyper-v关机就正常了
> 
> 你在lede目录里find下 `find ./ -name hv_utils*`

littoy, 2023-01-01:
从配置看是应该直接编译进内核的？不知道为什么默认没有加载

littoy, 2023-01-01:
看起来生成的build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.85/.config 跟target/linux/x86/64/config-5.15相去甚远

fishg, 2023-01-04:
[image]
弄个kmod-hv_utils 试试

malash, 2023-02-23:
@fishg 你好，请问这一段代码需要加在哪里？

fishg, 2023-02-24:
> @fishg 你好，请问这一段代码需要加在哪里？

主线已经修复了，pull最新的代码编译直接就好了。

malash, 2023-02-24:
@fishg 测试了一下，确实可以了，感谢～

coolsnowwolf, 2026-09-05:
该报告来自 R22.8.2/内核 5.15.60 的 Hyper-V 主机关机行为，未提供 OpenWrt 日志，错误码 0x800710DF 属于 Hyper-V 主机管理器/虚拟机配置层。当前仓库没有可验证的 LEDE 代码故障或复现材料，关闭为 not planned。

closed 2026-09-05 by coolsnowwolf