coolsnowwolf/lede #9970: Request features USB Interfaces network

GenzPN, 2022-08-19:
Feedback bug/issue template, please delete suggestions

## 1. About the question you want to submit

Q: Is the issue searched (use "x" to select)
* [X] No similar issue

## 2. Detailed description
I want developers to be able to add a USB interface section at
Ethernet Adapter: "eth1"

### (1) Specific questions

A: None

### (2) Router model and firmware version

A: Xiaomi R3G Gen1

### (3) Detailed log

A: No log
[image]

zhangguanzhang, 2022-08-19:
https://zhangguanzhang.github.io/2021/09/03/openwrt-usb-net
edit your `.config`  and `make defconfig && make`

GenzPN, 2022-08-19:
Let me try

zhangguanzhang, 2022-08-19:
> Let me try

should add this
```
CONFIG_PACKAGE_kmod-usb-net=y
```

zhangguanzhang, 2022-08-19:
used the phone turn on the usb net share, connect it to the device
run this command:
```
lsusb
ip a s 
```

GenzPN, 2022-08-19:
```

              ___         __        ____        _ __    ___  __
             /   | __  __/ / ____  / __ )__  __(_) /___/ / |/ /
            / /| |/ / / / __/ __ \/ __  / / / / / / __  /|   /
           / /__ / /_/ / /_/ /_/ / /_/ / /_/ / / / /_/ //   |
          /_/  |_\__,_/\__/\____/_____/\__,_/_/_/\__,_//_/|_|

                                                By GenzPN

                | Powered by GenzPN @ xiaowansm5 |

          ———————————————————————————————————————————————————
             Openwrt R22.08-20220819 / AutoUpdate V6.9.3
          ———————————————————————————————————————————————————

            Kernel version:             5.4.203
            Available memory:           120.5M  | 49.1% of 245.5M
            Available storage:          63.7M   | 91.5% of 69.6M
             Uptime:            0 Day 0 Hour 38 Min 34 Sec
            Logged in user:             root    | 192.168.15.171:49793

            ==================== Warning ========================
             The current device does not have a password set for the root user!
             Please use the passwd command to set a system password for increased security
             At the same time prevent unauthorized SSH use of root to log in illegally
            --------------------------------------------------

root@GenzPN:~# lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux 5.4.203 xhci-hcd xHCI Host Controller
Bus 001 Device 030: ID 1410:b024 Novatel Wireless USB800
Bus 001 Device 001: ID 1d6b:0002 Linux 5.4.203 xhci-hcd xHCI Host Controller
root@GenzPN:~# ip a s
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP group default qlen 1000
    link/ether 40:31:3c:00:dd:68 brd ff:ff:ff:ff:ff:ff
    inet6 fe80::4231:3cff:fe00:dd68/64 scope link
       valid_lft forever preferred_lft forever
3: wan@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000
    link/ether 40:31:3c:00:dd:67 brd ff:ff:ff:ff:ff:ff
    inet 192.168.10.183/24 brd 192.168.10.255 scope global wan
       valid_lft forever preferred_lft forever
    inet6 fdf3:834c:bf8f:0:4231:3cff:fe00:dd67/64 scope global noprefixroute
       valid_lft forever preferred_lft forever
    inet6 fd5b:b3b4:5eb0:4:4231:3cff:fe00:dd67/64 scope global dynamic noprefixroute
       valid_lft 25637sec preferred_lft 25637sec
    inet6 fdf3:834c:bf8f::f2e/128 scope global dynamic noprefixroute
       valid_lft 43160sec preferred_lft 25638sec
    inet6 fd5b:b3b4:5eb0:4::f2e/128 scope global dynamic noprefixroute
       valid_lft 25638sec preferred_lft 25638sec
    inet6 fe80::4231:3cff:fe00:dd67/64 scope link
       valid_lft forever preferred_lft forever
4: lan2@eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue master br-lan state LOWERLAYERDOWN group default qlen 1000
    link/ether 40:31:3c:00:dd:68 brd ff:ff:ff:ff:ff:ff
5: lan1@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br-lan state UP group default qlen 1000
    link/ether 40:31:3c:00:dd:68 brd ff:ff:ff:ff:ff:ff
6: sit0@NONE: <NOARP> mtu 1480 qdisc noop state DOWN group default qlen 1000
    link/sit 0.0.0.0 brd 0.0.0.0
10: br-lan: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000
    link/ether 40:31:3c:00:dd:68 brd ff:ff:ff:ff:ff:ff
    inet 192.168.15.1/24 brd 192.168.15.255 scope global br-lan
       valid_lft forever preferred_lft forever
    inet6 fdf3:834c:bf8f:2::1/63 scope global dynamic noprefixroute
       valid_lft 43160sec preferred_lft 25638sec
    inet6 fd5b:b3b4:5eb0:6::1/63 scope global dynamic noprefixroute
[... 58 lines cut ...]
kmod-usb-net-cdc-eem - 5.4.203-1
kmod-usb-net-cdc-ether - 5.4.203-1
kmod-usb-net-cdc-mbim - 5.4.203-1
kmod-usb-net-cdc-ncm - 5.4.203-1
kmod-usb-net-huawei-cdc-ncm - 5.4.203-1
kmod-usb-net-ipheth - 5.4.203-1
kmod-usb-net-qmi-wwan - 5.4.203-1
kmod-usb-net-rndis - 5.4.203-1
kmod-usb-net-rtl8152 - 5.4.203-1
kmod-usb-ohci - 5.4.203-1
kmod-usb-serial - 5.4.203-1
kmod-usb-serial-option - 5.4.203-1
kmod-usb-serial-pl2303 - 5.4.203-1
kmod-usb-serial-qualcomm - 5.4.203-1
kmod-usb-serial-sierrawireless - 5.4.203-1
kmod-usb-serial-wwan - 5.4.203-1
kmod-usb-wdm - 5.4.203-1
kmod-usb-xhci-hcd - 5.4.203-1
kmod-usb-xhci-mtk - 5.4.203-1
libusb-1.0-0 - 1.0.24-5
usb-modeswitch - 2022-02-24-3c8595a4-1
root@GenzPN:~#

```
Never mind download OPKG

zhangguanzhang, 2022-08-19:
`Novatel Wireless USB800`  is your phone?

GenzPN, 2022-08-19:
Yeah it a AT&T USB800 MiFi USB LTE
[image]

zhangguanzhang, 2022-08-19:
this might be qmi protocol device，I think you should enable this:
```
CONFIG_PACKAGE_luci-proto-qmi=y
```
and read this doc https://openwrt.org/docs/guide-user/network/wan/wwan/ltedongle#required_packages

GenzPN, 2022-08-19:
You can check my config in here

GenzPN, 2022-08-20:
Fix with here https://github.com/GenzPN/GenzPN-lede/commit/1f06fbf9fa350100fb6b78a22e58356abfee5e72

closed 2022-08-20 by GenzPN