coolsnowwolf/lede #9971: mt7921e网卡wlan开启wap-psk/wap2-psk 很难连接

littoy, 2022-08-20:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：如果我不选No Encrytion，用wap-psk/wap2-psk， 或者单独wap2-psk，都经常连上wifi不分配ip。手动指定ip也不通，即使ping网关也不通。用No Encrytion秒连，信号也正常。

### (2) 路由器型号和固件版本

A：r5s mt7921e网卡， 5.19.2 全新编译

### (3) 详细日志

A：

closed 2022-08-20 by aiamadeus

wlshdjj, 2022-08-21:
据我所知官方并没采纳7921的AP补丁。你的补丁来源可能需要提供一下，配置加密前后的wireless配置文件（去掉你的密码或设为12345678）也需要提供一下。

littoy, 2022-08-21:
之前可能日志没贴被关了，编辑下，其中7c:04:d0:1e:a6:cb就无法获取到ip

littoy, 2022-08-21:
> 据我所知官方并没采纳7921的AP补丁。你的补丁来源可能需要提供一下，配置加密前后的wireless配置文件（去掉你的密码或设为12345678）也需要提供一下。

是用的master分支自带的mt7921e驱动，5.19内核。

不加密配置：
cat /etc/config/wireless

config wifi-device 'radio0'
	option type 'mac80211'
	option path '3c0800000.pcie/pci0002:00/0002:00:00.0/0002:01:00.0'
	option channel 'auto'
	option legacy_rates '0'
	option band '5g'
	option htmode 'HE80'
	option country 'CN'
	option mu_beamformer '0'

config wifi-iface 'default_radio0'
	option device 'radio0'
	option mode 'ap'
	option ssid 'R5S'
	option key 'A1234567'
	option network 'lan'
	option macfilter 'allow'
	option encryption 'none'

开启加密的配置：
 cat /etc/config/wireless

config wifi-device 'radio0'
	option type 'mac80211'
	option path '3c0800000.pcie/pci0002:00/0002:00:00.0/0002:01:00.0'
	option channel 'auto'
	option legacy_rates '0'
	option band '5g'
	option htmode 'HE80'
	option country 'CN'
	option mu_beamformer '0'

config wifi-iface 'default_radio0'
	option device 'radio0'
	option mode 'ap'
	option ssid 'R5S'
	option key 'A1234567'
	option network 'lan'
	option macfilter 'allow'
	option encryption 'psk-mixed'

wlshdjj, 2022-08-21:
配置没有问题，缺少hwmode项，不过翻看mac80211和hostapd都没有获取hwmode选项的地方，因此没有被利用到。mac80211.sh和hostapd.sh由于历史遗留问题，残留非常多的垃圾代码，看起来难以琢磨，用起来有没有问题也不知道。
不清楚配置无线后有无重启系统再试过？
获取不到IP，说明wlan0并没有加入到lan去，这里需要注意一点：openwrt官方master系列的network和luci1.x产生的配置不同，luci1.x的配置是interface 'lan'里面直接绑定ifname，而master却是每个port设置一个device作为一个配置项，再将其定向至br-lan桥接（类似于咱们添加虚拟ssid的过程）。lean使用的是1.x luci，你如果在luci上进行了配置，原配置格式就会被更改，配置是否不兼容？
这个报错疑似是这个问题：
Sun Aug 21 15:28:30 2022 daemon.notice hostapd: nl80211: Failed to remove interface wlan0 from bridge br-lan: Invalid argument

wlshdjj, 2022-08-21:
可能你还要提供network配置内容

littoy, 2022-08-21:
> 配置没有问题，缺少hwmode项，不过翻看mac80211和hostapd都没有获取hwmode选项的地方，因此没有被利用到。mac80211.sh和hostapd.sh由于历史遗留问题，残留非常多的垃圾代码，看起来难以琢磨，用起来有没有问题也不知道。 不清楚配置无线后有无重启系统再试过？ [...]

我试过多次，改配置有重启，也有没重启的，开启加密第一个链接的设备大概率能正常链接。第二个第三个不能正常链接，手动ip也不通，而且这种情况下，第一个已经连上的也会断网。

是用的lede master，通过luci配置的。以下是network:
config interface 'loopback'
	option ifname 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config globals 'globals'
	option ula_prefix 'fd30:e16e:e8a4::/48'

config interface 'lan'
	option type 'bridge'
	option proto 'static'
	option netmask '255.255.255.0'
	option ip6assign '60'
	option _orig_ifname 'eth1 eth2'
	option _orig_bridge 'true'
	option ipaddr '192.168.100.1'
	option ifname 'eth0 eth1'

config device 'lan_eth1_dev'
	option name 'eth1'
	option macaddr '42:cb:3f:32:18:cc'

config device 'lan_eth2_dev'
	option name 'eth2'
	option macaddr '42:cb:3f:32:18:cc'

config interface 'wan'
	option proto 'dhcp'
	option _orig_ifname 'eth0'
	option _orig_bridge 'false'
	option ifname 'eth2'

config device 'wan_eth0_dev'
	option name 'eth0'
	option macaddr '42:cb:3f:32:18:cb'

config interface 'VPN'
	option ifname 'ipsec0'
	option proto 'static'
	option ipaddr '10.10.10.1'
	option netmask '255.255.255.0'

wlshdjj, 2022-08-21:
上面这个配置很奇怪，既不是openwrt官方纯master产生的又不是lean适配luci1.x的config_generate产生的，而是两种混杂在一起，这会产生什么问题我不清楚，你使用了官方的config_generate，同时又使用了lean的luci来配置，乱了套了。暂时无解。我当时测试7921也没注意到加密情况。

littoy, 2022-08-21:
我这r5s三网口基本情况说下：eth2做wan，eth0和eth1 做lan桥接开dhcp，wan和lan全部放入了防火墙Lan区域做旁路由。
可否提供一下相对标准和配置做参考呢？🙏

wlshdjj, 2022-08-22:
为了验证你这个问题，我把家里使用的x86拆下来测试。现在遇到的问题比你说的还要怪异。当初测试MT7921时，HT/VHT/HE都测试了，但是没有加密，电脑手机都可以连接，还进行了测速。但是现在再试时，怪异事情来了：仅有HT/VHT的信号能被电脑搜索到，HE的电脑不行手机可以。再继续折腾，现在手机也搜不到。更别说加密的事，信号都搜不到。但是luci上显示信号是正常的。我见鬼了。前些天电脑还原过，但驱动还是之前的。mac80211和luci做过很多修改，现在我都搞不清是什么问题了。我使用的是官方源码，这不关lean的事，怪了去了。

wlshdjj, 2022-08-22:
这是我的wireless配置，现在找了两台手机测都可以秒连，电脑没办法测，intel9265这个怪胎

config wifi-device 'radio0'
	option type 'mac80211'
	option path 'pci0000:00/0000:00:1c.1/0000:02:00.0'
	option hwmode '11a'
	option band '5g'
	option mu_beamformer '1'
	option cell_density '0'
	option country 'CN'
	option txpower '20'
	option channel '36'
	option htmode 'HE80'
	option legacy_rates '0'

config wifi-iface 'default_radio0'
	option device 'radio0'
	option network 'lan'
	option mode 'ap'
	option ssid 'Openwrt'
	option key '12345678'
	option skip_inactivity_poll '0'
	option short_preamble '1'
	option encryption 'psk2'
	option disassoc_low_ack '1'

这是我的network配置，真没你的那些乱七八糟的东西：

config interface 'loopback'
	option ifname 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config globals 'globals'
	option ula_prefix 'fdf5:65f5:34e1::/48'
	option packet_steering '1'

config interface 'lan'
	option type 'bridge'
	list ifname 'eth0'
	option proto 'static'
	option netmask '255.255.255.0'
	option ip6assign '60'
	option ipaddr '192.168.18.1'

config interface 'wan'
	option ifname 'eth1'
	option proto 'dhcp'

config interface 'wan6'
	option ifname 'eth1'
	option proto 'dhcpv6'

fishg, 2022-08-23:
谢谢，我参照改个试试

wlshdjj, 2022-08-23:
再三验证，不存在这个问题，除了怪胎intel9625搜索不到信号外，手机都是秒连，联网正常。

littoy, 2022-08-23:
> 再三验证，不存在这个问题，除了怪胎intel9625搜索不到信号外，手机都是秒连，联网正常。

感谢，我再对照测试下。有没有可能是我的天线不行，7块的号称3db内置天线，不加密放旁边倒是是1080Mbps