coolsnowwolf/lede #9973: 请教大佬，SSR Plus +检测xtls节点可用，但是设置成主服务器后扶不了墙

q405812306, 2022-08-20:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：在同样的xtls信息配置下，之前在N1上面更新了一个版本后，n1也是的ssr+用xtls节点也是扶不了墙。。。现在想把树莓派当软路由，但是ssr+用xtls节点，扶不了墙，但是它自己检测节点的可用性又显示OK

[image]
[image]

### (2) 路由器型号和固件版本

A：
[image]

### (3) 详细日志

A：
2022-08-20 22:05:02: -----------end------------
2022-08-20 22:04:58: Main node:V2Ray 5.0.7 (V2Fly, a community-driven edition of V2Ray.) OpenWrt (go1.18.4 linux/arm64) Started!
2022-08-20 22:04:58: ----------start------------
2022-08-20 21:58:11: -----------end------------
2022-08-20 21:58:11: 禁止连接的域名加载完毕。
2022-08-20 21:58:11: 未启动主节点，禁止连接的域名正在加载。
2022-08-20 21:58:11: ----------start------------

minlang112, 2022-08-20:
你提错地方了→ https://github.com/fw876/helloworld

q405812306, 2022-08-20:
> 你提错地方了→ https://github.com/fw876/helloworld
好的，谢谢