coolsnowwolf/lede #9976: AX200 5GHz AP AX mode works, tildearrow's method.

ayyyuki1, 2022-08-21:
AX200 5GHz AP mode, tildearrow's method. No need for lar_disable!

https://tildearrow.org/?p=post&month=7&year=2022&item=lar

here it is with openwrt.

[image]

patch cleaned up for openwrt

[999-hostapd-2.10-lar.patch.txt](https://github.com/coolsnowwolf/lede/files/9388276/999-hostapd-2.10-lar.patch.txt)

ayyyuki1, 2022-08-21:
some more better tests

[image]

wlshdjj, 2022-08-21:
So, there must be another 5g AP that AX200 5g will work ?

ayyyuki1, 2022-08-21:
> So, there must be another 5g AP that AX200 5g will work ?

I think so since thats how it knows its own regulatory domain. It does 802.11d compliance this way.

aiamadeus, 2022-08-21:
Wow, so exciting, thank you so much for sharing this.

ayyyuki1, 2022-08-22:
> Wow, so exciting, thank you so much for sharing this.

I added 

> if (!iface->conf->noscan) { 
> //scan
> }

as you see this is to do the lar scan you have to do option noscan '0'
this will be handled better if lets say you make option lar_scan or something like that. 

also wise to add these radio options

>    option he_bss_color '8'
>    option he_su_beamformee '1'

from: https://forum.openwrt.org/t/802-11ax-worse-than-802-11ac-with-mt76-driver/126466/50

and it results to this
[image]

Would be nice to see if someone tests this too on their own especially on x86 builds.

wlshdjj, 2022-08-22:
I have no AX200 to test. All I want to do now is hope to make MT7921 work better.

ayyyuki1, 2022-08-22:
> I have no AX200 to test. All I want to do now is hope to make MT7921 work better.

I think AP mode on MT7921 and MT7921K already works.

also on another note, is MT7922 already available in china?

wlshdjj, 2022-08-22:
I haven't seen anyone selling the MT7922 yet.

aiamadeus, 2022-08-23:
> also on another note, is MT7922 already available in china?

It's available on the second-hand market, but it's rare and a little more expensive.

ayyyuki1, 2022-08-25:
Seems to me that you don't really need a 5GHz AP around to bring up 5GHz AP on this.

I hope somebody else test this.

@wlshdjj are you gonna build https://www.right.com.cn/FORUM/thread-928319-1-2.html with this patch? would be nice to see other people's test.

ayyyuki1, 2022-08-30:
seems this affects other people wiuth multiple radios or non-intel radios, this might be a better way of handling that issue

[999-hostapd-2.10-lar.patch.txt](https://github.com/coolsnowwolf/lede/files/9450357/999-hostapd-2.10-lar.patch.txt)

add this to radio config

`        option intel_lar '1'
        option channel '149'
        option noscan '0'
`

tildearrow, 2022-08-30:
Hello! Author of the original patch here.
Note that the way I implemented it is a hack which may not work properly sometimes.

SunSgr, 2022-09-07:
> 
Sure,I got the MT7922 two days ago, but I don't know how to drive it.

ayyyuki1, 2022-09-09:
> > 
> 
> Sure,I got the MT7922 two days ago, but I don't know how to drive it.

https://github.com/openwrt/openwrt/pull/10564

ayyyuki1, 2022-09-10:
@aiamadeus my theory goes like this

[999-hostapd-2.10-lar.patch.txt](https://github.com/coolsnowwolf/lede/files/9539763/999-hostapd-2.10-lar.patch.txt)

but I dont know the way on how to add it in uci/luci

ayyyuki1, 2022-09-19:
@aiamadeus @coolsnowwolf I think this is good enough.

closed 2022-09-19 by ayyyuki1

qq897042818, 2022-12-02:
> @aiamadeus @coolsnowwolf I think this is good enough.

> @aiamadeus @coolsnowwolf I think this is good enough.

> my theory on how to make this patch to not affect other radios (non intel 5GHz) goes like this
> 
> ```
> if (iface->conf->intel_lar && !iface->conf->noscan) {
>       do_lar_scan!
>  }
> ```
> 
> [999-hostapd-2.10-lar.patch.txt](https://github.com/coolsnowwolf/lede/files/9539763/999-hostapd-2.10-lar.patch.txt)
> 
> then add `option intel_lar '1'` on /etc/wireless/config then it should show up intel_lar=1 on /var/run/hostapd-phyX.conf
> 
> [image]
> 
> [image]
> 
> it already works on hostapd ~but I dont know the way on how to add it in uci/luci but I dont know the proper edit to be done on hostapd.sh and other scripts~
> 
> EDIT: I think I've got it
> 
> ```
> --- hostapd.sh.orig	2022-09-10 01:55:41.409228000 +0800
> +++ hostapd.sh	2022-09-10 16:32:39.349899173 +0800
> @@ -117,6 +117,7 @@
>  	config_add_int rssi_ignore_probe_request
>  	config_add_int maxassoc
>  	config_add_boolean vendor_vht
> +	config_add_boolean intel_lar
>  
>  	config_add_string acs_chan_bias
>  	config_add_array hostapd_options
> @@ -134,7 +135,7 @@
>  
>  	json_get_vars country country3 country_ie beacon_int:100 dtim_period:2 doth require_mode legacy_rates \
>  		acs_chan_bias local_pwr_constraint spectrum_mgmt_required airtime_mode cell_density \
> -		rts_threshold beacon_rate rssi_reject_assoc_rssi rssi_ignore_probe_request maxassoc vendor_vht
> +		rts_threshold beacon_rate rssi_reject_assoc_rssi rssi_ignore_probe_request maxassoc vendor_vht intel_lar
>  
>  	hostapd_set_log_options base_cfg
>  
> @@ -216,6 +217,7 @@
>  				set_default rate_list "24000 36000 48000 54000"
>  				set_default basic_rate_list "24000"
>  			fi
> +			[ -n "$intel_lar" ] && append base_cfg "intel_lar=$intel_lar" "$N"
>  		;;
>  	esac
>  
> ```
> 
> so now in radio config you have to
> 
> ```
> option intel_lar '1'
> option noscan '0'
> option channel '149'
> option country 'XX'
> ```
> 
> so you can operate this.
> 
> Also yep, It does work even if theres no other 5GHz radio broadcasting around. Just set the country properly. (It seems to accept the hint.) And even if theres other APs around broadcasting the wrong country, channel 149 is always available on all countries except Qatar. Feel free to try ;)

> option noscan '0'

I need some help . maybe my steps were wrong ,please see my steps in word file .
please help me, tks!
[my wrong steps.docx](https://github.com/coolsnowwolf/lede/files/10140840/my.wrong.steps.docx)

backslashxx, 2022-12-03:
> ### I need some help . maybe my steps were wrong ,please see my steps in word file .
> please help me, tks! [my wrong steps.docx](https://github.com/coolsnowwolf/lede/files/10140840/my.wrong.steps.docx)

you need to add the hostapd.sh edit

qq897042818, 2022-12-04:
I have edit /hostapd/files/hostapd.sh, but still phyX self-managed country is CN,Any other comment ?
[hostapd.zip](https://github.com/coolsnowwolf/lede/files/10147950/hostapd.zip)

backslashxx, 2022-12-04:
@qq897042818 seems about right, cant open an AP on channel 149~161?

qq897042818, 2022-12-04:
@backslashxx Yes, only 2.4g 144M is available now

06:1c.0 Network controller: Intel Corporation Wi-Fi 6 AX210/AX211/AX411 160MHz (rev 1a)
        Subsystem: Intel Corporation Wi-Fi 6 AX210 160MHz
        Kernel driver in use: iwlwifi
09:01.0 SCSI storage controller: Red Hat, Inc. Virtio SCSI
        Subsystem: Red Hat, Inc. Device 0008
        Kernel driver in use: virtio-pci
root@OpenWrt:~#
root@OpenWrt:~# iw reg get
global
country DE: DFS-ETSI
        (2400 - 2483 @ 40), (N/A, 20), (N/A)
        (5150 - 5250 @ 80), (N/A, 23), (N/A), NO-OUTDOOR, AUTO-BW
        (5250 - 5350 @ 80), (N/A, 20), (0 ms), NO-OUTDOOR, DFS, AUTO-BW
        (5470 - 5725 @ 160), (N/A, 26), (0 ms), DFS
        (5725 - 5875 @ 80), (N/A, 13), (N/A)
        (5945 - 6425 @ 160), (N/A, 23), (N/A), NO-OUTDOOR
        (57000 - 66000 @ 2160), (N/A, 40), (N/A)

phy#0 (self-managed)
country CN: DFS-UNSET
        (2402 - 2437 @ 40), (6, 22), (N/A), AUTO-BW, NO-HT40MINUS, NO-80MHZ, NO-160MHZ
        (2422 - 2462 @ 40), (6, 22), (N/A), AUTO-BW, NO-80MHZ, NO-160MHZ
        (2447 - 2482 @ 40), (6, 22), (N/A), AUTO-BW, NO-HT40PLUS, NO-80MHZ, NO-160MHZ
        (5170 - 5190 @ 160), (6, 22), (N/A), NO-OUTDOOR, AUTO-BW, IR-CONCURRENT, NO-HT40MINUS, PASSIVE-SCAN
        (5190 - 5210 @ 160), (6, 22), (N/A), NO-OUTDOOR, AUTO-BW, IR-CONCURRENT, NO-HT40PLUS, PASSIVE-SCAN
        (5210 - 5230 @ 160), (6, 22), (N/A), NO-OUTDOOR, AUTO-BW, IR-CONCURRENT, NO-HT40MINUS, PASSIVE-SCAN
        (5230 - 5250 @ 160), (6, 22), (N/A), NO-OUTDOOR, AUTO-BW, IR-CONCURRENT, NO-HT40PLUS, PASSIVE-SCAN
        (5250 - 5270 @ 160), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40MINUS, PASSIVE-SCAN
        (5270 - 5290 @ 160), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40PLUS, PASSIVE-SCAN
        (5290 - 5310 @ 160), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40MINUS, PASSIVE-SCAN
        (5310 - 5330 @ 160), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40PLUS, PASSIVE-SCAN
        (5490 - 5510 @ 240), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40MINUS, PASSIVE-SCAN
        (5510 - 5530 @ 240), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40PLUS, PASSIVE-SCAN
        (5530 - 5550 @ 240), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40MINUS, PASSIVE-SCAN
        (5550 - 5570 @ 240), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40PLUS, PASSIVE-SCAN
        (5570 - 5590 @ 240), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40MINUS, PASSIVE-SCAN
        (5590 - 5610 @ 240), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40PLUS, PASSIVE-SCAN
        (5610 - 5630 @ 240), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40MINUS, PASSIVE-SCAN
        (5630 - 5650 @ 240), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40PLUS, PASSIVE-SCAN
        (5650 - 5670 @ 80), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40MINUS, NO-160MHZ, PASSIVE-SCAN
        (5670 - 5690 @ 80), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40PLUS, NO-160MHZ, PASSIVE-SCAN
        (5690 - 5710 @ 80), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40MINUS, NO-160MHZ, PASSIVE-SCAN
        (5710 - 5730 @ 80), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40PLUS, NO-160MHZ, PASSIVE-SCAN
        (5735 - 5755 @ 80), (6, 22), (N/A), AUTO-BW, IR-CONCURRENT, NO-HT40MINUS, NO-160MHZ, PASSIVE-SCAN
        (5755 - 5775 @ 80), (6, 22), (N/A), AUTO-BW, IR-CONCURRENT, NO-HT40PLUS, NO-160MHZ, PASSIVE-SCAN
        (5775 - 5795 @ 80), (6, 22), (N/A), AUTO-BW, IR-CONCURRENT, NO-HT40MINUS, NO-160MHZ, PASSIVE-SCAN
        (5795 - 5815 @ 80), (6, 22), (N/A), AUTO-BW, IR-CONCURRENT, NO-HT40PLUS, NO-160MHZ, PASSIVE-SCAN
        (5815 - 5835 @ 40), (6, 22), (N/A), AUTO-BW, IR-CONCURRENT, NO-HT40MINUS, NO-80MHZ, NO-160MHZ, PASSIVE-SCAN

root@OpenWrt:~#

backslashxx, 2022-12-04:
> @backslashxx Yes, only 2.4g 144M is available now
> 
> [image]

show me "iw phy"

qq897042818, 2022-12-04:
> 

[hostapd-phy0.conf.txt](https://github.com/coolsnowwolf/lede/files/10149033/hostapd-phy0.conf.txt)
[iwphy.txt](https://github.com/coolsnowwolf/lede/files/10149034/iwphy.txt)
[wireless.txt](https://github.com/coolsnowwolf/lede/files/10149035/wireless.txt)
[dmesg.txt](https://github.com/coolsnowwolf/lede/files/10149036/dmesg.txt)

backslashxx, 2022-12-05:
> [hostapd-phy0.conf.txt](https://github.com/coolsnowwolf/lede/files/10149033/hostapd-phy0.conf.txt) [iwphy.txt](https://github.com/coolsnowwolf/lede/files/10149034/iwphy.txt) [...]

heres the issue

			* 5180 MHz [36] (22.0 dBm) (no IR)
			* 5200 MHz [40] (22.0 dBm) (no IR)
			* 5220 MHz [44] (22.0 dBm) (no IR)
			* 5240 MHz [48] (22.0 dBm) (no IR)
			* 5260 MHz [52] (22.0 dBm) (no IR, radar detection)
			* 5280 MHz [56] (22.0 dBm) (no IR, radar detection)
			* 5300 MHz [60] (22.0 dBm) (no IR, radar detection)
			* 5320 MHz [64] (22.0 dBm) (no IR, radar detection)
			* 5340 MHz [68] (disabled)
			* 5360 MHz [72] (disabled)
			* 5380 MHz [76] (disabled)
			* 5400 MHz [80] (disabled)
			* 5420 MHz [84] (disabled)
			* 5440 MHz [88] (disabled)
			* 5460 MHz [92] (disabled)
			* 5480 MHz [96] (disabled)
			* 5500 MHz [100] (22.0 dBm) (no IR, radar detection)
			* 5520 MHz [104] (22.0 dBm) (no IR, radar detection)
			* 5540 MHz [108] (22.0 dBm) (no IR, radar detection)
			* 5560 MHz [112] (22.0 dBm) (no IR, radar detection)
			* 5580 MHz [116] (22.0 dBm) (no IR, radar detection)
			* 5600 MHz [120] (22.0 dBm) (no IR, radar detection)
			* 5620 MHz [124] (22.0 dBm) (no IR, radar detection)
			* 5640 MHz [128] (22.0 dBm) (no IR, radar detection)
			* 5660 MHz [132] (22.0 dBm) (no IR, radar detection)
			* 5680 MHz [136] (22.0 dBm) (no IR, radar detection)
			* 5700 MHz [140] (22.0 dBm) (no IR, radar detection)
			* 5720 MHz [144] (22.0 dBm) (no IR, radar detection)
			* 5745 MHz [149] (22.0 dBm) (no IR)
			* 5765 MHz [153] (22.0 dBm) (no IR)
			* 5785 MHz [157] (22.0 dBm) (no IR)
			* 5805 MHz [161] (22.0 dBm) (no IR)
			* 5825 MHz [165] (22.0 dBm) (no IR)
			* 5845 MHz [169] (disabled)
			* 5865 MHz [173] (disabled)
			* 5885 MHz [177] (disabled)
			* 5905 MHz [181] (disabled)
			
			
			
so china is no IR all 5GHz channels? hmmm
try this, 

iw dev phy0 scan && iw phy, tell me if any 5Ghz channels have no IR after that

qq897042818, 2022-12-05:
> 
all 5g channels are disable or no-IR in china 
[iw scan.txt](https://github.com/coolsnowwolf/lede/files/10151028/iw.scan.txt)

Github-Newbie5432, 2023-01-10:
Forgive my ignorance .... I am primarily a Windows user! :P

I looked into this sometime ago and am pleased to see that some progress has been made since then in getting 5Ghz AP mode "sort of" working at least ....

As someone else mentioned above, in Windows 11 you can share your internet connection over WiFi as a hotspot.

I tried this last night on my 12th Gen Intel NUC which has the Intel AX211 wireless card.

In the hotspot settings in Windows it allows you to select either 2.4Ghz or 5Ghz.

I selected 5Ghz and sure enough a WiFi 6 5Ghz network appeared that I could connect to.

That would suggest to me (and I apologise if I've missed something) that the hardware is capable of being a 5Ghz AP

Why would it work in Windows but then it not be possible to do the same under Linux?

Would be awesome if we could make access points out of these WiFi 6 / 6E cards, they're pretty affordable after all!

Or is the Windows "Mobile Hotspot" implementation still with the same limitations outlined above that are on Linux ie. No DFS etc. ?

tildearrow, 2023-01-10:
> Forgive my ignorance .... I am primarily a Windows user! :P
>
> I looked into this sometime ago and am pleased to see that some progress has been made since then in getting 5Ghz AP mode "sort of" [...]

Were there any 5GHz networks nearby? If so, can you try again under Windows on a 5GHz-less area and see whether the 5GHz option still exists?

Github-Newbie5432, 2023-01-10:
> Were there any 5GHz networks nearby? If so, can you try again under Windows on a 5GHz-less area and see whether the 5GHz option still exists?

Yes there were 5Ghz networks nearby - unfortunately seeing as I live in a built-up suburb I don't have a "5Ghz-less area" to try (where I would have a computer available to test!)

Maybe that's all

xpratg, 2023-08-25:
Hi thanks for showing us. Can you help me to implement this change to an OpenWRT x86 instance? I`m trying to use an ax200 card as an AP in a proxmox OpenWRT vm.

Thanks in advance!

RSSYLY, 2025-12-31:
> 小白来报道。AX211求大神指点。openwrt N100设备

无法使用的，这个方法是通过强制开启位置感知扫描，从而在合规的地区开启5Ghz，而英特尔的规则中中国大陆不允许5Ghz热点（来自一个愤怒的、2年前购买了板载AX211网卡的主板的用户