coolsnowwolf/lede #9977: pppd 会crash kernel @5.19

GreatMichaelLee, 2022-08-21:
内核日志如下

Sun Aug 21 10:40:54 2022 user.info mwan3track[10049]: Check (ping) failed for target "223.5.5.5" on interface vwan4 (pppoe-vwan4)
Sun Aug 21 10:40:56 2022 daemon.info pppd[16134]: LCP terminated by peer
Sun Aug 21 10:40:56 2022 daemon.info pppd[16134]: Connect time 990.2 minutes.
Sun Aug 21 10:40:56 2022 daemon.info pppd[16134]: Sent 1301830369 bytes, received 2410974510 bytes.
Sun Aug 21 10:40:56 2022 kern.err kernel: [59514.636893] rcu: INFO: rcu_preempt detected expedited stalls on CPUs/tasks: { 3-... } 3 jiffies s: 1161 root: 0x8/.
Sun Aug 21 10:40:56 2022 kern.err kernel: [59514.638220] rcu: blocking rcu_node structures (internal RCU debug):
Sun Aug 21 10:40:56 2022 kern.info kernel: [59514.638900] Task dump for CPU 3:
Sun Aug 21 10:40:56 2022 kern.info kernel: [59514.639359] task:pppd            state:R  running task     stack:    0 pid:24651 ppid: 16134 flags:0x00004008
Sun Aug 21 10:40:56 2022 kern.info kernel: [59514.640374] Call Trace:
Sun Aug 21 10:40:56 2022 kern.info kernel: [59514.640845]  <TASK>
Sun Aug 21 10:40:56 2022 kern.info kernel: [59514.641243]  ? .m.sve.%i.^G..^Mr.t+0x1f/0x30
Sun Aug 21 10:40:56 2022 kern.info kernel: [59514.641771]  ? lz._..+0x4be/0x970
Sun Aug 21 10:40:56 2022 kern.info kernel: [59514.642270]  ? lz._..+0x5d5/0x970
Sun Aug 21 10:40:56 2022 kern.info kernel: [59514.642781]  ? xz_..lz.2_r.+0x6e1/0x9a0
Sun Aug 21 10:40:56 2022 kern.info kernel: [59514.643341]  ? .l...+0x46/0x90
Sun Aug 21 10:40:56 2022 kern.info kernel: [59514.643812]  ? xz_..r.+0x6e0/0x9e0
Sun Aug 21 10:40:56 2022 kern.info kernel: [59514.644282]  ? ....f..e+0xa0/0xc0
Sun Aug 21 10:40:56 2022 kern.info kernel: [59514.644717]  ? . ..+0x8e/0xa0
Sun Aug 21 10:40:56 2022 kern.info kernel: [59514.645119]  ? ^K. ..+0x65/0xa0
Sun Aug 21 10:40:56 2022 kern.info kernel: [59514.645529]  ? ..h..~d]+0xf6/0x5d0
Sun Aug 21 10:40:56 2022 kern.info kernel: [59514.645960]  ? ..h..p.d]+0x18/0x40
Sun Aug 21 10:40:56 2022 kern.info kernel: [59514.646400]  ? ..=.f/.+0x103/0x480
Sun Aug 21 10:40:56 2022 kern.info kernel: [59514.646918]  ? .p..f...^L.+0x4be/0x9c0
Sun Aug 21 10:40:56 2022 daemon.notice netifd: Interface 'vwan4' has lost the connection
Sun Aug 21 10:40:56 2022 kern.info kernel: [59514.647457]  ? .p..f...^L.+0x4be/0x9c0
Sun Aug 21 10:40:56 2022 kern.info kernel: [59514.648029]  ? .p..f...^L.+0x4be/0x9c0
Sun Aug 21 10:40:56 2022 kern.info kernel: [59514.648606]  ? .w.p._....q..o.+0xd/0x40
Sun Aug 21 10:40:56 2022 kern.info kernel: [59514.649140]  ? ..new..e+0x17e/0x1a0
Sun Aug 21 10:40:56 2022 kern.info kernel: [59514.649664]  ? ..new..e+0x17e/0x1a0
Sun Aug 21 10:40:56 2022 kern.info kernel: [59514.650167]  ? ..new..e+0x17e/0x1a0
Sun Aug 21 10:40:56 2022 kern.info kernel: [59514.650677]  ? .p..f...^L.+0x4be/0x9c0
Sun Aug 21 10:40:56 2022 kern.info kernel: [59514.651231]  ? .p..f...^L.+0x4be/0x9c0
Sun Aug 21 10:40:56 2022 kern.info kernel: [59514.651930]  ? k..c.h...+0x36/0x460
Sun Aug 21 10:40:56 2022 kern.info kernel: [59514.652611]  ? v.a.a_d.+0x24/0xb0
Sun Aug 21 10:40:56 2022 kern.info kernel: [59514.653197]  ? l..tl.f.c+0x113/0x140
Sun Aug 21 10:40:56 2022 kern.info kernel: [59514.653808]  ? v...._v..+0xa0/0xa0
Sun Aug 21 10:40:56 2022 kern.info kernel: [59514.654375]  ? m.c..f.c.._m...d+0x12a/0x2c0
Sun Aug 21 10:40:56 2022 kern.info kernel: [59514.654961]  ? .,...ty+0x166/0x810
Sun Aug 21 10:40:56 2022 kern.info kernel: [59514.655527]  ? ew..b..ce^F149+0x274/0x330
Sun Aug 21 10:40:56 2022 kern.info kernel: [59514.656101]  ? ..$u.+0x1b9/0x4f0
Sun Aug 21 10:40:56 2022 kern.info kernel: [59514.656603]  ? .$u.+0x50/0xa0
Sun Aug 21 10:40:56 2022 kern.info kernel: [59514.657095]  ? d.[.+0x13f/0x2e0
Sun Aug 21 10:40:56 2022 kern.info kernel: [59514.657578]  ? k.ne.[.4+0x96/0x120
Sun Aug 21 10:40:56 2022 kern.info kernel: [59514.658102]  ? .k..pe...+0x50/0x50
Sun Aug 21 10:40:56 2022 kern.info kernel: [59514.658620]  ? do..[.4+0x6f/0x90
Sun Aug 21 10:40:56 2022 kern.info kernel: [59514.659109]  ? .q.....^\....+0xd/0x30
Sun Aug 21 10:40:56 2022 kern.info kernel: [59514.659621]  ? .q.....+0x3f/0x50
Sun Aug 21 10:40:56 2022 kern.info kernel: [59514.660101]  ? .c...fa^^t+0x270/0x560
Sun Aug 21 10:40:56 2022 kern.info kernel: [59514.660592]  ? x^H..[.4+0x1d/0x30
Sun Aug 21 10:40:56 2022 kern.info kernel: [59514.661003]  ? do.sc..^H+0x43/0xc0
Sun Aug 21 10:40:56 2022 kern.info kernel: [59514.661477]  ? ...SYSCALL_6.aft.hwf..+0x63/0xcd
Sun Aug 21 10:40:56 2022 kern.info kernel: [59514.662172]  </TASK>
Sun Aug 21 10:40:56 2022 daemon.notice netifd: Network device 'pppoe-vwan4' link is down

GreatMichaelLee, 2022-08-22:
[   99.251488] device eth0 entered promiscuous mode
[   99.465751] device eth1 entered promiscuous mode
[   99.767671] pppoe-vwan1: renamed from ppp0
[   99.855345] pppoe-vwan3: renamed from ppp1
[   99.947816] pppoe-vwan2: renamed from ppp2
[  100.039124] pppoe-vwan4: renamed from ppp3
[  192.565409] ------------[ cut here ]------------
[  192.566030] WARNING: CPU: 2 PID: 2152 at k?e+0x456/0x550
[  192.566610] Modules linked in: ksmbd xt_FULLCONENAT pppoe ppp_async wireguard snd_usb_audio snd_hda_intel snd_hda_codec_via snd_hda_codec_realtek snd_hda_codec_hdmi snd_hda_codec_generic snd_hda_codec pppox ppp_mppe ppp_generic libchacha20poly1305 iptable_nat ipt_REJECT curve25519_x86_64 chacha_x86_64 ax88179_178a zstd xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_socket xt_recent xt_quota xt_policy xt_pkttype xt_owner xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_iprange xt_hl xt_helper xt_geoip xt_esp xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connlabel xt_connbytes xt_comment xt_cgroup xt_addrtype xt_TPROXY xt_TCPMSS xt_REDIRECT xt_MASQUERADE xt_LUA xt_LOG xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CT xt_CLASSIFY via_velocity usbnet usblp ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda tulip ts_fsm ts_bm tcp_bbr snd_usbmidi_lib snd_intel_dspcfg snd_hda_core slhc sch_cake rtl8150 regmap_core r8168
[  192.566674]  r8152 r8125 poly1305_x86_64 pcnet32 nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nft_reject nft_redir nft_quota nft_objref nft_numgen nft_meta_bridge nft_log nft_limit nft_hash nft_fwd_netdev nft_dup_netdev nft_ct nf_tproxy_ipv6 nf_tproxy_ipv4 nf_tables nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_nat nf_flow_table nf_dup_netdev nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast ts_kmp nf_conntrack_amanda nf_conncount mlx5_core mlx4_en mlx4_core macvlan lzo libcurve25519_generic libchacha iptable_raw iptable_mangle iptable_filter ipt_ah ipt_ECN ip_tables igc iavf i915 i40e forcedeth e1000e drm_display_helper drm_buddy crc_ccitt br_netfilter bnx2x bnx2 alx 8139too 8139cp act_connmark nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 sch_tbf
[  192.575338]  sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact configs evdev i2c_dev ledtrig_audio cryptodev xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 nfsv4 nfsd nfs bonding ip_gre gre snd_via82xx snd_intel8x0 snd_mpu401_uart ixgbe igbvf e1000 amd_xgbe mdio_devres snd_ac97_codec ac97_bus ifb dummy ip6_udp_tunnel udp_tunnel mdio ipcomp6 xfrm6_tunnel esp6 ah6 xfrm4_tunnel ipcomp esp4 ah4 tunnel6 tunnel4 ip_tunnel rpcsec_gss_krb5 auth_rpcgss veth tun snd_rawmidi snd_seq_device snd_pcm_oss snd_mixer_oss snd_hwdep snd_compress snd_pcm snd_timer snd soundcore xfrm_user
[  192.584387]  xfrm_ipcomp af_key xfrm_algo ntfs3 ntfs lockd sunrpc grace cifs oid_registry cifs_md4 cifs_arc4 asn1_decoder dns_resolver dm_mirror dm_region_hash dm_log dm_crypt dm_mod dax nls_utf8 ena crypto_user algif_skcipher algif_rng algif_hash algif_aead af_alg sha512_ssse3 sha512_generic sha256_ssse3 sha1_ssse3 sha1_generic seqiv jitterentropy_rng drbg md5 md4 hmac echainiv des_generic libdes deflate cts cmac authenc arc4 crypto_acompress nls_iso8859_1 nls_cp437 vfat fat uas sdhci_pltfm igb exfat btrfs zstd_decompress zstd_compress xxhash xor raid6_pq lzo_decompress lzo_compress libcrc32c button_hotplug ptp pps_core mii libphy tpm cbc encrypted_keys trusted [last unloaded: ksmbd]
[  192.602386] CPU: 2 PID: 2152 Comm: kworker/2:3 Not tainted 5.19.3 #0
[  192.603581] Hardware name: QEMU Standard PC (Q35 + ICH9, 2009), BIOS 0.0.0 02/06/2015
[  192.604820] Workqueue: events ??w?k
[  192.605786] RIP: 0010:k?e+0x456/0x550
[  192.606782] Code: ce 48 85 c0 0f 84 a1 00 00 00 0f b6 d2 48 c1 e2 04 48 8b 04 10 48 83 e0 e0 4a 8d 3c 20 44 89 ee e8 9f fb fe ff e9 40 fe ff ff <0f> 0b 45 31 ed 80 3d d2 75 0d 01 00 ba 00 f0 ff ff 0f 85 c1 fe ff
[  192.609663] RSP: 0018:ffffc9000028bde0 EFLAGS: 00010246
[  192.610710] RAX: 0004500000000000 RBX: ffff8881157473e8 RCX: 0000000000000220
[  192.611884] RDX: ffffffffffffffff RSI: 000000000455d1c0 RDI: ffff8881157473e8
[  192.613204] RBP: ffffc9000028be10 R08: ffff88817bd27560 R09: ffffffff82308d00
[  192.614515] R10: 0000000000000000 R11: 0000000000000980 R12: ffff88817e35d1c0
[  192.615842] R13: 0000000000000000 R14: 0000000000000000 R15: ffff8881957473e8
[  192.617193] FS:  0000000000000000(0000) GS:ffff88817bd00000(0000) knlGS:0000000000000000
[  192.618657] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  192.619892] CR2: 00007f2bea4dd900 CR3: 000000000420a000 CR4: 0000000000350ee0
[  192.621274] Call Trace:
[  192.622219]  <TASK>
[  192.623347]  ? ????f?e+0xa0/0xc0
[  192.624445]  kv?e+0x29/0x30
[  192.625454]  v?+0x328/0x380
[  192.626447]  ??w?k+0x24/0x40
[  192.627501]  ?c????w?k+0x202/0x370
[... cut ...]
[  192.783868] RSP: 002b:00007fffe239d920 EFLAGS: 00000246 ORIG_RAX: 000000000000003b
[  192.786504] RAX: ffffffffffffffda RBX: 00007f5cc7bcaf40 RCX: 00007f5cc7c67d34
[  192.787390] RDX: 00007f5cc7be87e8 RSI: 00007f5cc7bcaf40 RDI: 00007f5cc7be8c90
[  192.788241] RBP: 00007f5cc7be8c90 R08: fefefefefefefeff R09: 8080808080808080
[  192.789177] R10: 0000000000000000 R11: 0000000000000246 R12: 00007f5cc7be87e8
[  192.790216] R13: 00000000ffffffff R14: 00007f5cc7be87e8 R15: 0000000000000000
[  192.791181]  </TASK>
[  192.791735] Disabling lock debugging due to kernel taint

GreatMichaelLee, 2022-08-23:
网上查了一下 544 是512+32， 32是有内核污染，表示有bad page(内核日志也提示有Bad page state in process dynamic_dns_upd pfn:115747）具体原因未知，看文档说是可能有加载第三方驱动造成的? 关键我这没加载其他啥....都是仓库编译的东西，哪位大佬给看看？

root@OWRT:~# sysctl -a|grep tain
kernel.tainted = 544

shu0802, 2022-08-24:
我这个是不是没有安装ipv6的原因
[image]

GreatMichaelLee, 2022-08-25:
> 我这个是不是没有安装ipv6的原因 [image]

我的也有这个error, 我也没装iptables v6,应该是这个原因，这个忽略就好，关系不大。

GreatMichaelLee, 2022-08-27:
今天的代码是修复了这个问题吗，L大？@coolsnowwolf

GreatMichaelLee, 2022-08-27:
刚用最新的代码试了一下，启动时内核还是会报一些类似内核污染的日志，我也不知道是不是问题

[   33.959973] br-lan: port 1(eth2) entered disabled state
[   33.960746] device eth2 entered promiscuous mode
[   34.470981] rcu: INFO: rcu_preempt detected expedited stalls on CPUs/tasks: { 0-... } 3 jiffies s: 77 root: 0x1/.
[   34.472920] rcu: blocking rcu_node structures (internal RCU debug):
[   34.474051] Task dump for CPU 0:
[   34.474588] task:procd           state:R  running task     stack:    0 pid: 5842 ppid:     1 flags:0x00004008
[   34.476067] Call Trace:
[   34.476680]  <TASK>
[   34.477106]  ? lz?_??+0x243/0x970
[   34.477619]  ? a?c?+0x234/0x3e0
[   34.478164]  ? ???mc?lruv???+0x44/0x80
[   34.479438]  ? ???$u???o>+0x1d/0x50
[   34.480022]  ? ?ubm?b?+0xba/0x130
[   34.480648]  ? xz_??lz?2_r?+0x669/0x9a0
[   34.481291]  ? xz_??bcj_r?+0x73/0x250
[   34.481807]  ? xz_??r?+0x32a/0x9e0
[   34.482352]  ? ??h?xz_??s+0x95/0x220
[   34.482850]  ? ??h???s+0x3c/0x70
[   34.483374]  ? ??h??~d]+0xd1/0x5d0
[   34.483935]  ? [k??+0x12/0x20
[   34.484404]  ? ??h?c?h?g?+0x12a/0x2c0
[   34.484899]  ? d?pl???i?+0x1d/0x30
[   34.485438]  ? ??h??ok?+0x31d/0x370
[   34.485895]  ? ??h??f?gm?t+0x20/0x30
[   34.486354]  ? ??h??~f/?+0x94/0x680
[   34.486799]  ? ??lruv???+0x3d/0x50
[   34.487262]  ? ??=??f/?+0x1ef/0x340
[   34.487716]  ? ??=?~f/?78+0x28/0x1e0
[   34.490538]  ? /??lru+0x2d/0x60
[   34.491035]  ? ??=?p??+0x38a/0x580
[   34.491495]  ? [k??+0x12/0x20
[   34.491912]  ? ??=??+0xc5/0x340
[   34.492345]  ? v՟?_????10+0x7c/0xd0
[   34.492842]  ? mn?u?n??pi?+0x3d/0x220
[   34.493315]  ? v՟?+0x69/0xc0
[   34.493730]  ? mn?>+0x23/0x40
[   34.494149]  ? ?m?w?k+0xe3/0xf0
[   34.494588]  ? g??i????~??+0xe6/0x160
[   34.495079]  ? d?????v_?+0x146/0x170
[   34.495547]  ? d?????+0x7b/0x1c0
[   34.495976]  ? v?????+0x18/0x30
[   34.496406]  ? v??~??+0x19e/0x220
[   34.496861]  ? ew?n???+0xe7/0x150
[   34.497294]  ? v???+0x111/0x160
[   34.497709]  ? ksy???+0x51/0xd0
[   34.498124]  ? x????+0x19/0x20
[   34.498535]  ? do?sc??+0x43/0xc0
[   34.498970]  ? ???SYSCALL_6?aft?hwf??+0x63/0xcd
[   34.499482]  </TASK>
[   34.781126] device br-lan entered promiscuous mode
[   37.771427] igc 0000:01:00.0 eth0: NIC Link is Up 2500 Mbps Full Duplex, Flow Control: RX/TX
[   37.801410] igc 0000:03:00.0 eth2: NIC Link is Up 2500 Mbps Full Duplex, Flow Control: RX/TX
[   37.847251] br-lan: port 1(eth2) entered blocking state
[   37.848003] br-lan: port 1(eth2) entered listening state
[   38.141401] igc 0000:02:00.0 eth1: NIC Link is Up 2500 Mbps Full Duplex, Flow Control: RX/TX
[   39.890967] br-lan: port 1(eth2) entered learning state

shu0802, 2022-08-28:
做个减法，先默认编译，一个插件一个插件的加，看看是哪个插件的原因

moonerick, 2022-10-24:
解决了么，我目前使用5.15没有问题，但是测试5.19和6.0都有这个问题

closed 2026-09-05 by coolsnowwolf