coolsnowwolf/lede #9978: x86云编译报错

binge8, 2022-08-21:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：x86云编译报错

### (2) 路由器型号和固件版本

A：最新源码，x86

### (3) 详细日志

A：看截图
[image]

ecrasy, 2022-08-21:
configure: error: Cross compiling required --hot=HOST_TUPLE and --build=ARCH

binge8, 2022-08-21:
> configure: error: Cross compiling required --hot=HOST_TUPLE and --build=ARCH

感谢，但是不大懂怎么搞啊

echoboss, 2022-08-21:
解决了吗？我也遇到了。

echoboss, 2022-08-21:
> 

> > configure: error: Cross compiling required --hot=HOST_TUPLE and --build=ARCH
> 
> 感谢，但是不大懂怎么搞啊

解决了吗？我也遇到了。

ecrasy, 2022-08-22:
> > configure: error: Cross compiling required --hot=HOST_TUPLE and --build=ARCH
> 
> 感谢，但是不大懂怎么搞啊

不清楚你的配置config是怎么生成的
无法判断你的问题出在哪里
log显示你的配置编译选项出错了
我今天的x86云编译成功
没发现你出现的问题

xiangfeidexiaohuo, 2022-08-26:
python安装版本太高，我实体机Ubuntu  20.04，里面的python是3.8.10，编译不会出现这个错误。
但是云编译，只是apt-get install -y python3，就默认安装最新的3.9.xx了，所以云编译脚本改下， 指定安装py3.8就成。

ookinasiawase, 2022-10-21:
我也是这个问题，解决了吗？求解答

closed 2026-09-05 by coolsnowwolf