coolsnowwolf/lede #9981: 小米路由3启动不了

yuzhe1129, 2022-08-21:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
先刷了别人编译带pbboot的固件，然后在升级编译的固件，可以启动，但是WAN口失效的，没无线，
请问下百兆的R3，这个固件是要修改什么编译的吗？怎么我编译WAN口失效 没无线的，

### (2) 路由器型号和固件版本

小米R3

### (3) 详细日志
PandoraBox-Boot Version 2.1
Build:Mar 17 2▒▒▒▒▒▒▒▒▒:13
SoC:MT7620A
Memory-Testing...131072K OK
MT7620:CPU clock set to 580Mhz
PCIe:reset controller
PB-Boot running at 0x87FA4000!
NAND: F59L1G▒▒▒▒▒▒▒▒iB
▒▒▒ng internal default config
Board:X▒▒▒▒▒ҳ
PandoraBox GPIO subsystem init
PandoraBox MT762x GPIO driver initiali▒▒䮊▒▒▒▒O_Mode_REG: 0x0012391C
PandoraBox LED subsystem init
LED [Power] regis▒▒▒▒䊍▒andoraBox Button subsystem init
Button [Reset] registered
MT76▒ؠ▒▒נ▒nitialized                                                          0
Booting from Fla▒议▒▒▒▒▒oting image from bc200000 ...
   Image Name:   MIPS OpenWrt Linux-5.4.86
   Image Type▒▒▒▒▒▒▒ӠLinux Kernel Image (lzma compressed)
   Data Size:    2120213 Bytes = ▒▒▒▒�▒▒▒ Load Address: 80000000
   Entry Point:  80000000
   Veri▒▒▒▒▒▒▒▒cksum ... OK
   Uncompr▒▒▒▒▒▒▒ernel Image ... ▒ˊ▒   Flushing cache from 0x80000000 size 66625                                                                                                                                                             48 ... OK
[    0.000000] Linux versi▒▒▒▒▒▒ (zmf@zmf-virtual-machine) (gcc version 8.4.0 (                                                                                                                                                             OpenWrt GCC 8.4.0 r3019-1▒巳▒䩩 #0 Sun Jan 17 03:10:55 2021
[    0.000000] Board has DDR2
[    0.0000▒▒ݠ▒▒▒▒▒g PMU set to hw control
[    0.000000] Digital PMU set to hw control
[▒▒▒▒▒▒▒▒0000] SoC Type: MediaTek MT7620A ver:2 eco:6
[    0.000000] printk: boo▒▒▒▒▒▒▒▒[early0] enabled
[    0.000000] CPU0 revision is: 00019650 (MIPS 24KEc)▒▒[    0.000000] MIPS: mac                                                                                                                                                             hine is Xiaom▒▒▒▒▒▒ter R3
▒▒▒▒▒▒00000] Initrd not found or empty - disabling initrd
[    0.000000] Prim▒▒▒▒▒▒▒▒uction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Prim▒▒▒▒▒▒▒cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] Zone ran▒▒󺍊۠   0.000000]   Normal   [mem 0x0000000000000000-0x0000                                                                                                                                                             000007ffffff]
[  ▒▒▒▒▒▒▒▒▒0] Movable zone start for each node
[    0.000000] Early memory node ra▒▒▒▒۠   0.000000]   node   0: [mem 0x000000000                                                                                                                                                             0000000-0x0000000007ffffff]
[ ▒▒▒▒▒▒▒▒▒00] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
[    0▒▒▒▒▒▒▒ݠBuilt 1 zonelists, mobility grouping on.  Total pages: 32480
[    0.000▒▒▒ݠ▒▒▒▒el command line: console=ttyS0,115200 rootfstype=squashfs,jffs                                                                                                                                                             2
▒▒▒▒▒▒00000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes, li▒▒                                                                                                                                                             ▒򩍊[ ▒▒▒▒▒▒▒▒▒00] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes, li                                                                                                                                                             near)▒▒[    0.000000] Writing ErrCtl register=0007a61f
[    0.000000] Rea▒▒▒▒▒▒▒▒Ctl register=0007a61f
[    0.000000] mem auto-init: stack:off, heap all▒▒▒▒欠heap free:off
[    0.000000] Memory: 121960K/131072ˠ▒▒▒▒▒▒ble (5018K kernel code, 209K rwdata,                                                                                                                                                              1104K rodata, 1212K init, 209K bss,▒▒▒▒▒ˠ▒▒served, 0K cma-reserved)
[    0.000000] SLUB: HWalign=32, ▒▒▒▒򽰭▒, MinObjects=0, CPUs=1, Nodes=1
▒▒▒▒▒▒▒0000] NR_IRQS: 256
[    0.000000] random: get_random_bytes called from start_kerne쫰▒▒0x518 with cr                                                                                                                                                             ng_init=0
[    0.000000] CPU Clock: 580MHz
[    0.000000] clocksource:▒▒▒▒▒▒▒▒: mask: 0xffff max_cycles: 0xffff, max_idle_n                                                                                                                                                             s: 583261500 ns
[    0.000▒▒▒ݠ▒▒▒▒ick: enable autosleep mode
[    0.000000] systick: running - mult: 2147▒▒▒▒▒▒▒▒t: 32
[    0.000000] clocksource: MIPS: mask: 0xffffffff ma▒▒▒▒▒▒▒▒ 0xffffffff, max_id                                                                                                                                                             le_ns: 6590553264 ns
[    0.000012] sched_clock: 32 ▒▒▒▒▒▒90MHz, resolution 3ns, wraps every 74051159                                                                                                                                                             02ns
[... 584 lines cut ...]
ash: missing ]
root@OpenWrt:/# [    1.059593] 0x000000140000-0x000000180000 : "crash_syslog"
ash: missing ]
root@OpenWrt:/# [    1.066338] 0x000000180000-0x000000200000 : "reserved0"
ash: missing ]
root@OpenWrt:/# [    1.072847] 0x000000200000-0x000000600000 : "kernel_stock"
ash: missing ]
root@OpenWrt:/# [    1.080395] 0x000000600000-0x000000a00000 : "kernel"
ash: missing ]
root@OpenWrt:/# [    1.087371] 0x000000a00000-0x000008000000 : "ubi"
ash: missing ]
root@OpenWrt:/# [    1.118982] libphy: Fixed MDIO Bus: probed
ash: missing ]
root@OpenWrt:/# [    1.139889] gsw: setting port4 to ephy mode
ash: missing ]
root@OpenWrt:/# [    1.144243] ralink_soc_eth 10100000.ethernet eth0 (uninitiali
zed): port 0 link up (100Mbps/Full duplex)
/bin/ash: syntax error: unexpected "("
root@OpenWrt:/# [    1.153829] ralink_soc_eth 10100000.ethernet eth0 (uninitiali
zed): port 1 link up (100Mbps/Full duplex)
/bin/ash: syntax error: unexpected "("
root@OpenWrt:/# [    1.163928] ralink_soc_eth 10100000.ethernet: loaded mt7620 d
river
ash: missing ]
root@OpenWrt:/# [    1.170974] ralink_soc_eth 10100000.ethernet eth0: mediatek

yuzhe1129, 2022-08-21:
一开始是默认的，
define Device/xiaomi_miwifi-r3
  SOC := mt7620a
  BLOCKSIZE := 128k
  PAGESIZE := 2048
  KERNEL_SIZE := 4096k
  IMAGE_SIZE := 32768k
  UBINIZE_OPTS := -E 5
  IMAGES += kernel1.bin rootfs0.bin
  IMAGE/kernel1.bin := append-kernel | check-size $$$$(KERNEL_SIZE)
  IMAGE/rootfs0.bin := append-ubi | check-size $$$$(IMAGE_SIZE)
  IMAGE/sysupgrade.bin := sysupgrade-tar | append-metadata
  DEVICE_VENDOR := Xiaomi
  DEVICE_MODEL := Mi Router R3
  DEVICE_PACKAGES := kmod-mt76x2 kmod-usb2 kmod-usb-ohci uboot-envtools
endef
TARGET_DEVICES += xiaomi_miwifi-r3
然后刷了没无线 WAN用不了就修改分区
KERNEL_SIZE := 4096k
  IMAGE_SIZE := 32768k
  UBINIZE_OPTS := -E 5
  IMAGES += kernel1.bin rootfs0.bin
  IMAGES += kernel1.bin rootfs0.bin pb-boot.bin boot.bin
  IMAGE/kernel1.bin := append-kernel | check-size $$$$(KERNEL_SIZE)
  IMAGE/rootfs0.bin := append-ubi | check-size $$$$(IMAGE_SIZE)
  IMAGE/rootfs0.bin := append-ubi | check-size
  IMAGE/sysupgrade.bin := sysupgrade-tar | append-metadata
  IMAGE/boot.bin := append-kernel | pad-to $$(KERNEL_SIZE) | append-ubi | check-size
  IMAGE/pb-boot.bin := append-kernel | pad-to $$(KERNEL_SIZE) | \
			     append-kernel | pad-to $$(KERNEL_SIZE) | \
			     append-ubi | check-size
  DEVICE_VENDOR := Xiaomi
  DEVICE_MODEL := Mi Router R3
  DEVICE_PACKAGES := kmod-mt76x2 kmod-usb2 kmod-usb-ohci uboot-envtools
生成的后缀有个boot的文件，刷了无线重启，

coolsnowwolf, 2026-09-05:
日志显示的是第三方 PandoraBox bootloader/旧版 5.4 固件，且无线和 WAN 失效缺少 DTS、校准分区和完整启动日志。不能归因于当前 LEDE，建议使用对应设备镜像并确认 ART/factory 分区，关闭为“不计划处理”。

closed 2026-09-05 by coolsnowwolf