coolsnowwolf/lede #9996: mac80211 backports-5.15.33-1/drivers/net/wireless/ath/ath11k/ath11k.ko' is missing

zhangguanzhang, 2022-08-25:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

```
make[2]: Entering directory '/workdir/openwrt/package/kernel/mac80211'
mkdir -p /workdir/openwrt/dl
SHELL= flock /workdir/openwrt/tmp/.backports-5.15.33-1.tar.xz.flock -c '  	/workdir/openwrt/scripts/download.pl "/workdir/openwrt/dl" "backports-5.15.33-1.tar.xz" "1b6b3bded4c81814ebebe2d194c2f8966d2399005b85ebb0557285b6e73f5422" "" "@KERNEL/linux/kernel/projects/backports/stable/v5.15.33/"    '
mkdir -p /workdir/openwrt/bin/targets/rockchip/armv8/packages /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1/ipkg-aarch64_generic/kmod-ath11k/CONTROL /workdir/openwrt/staging_dir/target-aarch64_generic_musl/pkginfo
ERROR: module '/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1/drivers/net/wireless/ath/ath11k/ath11k.ko' is missing.
make[2]: *** [Makefile:555: /workdir/openwrt/bin/targets/rockchip/armv8/packages/kmod-ath11k_5.19.3+5.15.33-1-1_aarch64_generic.ipk] Error 1
make[2]: Leaving directory '/workdir/openwrt/package/kernel/mac80211'
time: package/kernel/mac80211/regular/compile#0.43#0.16#0.52
    ERROR: package/kernel/mac80211 failed to build (build variant: regular).
make[1]: *** [package/Makefile:116: package/kernel/mac80211/compile] Error 1
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:230: package/kernel/mac80211/compile] Error 2
```

ecrasy, 2022-08-26:
编译树莓派4b固件也有同样的问题
make menuconfig里面mac80211已经是默认勾选了
但是scripts/diffscripts.sh 生成的config文件没有把mac80211添加进去
导致编译缺少模块
虽然可以通过手动添加解决
但还是希望这个Bug可以尽快修复

zhangguanzhang, 2022-08-26:
> 编译树莓派4b固件也有同样的问题 make menuconfig里面mac80211已经是默认勾选了 但是scripts/diffscripts.sh 生成的config文件没有把mac80211添加进去 导致编译缺少模块 虽然可以通过手动添加解决 但还是希望这个Bug可以尽快修复

我去你仓库搜索 mac80211 没搜到 sh 脚本里的修改，你现在是怎么避免这个的

hyy-666, 2022-08-26:
......,请问你有高通系的802.11ax的网卡吗，没有的话，加这个东西的意义何在.........

ecrasy, 2022-08-26:
> > 编译树莓派4b固件也有同样的问题 make menuconfig里面mac80211已经是默认勾选了 但是scripts/diffscripts.sh 生成的config文件没有把mac80211添加进去 导致编译缺少模块 虽然可以通过手动添加解决 但还是希望这个Bug可以尽快修复
> 
> 我去你仓库搜索 mac80211 没搜到 sh 脚本里的修改，你现在是怎么避免这个的

我没在脚本修改
我是回滚config到老版本解决
新老版本config就差在少了mac80211

ecrasy, 2022-08-26:
> ......,请问你有高通系的802.11ax的网卡吗，没有的话，加这个东西的意义何在.........

make defconfig 默认钩上的
这不是手选的
但是在生产config文件时没有生成对应的项
导致编译出错

zhangguanzhang, 2022-08-27:
> ......,请问你有高通系的802.11ax的网卡吗，没有的话，加这个东西的意义何在.........

我没开ath11k都报错这个

zhangguanzhang, 2022-08-30:
@aiamadeus 大佬，是不是这行

https://github.com/coolsnowwolf/lede/blob/0e14f147653461e7d9ffd30291c4f109171a532d/package/kernel/mac80211/ath.mk#L310

需要结尾加版本号判断

zhangguanzhang, 2022-08-31:
貌似这样
```
FILES:=$(PKG_BUILD_DIR)/drivers/net/wireless/ath/ath11k/ath11k.ko@ge5.19
```

closed 2022-09-06 by aiamadeus