coolsnowwolf/lede #10000: 最新版源码trojan编译失败

ricksuzade-maker, 2022-08-26:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：变异trojan时，提示libboost_program_options.so.1.79.0 和 libboost_system.so.1.79.0 缺失

### (2) 路由器型号和固件版本

A：x86-64, mips, arm64平台均出现这个问题

### (3) 详细日志

A：
touch /home/zhiliang/scripts/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.boost_installed
echo "boost" >> /home/zhiliang/scripts/lede/staging_dir/target-x86_64_musl/pkginfo/boost.default.install
make[3]: Leaving directory '/home/zhiliang/scripts/lede/feeds/packages/libs/boost'
time: package/feeds/packages/boost/compile#420.78#85.78#83.50
make[3]: Entering directory '/home/zhiliang/scripts/lede/feeds/helloworld/trojan'
mkdir -p /home/zhiliang/scripts/lede/dl
SHELL= flock /home/zhiliang/scripts/lede/tmp/.trojan-1.16.0.tar.gz.flock -c '  	/home/zhiliang/scripts/lede/scripts/download.pl "/home/zhiliang/scripts/lede/dl" "trojan-1.16.0.tar.gz" "86cdb2685bb03a63b62ce06545c41189952f1ec4a0cd9147450312ed70956cbc" "" "https://codeload.github.com/trojan-gfw/trojan/tar.gz/v1.16.0?"    '
mkdir -p /home/zhiliang/scripts/lede/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/zhiliang/scripts/lede/tmp/.root-copy.flock -c 'cp -fpR /home/zhiliang/scripts/lede/build_dir/target-x86_64_musl/trojan-1.16.0/.pkgdir/trojan/. /home/zhiliang/scripts/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/zhiliang/scripts/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.trojan_installed
mkdir -p /home/zhiliang/scripts/lede/bin/targets/x86/64/packages /home/zhiliang/scripts/lede/build_dir/target-x86_64_musl/trojan-1.16.0/ipkg-x86_64/trojan/CONTROL /home/zhiliang/scripts/lede/staging_dir/target-x86_64_musl/pkginfo
install -d -m0755 /home/zhiliang/scripts/lede/build_dir/target-x86_64_musl/trojan-1.16.0/ipkg-x86_64/trojan/usr/sbin
install -m0755 /home/zhiliang/scripts/lede/build_dir/target-x86_64_musl/trojan-1.16.0/ipkg-install/usr/bin/trojan /home/zhiliang/scripts/lede/build_dir/target-x86_64_musl/trojan-1.16.0/ipkg-x86_64/trojan/usr/sbin/trojan
find /home/zhiliang/scripts/lede/build_dir/target-x86_64_musl/trojan-1.16.0/ipkg-x86_64/trojan -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
Package trojan is missing dependencies for the following libraries:
libboost_program_options.so.1.79.0
libboost_system.so.1.79.0
make[3]: *** [Makefile:77: /home/zhiliang/scripts/lede/bin/packages/x86_64/helloworld/trojan_1.16.0-2_x86_64.ipk] Error 1
make[3]: Leaving directory '/home/zhiliang/scripts/lede/feeds/helloworld/trojan'
time: package/feeds/helloworld/trojan/compile#0.28#0.06#0.30
    ERROR: package/feeds/helloworld/trojan failed to build.
make[2]: *** [package/Makefile:116: package/feeds/helloworld/trojan/compile] Error 1
make[2]: Leaving directory '/home/zhiliang/scripts/lede'
make[1]: *** [package/Makefile:110: /home/zhiliang/scripts/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/zhiliang/scripts/lede'
make: *** [/home/zhiliang/scripts/lede/include/toplevel.mk:230: world] Error 2

hyy-666, 2022-08-26:
先make clean。。。。。,大雕更新boost了，你这个有缓存在，可以单独删除boost的源码包，但是更推荐make clean毕竟很多都依赖boost

CYaNu, 2022-08-26:
删除./tmp ./dl下trojan的下载包 然后make V=s download下载一下就行了

yzq159, 2022-08-27:
缺少libboost，sudo apt-get install libboost-all-dev

closed 2022-10-05 by aiamadeus