coolsnowwolf/lede #10001: 内核日志乱码

mitsukileung, 2022-08-26:
X86

[   89.492166] rcu: INFO: rcu_preempt detected expedited stalls on CPUs/tasks: { 1-... } 3 jiffies s: 69 root: 0x2/.
[   89.501448] rcu: blocking rcu_node structures (internal RCU debug):
[   89.566556] Task dump for CPU 1:
[   89.569459] task:grep            state:R  running task     stack:    0 pid:11211 ppid: 11135 flags:0x00000002
[   89.576162] Call Trace:
[   89.579184]  <TASK>
[   89.582014]  ??!u?+0x23c/0x5e0
[   89.591002]  ? ????k?n??k+0x40/0x70
[   89.604760]  d?t????+0x43/0x50
[   89.607701]  d???+0x74b/0xa20
[   89.610467]  d?#???+0x34/0xa0
[   89.620283]  x??ٰ#?+0x17/0x20
[   89.816895]  ?sc??+0x46/0xc0
[   89.819842]  ???SYSCALL_6{aft?hwf??+0x63/0xcd
[   89.829445] RIP: 0033:0x7f2bf325230d
[   89.832294] RSP: 002b:00007ffda72a6338 EFLAGS: 00000246 ORIG_RAX: 00000000000000e7
[   89.860822] RAX: ffffffffffffffda RBX: 0000000000000001 RCX: 00007f2bf325230d
[   89.872185] RDX: 0000000000000000 RSI: 0000000000000000 RDI: 0000000000000001
[   90.052713] RBP: 00007ffda72a63e8 R08: 0000000000000000 R09: 0000000000000000
[   90.062044] R10: 0000000000000000 R11: 0000000000000246 R12: 000000000044b9bf
[   90.071597] R13: 00007ffda72a6410 R14: 00007f2bf32a7ac8 R15: 0000000000000000
[   90.090265]  </TASK>

大佬们，这个内核日志乱码是什么鬼呀？

lovehackintosh, 2022-08-26:
搜一下以前的issue

mitsukileung, 2022-08-26:
搜过了，我没开serverchan也会

closed 2022-08-27 by coolsnowwolf