coolsnowwolf/lede #10002: R2S编译出错，大佬帮看看什么原因。。。

drw2005, 2022-08-26:
最近一段时间一直编译错误，也看不懂日志提示的错误原因，请大佬给看看指导指导，感谢！

ecrasy, 2022-08-26:
make -j1 看看
你这LOG都没截完整

zhangguanzhang, 2022-08-26:
你这不按照issue的模板带日志，可能没人理你

drw2005, 2022-08-26:
> 你这不按照issue的模板带日志，可能没人理你

大佬批评的是，我又重新上传了二次编译出错的是部分日志。

ecrasy, 2022-08-26:
fatal error: Python.h: No such file or directory
你环境没搭好？

zhangguanzhang, 2022-08-26:
1. ubuntu 系统版本多少，2204好像有个啥注意事项来着
2. 初次编译失败的话可能是依赖没安装全

drw2005, 2022-08-26:
> fatal error: Python.h: No such file or directory 你环境没搭好？
环境没问题，编译x86都没问题。

drw2005, 2022-08-26:
> 1. ubuntu 系统版本多少，2204好像有个啥注意事项来着
20.04
> 2. 初次编译失败的话可能是依赖没安装全
依赖也重新安装了。

lifetyper, 2022-08-26:
也碰到一样的错误了，不是环境问题，应该是uboot部分的问题。

coolsnowwolf, 2022-08-26:
明显环境问题，缺 libpython3-dev 这个包

closed 2022-08-28 by aiamadeus

drw2005, 2022-08-29:
> 明显环境问题，缺 libpython3-dev 这个包

确实，又重新安装了一遍环境，现在可以了，编译成功。

yance-yang, 2024-01-05:
请问 libpython3-dev， libpython2-dev， python3-dev， python2-dev 都安装了，编译uboot还是提示没有Python.h头文件。