coolsnowwolf/lede #10003: 编译x86 5.10内核报错

bi4tms, 2022-08-26:
日志如下：
Applying /home/x86.510/lede/target/linux/generic/hack-5.10/952-net-conntrack-events-support-multiple-registrant.patch using plaintext:
patching file include/net/netfilter/nf_conntrack_ecache.h
patching file include/net/netns/conntrack.h
patching file net/netfilter/Kconfig
patching file net/netfilter/nf_conntrack_core.c
patching file net/netfilter/nf_conntrack_ecache.c
patching file net/netfilter/nf_conntrack_netlink.c
Hunk #1 FAILED at 706.
Hunk #2 succeeded at 3752 (offset -35 lines).
1 out of 2 hunks FAILED -- saving rejects to file net/netfilter/nf_conntrack_netlink.c.rej
Patch failed!  Please fix /home/x86.510/lede/target/linux/generic/hack-5.10/952-net-conntrack-events-support-multiple-registrant.patch!
Makefile:27: recipe for target '/home/x86.510/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.138/.prepared_09169c28e5446ab545029c4831373065' failed
make[4]: *** [/home/x86.510/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.138/.prepared_09169c28e5446ab545029c4831373065] Error 1
make[4]: Leaving directory '/home/x86.510/lede/target/linux/x86'
Makefile:11: recipe for target 'compile' failed
make[3]: *** [compile] Error 2
make[3]: Leaving directory '/home/x86.510/lede/target/linux'
time: target/linux/compile#18.15#70.89#69.42
    ERROR: target/linux failed to build.
target/Makefile:23: recipe for target 'target/linux/compile' failed
make[2]: *** [target/linux/compile] Error 1
make[2]: Leaving directory '/home/x86.510/lede'
target/Makefile:18: recipe for target '/home/x86.510/lede/staging_dir/target-x86_64_musl/stamp/.target_compile' failed
make[1]: *** [/home/x86.510/lede/staging_dir/target-x86_64_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/x86.510/lede'
/home/x86.510/lede/include/toplevel.mk:228: recipe for target 'world' failed
make: *** [world] Error 2

coolsnowwolf, 2022-08-26:
悲剧 @lovehackintosh 有空可以看看么？

lovehackintosh, 2022-08-26:
> 悲剧 @lovehackintosh 有空可以看看么？

不好意思,以后再bump我会先编译一遍的

closed 2022-08-26 by aiamadeus