coolsnowwolf/lede #10004: 编译x86 5.15内核报错

bi4tms, 2022-08-26:
日志如下：
Applying /home/x86/lede/target/linux/generic/pending-5.15/851-0006-Revert-PCI-aardvark-Fix-initialization-with-old-Marv.patch using plaintext:
patching file drivers/pci/controller/pci-aardvark.c
patch: **** malformed patch at line 29: @@ -1631,9 +1631,7 @@ static int advk_pcie_enable_phy(struct a

Patch failed!  Please fix /home/x86/lede/target/linux/generic/pending-5.15/851-0006-Revert-PCI-aardvark-Fix-initialization-with-old-Marv.patch!
Makefile:112: recipe for target '/home/x86/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/linux-5.15.63/.prepared' failed
make[3]: *** [/home/x86/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/linux-5.15.63/.prepared] Error 1
make[3]: Leaving directory '/home/x86/lede/toolchain/kernel-headers'
time: toolchain/kernel-headers/compile#16.87#72.20#72.83
    ERROR: toolchain/kernel-headers failed to build.
toolchain/Makefile:95: recipe for target 'toolchain/kernel-headers/compile' failed
make[2]: *** [toolchain/kernel-headers/compile] Error 1
make[2]: Leaving directory '/home/x86/lede'
toolchain/Makefile:93: recipe for target '/home/x86/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/stamp/.toolchain_compile' failed
make[1]: *** [/home/x86/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/stamp/.toolchain_compile] Error 2
make[1]: Leaving directory '/home/x86/lede'
/home/x86/lede/include/toplevel.mk:228: recipe for target 'world' failed

lifetyper, 2022-08-26:
一样的错误

759071793, 2022-08-26:
一样的错误

fxvbv, 2022-08-26:
确实出现了这个错误。

zhangguanzhang, 2022-08-26:
```
Patch failed! Please fix /home/x86/lede/target/linux/generic/pending-5.15/851-0006-Revert-PCI-aardvark-Fix-initialization-with-old-Marv.patch!
```
这个 patch有问题

CoderKwong, 2022-08-26:
同一个问题

MIAOBUSI, 2022-08-26:
pull完最新源码还是这个错误 日志如下：

MIAOBUSI, 2022-08-26:
Applying /home/miao/lede/target/linux/generic/pending-5.15/851-0006-Revert-PCI-aardvark-Fix-initialization-with-old-Marv.patch using plaintext: 
patching file drivers/pci/controller/pci-aardvark.c
patch: **** malformed patch at line 29: @@ -1631,9 +1631,7 @@ static int advk_pcie_enable_phy(struct a

Patch failed!  Please fix /home/miao/lede/target/linux/generic/pending-5.15/851-0006-Revert-PCI-aardvark-Fix-initialization-with-old-Marv.patch!
make[3]: *** [Makefile:122: /home/miao/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/linux-5.15.63/.prepared] Error 1
make[3]: Leaving directory '/home/miao/lede/toolchain/kernel-headers'
time: toolchain/kernel-headers/compile#8.85#6.10#13.97
    ERROR: toolchain/kernel-headers failed to build.
make[2]: *** [toolchain/Makefile:97: toolchain/kernel-headers/compile] Error 1
make[2]: Leaving directory '/home/miao/lede'
make[1]: *** [toolchain/Makefile:93: /home/miao/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/stamp/.toolchain_compile] Error 2
make[1]: Leaving directory '/home/miao/lede'
make: *** [/home/miao/lede/include/toplevel.mk:230：world] 错误 2

lovehackintosh, 2022-08-26:
临时解决办法:
```shell
```

closed 2022-08-26 by aiamadeus

MIAOBUSI, 2022-08-26:
> 临时解决办法:
> 
> ```shell
> wget -O target/linux/generic/pending-5.15/851-0006-Revert-PCI-aardvark-Fix-initialization-with-old-Marv.patch https://github.com/benihi/lede/raw/5db1b1986a1b1b9d7c294f1c6526c54c33a43814/target/linux/generic/pending-5.15/851-0006-Revert-PCI-aardvark-Fix-initialization-with-old-Marv.patch
> ```

OK 感谢

PR #10006 (2022-08-26): kernel: 5.15: fix bug in patch format
fixes: #10004
Q：你知道这是`pull request`吗？(使用 "x" 选择)
* [x] 我知道