coolsnowwolf/lede #10008: RTL8821CU 无线网卡不能开启AP模式

Hyy2001X, 2022-08-28:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

### 默认设置下的无线如图，红圈中的内容为异常情况, 这些信息怎么设置都显示为0
[image]

rtl8821cu 的驱动和固件已经编译进了

**root@Openwrt:~# lsmod | grep 8821cu**
> compat                 16384 17 rtw88_8822ce,rtw88_8822be,rtw88_8821cu,rtw88_8723de,rtl8821ae,rtl8192se,rtl8192de,rtl8192cu,rtl8192ce,iwlmvm,iwldvm,btcoexist,rtlwifi,rtl8xxxu,mac80211,iwlwifi,cfg80211,[permanent]
rtw88_8821c            86016  1 rtw88_8821cu
rtw88_8821cu           16384  0
rtw88_usb              20480  1 rtw88_8821cu

**root@Openwrt:~# opkg list | grep -E 'wpad|hostapd|mac80211|rtw|8821cu'**
> hostapd-common - 2022-01-16-cff80b4f-5
kmod-mac80211 - 5.19.4+5.15.33-1-1
wpad-openssl - 2022-01-16-cff80b4f-5
kmod-rtw88 - 5.19.4+5.15.33-1-1
rtl8821cu-firmware - 20220708-1

### (2) 路由器型号和固件版本

A：x86_64 , 固件编译日期 2022/08/26

### (3) 详细日志

**root@Openwrt:~# wifi up**
> Sun Aug 28 10:52:48 2022 daemon.notice hostapd: Configuration file: /var/run/hostapd-phy0.conf (phy wlan0) --> new PHY
Sun Aug 28 10:52:48 2022 daemon.err hostapd: Could not set interface wlan0 flags (UP): Resource busy
Sun Aug 28 10:52:48 2022 daemon.err hostapd: nl80211: Could not set interface 'wlan0' UP
Sun Aug 28 10:52:48 2022 daemon.notice hostapd: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Sun Aug 28 10:52:48 2022 daemon.err hostapd: nl80211 driver initialization failed.
Sun Aug 28 10:52:48 2022 daemon.notice hostapd: wlan0: CTRL-EVENT-TERMINATING
Sun Aug 28 10:52:48 2022 daemon.err hostapd: hostapd_free_hapd_data: Interface wlan0 wasn't started
Sun Aug 28 10:52:48 2022 daemon.notice netifd: radio0 (21615): Command failed: Invalid argument
Sun Aug 28 10:52:48 2022 daemon.notice netifd: radio0 (21615): Device setup failed: HOSTAPD_START_FAILED
Sun Aug 28 10:52:48 2022 kern.err kernel: [25307.885726] rtw_8821cu 1-2:1.0: error beacon valid
Sun Aug 28 10:52:48 2022 kern.err kernel: [25307.886269] rtw_8821cu 1-2:1.0: failed to download rsvd page
Sun Aug 28 10:52:48 2022 kern.err kernel: [25307.886940] rtw_8821cu 1-2:1.0: failed to download firmware

szdosar, 2022-08-29:
这个问题我报过，目前无解。于是我回退到以前的版本：
在 git clone 之后，`git reset --hard 77251e963f61f33504d24defef9dcfe1104d0a70`
然后该干嘛干嘛。

Hyy2001X, 2022-08-29:
> 这个问题我报过，目前无解。于是我回退到以前的版本： 在 git clone 之后，`git reset --hard 77251e963f61f33504d24defef9dcfe1104d0a70` 然后该干嘛干嘛。

之前看到过你的issue，但是症状不太一样，相同的固件用`AX200`网卡作AP的时候也是上图中的状况均显示`0`

closed 2022-09-01 by Hyy2001X

Purestreams, 2023-03-24:
请问您最后是怎么解决ax200无发射功率的问题的？我也遇到了一样的情况

Hyy2001X, 2023-03-25:
> 请问您最后是怎么解决ax200无发射功率的问题的？我也遇到了一样的情况

没解决，建议用`mt7921`，便宜好用

zzh151223, 2023-04-15:
> > 请问您最后是怎么解决ax200无发射功率的问题的？我也遇到了一样的情况
> 
> 没解决，建议用`mt7921`，便宜好用

网卡驱动能单独编译吗？我用的别人的固件，没有编译mt7921，我自己编md5又对不上，把md5生成改成固定的又会编译失败

Hyy2001X, 2023-04-15:
> > > 请问您最后是怎么解决ax200无发射功率的问题的？我也遇到了一样的情况
> > 
> > 
> > 没解决，建议用`mt7921`，便宜好用
> 
> 网卡驱动能单独编译吗？我用的别人的固件，没有编译mt7921，我自己编md5又对不上，把md5生成改成固定的又会编译失败

5.15内核，CONFIG_PACKAGE_kmod-mt7921e=y，可以编译出固件的