coolsnowwolf/lede #10010: 更新固件经常掉配置，密码恢复默认

xeyesen, 2022-08-28:
经常每隔一段时间编译出来的固件，更新后密码总是恢复到默认的password，修改了密码，下次再更新还是会变成password，每次都需要不保存配置更新，设置全部要重新再搞一遍，太麻烦了，有什么办法可以避免这个问题吗？太多设置了，要设置好久

xiangfeidexiaohuo, 2022-08-28:
我只发现 密码会还原

sol1999, 2022-09-11:
一样问题,保留升级刷固件密码会恢复到默认的password

xyl94k, 2022-09-14:
是的，密码会重置

zwillhill, 2022-09-18:
直接把密码编译到固件里，用改过密码的package/lean/default-settings/files/zzz-default-settings文件替换原文件
或者使用sed命令修改该文件
`sed -i 's/原加密后密码/新加密后密码/g' package/lean/default-settings/files/zzz-default-settings`

xeyesen, 2022-09-18:
> 直接把密码编译到固件里，用改过密码的package/lean/default-settings/files/zzz-default-settings文件替换原文件 或者使用sed命令修改该文件 `sed -i 's/原加密后密码/新加密后密码/g' package/lean/default-settings/files/zzz-default-settings`

不止是密码，有些配置也会没掉，比如我设置了备份root下的文件，更新后root下面的文件都不见了

naoki66, 2022-09-24:
是的   似乎是某个随机的版本  发现配置文件掉了好多 以至于我都不敢随便升级

sjkhsl, 2022-10-05:
我是保留数据更新或者不保留，之后导入备份数据都会进不去页面～

cxgreat2014, 2023-01-28:
相同的问题，很好奇原理是啥，浪费了不少时间

ps 另外这个issue id吉祥

coolsnowwolf, 2026-09-05:
固件升级后配置/密码是否保留取决于镜像类型、overlay 分区和是否执行保留配置升级；报告没有设备、分区布局或升级命令，无法归因于当前代码，关闭为 not planned。

closed 2026-09-05 by coolsnowwolf