coolsnowwolf/lede #10011: 哪位大神来帮忙一下 Sed 默认源 下载没速度

shedkiss, 2022-08-29:
make download -j8 后:
 make[2] -C tools/sed download  一直卡着没速度 ,是不是没有镜像？国内的源镜像找到了在https://mirrors.tuna.tsinghua.edu.cn/gnu/sed/   ，如何添加描述？或者能否请大神更新一下，万分感谢

1054009064, 2022-08-29:
默认有腾讯源，按道理不会没有速度吧，你要加个 V=s 看看是哪个卡了
https://github.com/coolsnowwolf/lede/blob/f8015dd5e9d738c9706361c52ddf89d813f8b519/scripts/download.pl#L227

xcyll, 2022-08-29:
最近2天 lean仓库action里的 make download也要17-8分钟了
以前3-5分钟吧

1054009064, 2022-08-29:
还是建议提供单线程的详细下载日志。😅

Racon94, 2022-08-30:
我全局翻墙解决了，用虚拟机编译

rahxyz, 2022-09-01:
SHELL= flock /home/raxx/lede/tmp/.xxd-1.10.tar.gz.flock -c '    /home/raxx/lede/scripts/download.pl "/home/raxx/lede/dl" "xxd-1.10.tar.gz" "9bf05c19b9084973e3cc877696a7f9881a5c87fa5a9fa438d9962519726559f9" "" "http://grail.cba.csuohio.edu/~somos/"    '
+ curl -f --connect-timeout 20 --retry 5 --location --insecure http://grail.cba.csuohio.edu/~somos/xxd-1.10.tar.gz

这个卡住了，几天了都。

1054009064, 2022-09-01:
你们可以参考这个取消的 pr 改。 #10015

shedkiss, 2022-09-01:
还是卡住

1054009064, 2022-09-01:
我改动后的

[image]

bihell, 2022-09-01:
是要卡一段时间的，挂代理吧，也要卡一会

1054009064, 2022-09-01:
卡还有一个原因就是某些包并不是打包好，需要拉源码到本地打包，如果那个源码比较大，这时候就要看你的电脑打包性能了

lifetyper, 2022-09-01:
老美太不敬业了，服务器挂了几天了都不修。grail.cba.csuohio.edu这个站点从美国服务器都拉不动。

Cathgao, 2022-09-02:
有个服务器挂掉了，包半天拉不下来。我这没开详细日志，不知道卡的是哪个包
[image]

closed 2022-09-03 by aiamadeus

coderhwz, 2022-10-10:
现在是全世界都ping不通，翻墙也没用，这位老哥临时提供了文件，来解决 https://github.com/openwrt/openwrt/issues/10555 。 在22.03 rc6里还存在这个东西，不过master 已经没了。即没用，又恶心人。   也可以  https://github.com/openwrt/openwrt/commit/c6d3f39ecce43c4a9858157e9e2ee8718750a9ab  去掉

xiaopang0v0, 2023-03-02:
> 老美太不敬业了，服务器挂了几天了都不修。grail.cba.csuohio.edu这个站点从美国服务器都拉不动。

你应该是卡在了 `xxd` 吧，这并不是由于这个网站挂掉了，而是因为这个网站**屏蔽了 http 请求**，只需要进入这个包的 Makefile 里，把 http 改成 https 就可以了
```
tools/xxd/Makefile
```