coolsnowwolf/lede #10028: 负载均衡未能正确支持VPN，显示接口禁用（但实际部分能用）

xingyunjohn1, 2022-08-31:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：有物理WAN口和L2TP两个接口，需求是两个接口进行负载均衡；L2TP接口被显示为已禁用，若策略仅指定L2tp接口，可以通过l2tp接口上网，若策略中存在两个接口，则仅通过WAN口（100%）。
相关参考内容：https://github.com/openwrt/packages/issues/13145

### (2) 路由器型号和固件版本

A：x86. OpenWrt Ipv6-Plus-S20220701 by Sirpdboy

### (3) 详细日志

A：
`Interface status:
 interface wan is online 00h:11m:23s, uptime 94h:33m:42s and tracking is active
 interface l2tp is offline and tracking is paused

Current ipv4 policies:
balanced:
 wan (100%)
firstl2tp:
 wan (100%)
onlyl2tp:
 default
onlywan:
 wan (100%)`

PS: 我可以顺便问一下mwan3的日志目录吗，只找到系统日志中有少许几条mwan3启停日志。

goodwinY, 2022-09-10:
你可以试试如下操作：
网络-接口-接口wan/l2tp-高级设置-设置网关跃点
wan设为41
l2tp设为42

coolsnowwolf, 2026-09-05:
报告基于 2022 年第三方固件和旧版 mwan3，只有接口状态摘要，没有 mwan3 配置、L2TP 建链/路由日志或当前版本复现。无法确认是 LEDE 包缺陷；请按当前 mwan3 文档提供完整配置和日志后再开新 issue，现关闭为“不计划处理”。

closed 2026-09-05 by coolsnowwolf