coolsnowwolf/lede #10029: MR7350 机型 SMB掉速严重

Lorde627, 2022-08-31:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：MR7350 Plus 版本，在年初时SMB 速度无线连接能读取跑到 70M/s，在近一段时间渐渐掉速到 20M/s。
移动硬盘检测无问题，插在电脑上也是读取满速 130M/s。
怀疑可能是 USB 掉到 2.0 ？

### (2) 路由器型号和固件版本

A：
MR7350 Plus 
openwrt-ipq60xx-R22.8.22-linksys_mr7350_plus-squashfs-nand-sysupgrade.rar

### (3) 详细日志

A：
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.4.60 (lean@lean-esxi) (gcc version 7.5.0 (LEDE GCC 7.5.0 r255-84c894fc0) ) #0 SMP PREEMPT Wed Jun 1 19:15:57 2022
[    0.000000] Boot CPU: AArch64 Processor [51af8014]
[    0.000000] Ignoring memory range 0x40000000 - 0x41000000
[    0.000000] Machine: Linksys MR7350_Plus IPQ6018/AP-CP03-C1
[    0.000000] efi: Getting EFI parameters from FDT:
[    0.000000] efi: UEFI not found.
[    0.000000] cma: dma_contiguous_reserve(limit c0000000)
[    0.000000] On node 0 totalpages: 503040
[    0.000000]   DMA zone: 8128 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 503040 pages, LIFO batch:31
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.0 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] PERCPU: Embedded 15 pages/cpu @ffffffc07ef54000 s23192 r8192 d30056 u61440
[    0.000000] pcpu-alloc: s23192 r8192 d30056 u61440 alloc=15*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 494912
[    0.000000] Kernel command line: console=ttyMSM0,115200n8 ubi.mtd=rootfs root=mtd:ubi_rootfs rootfstype=squashfs rootwait root=/dev/ubiblock0_1 swiotlb=1 coherent_pool=2M
[    0.000000] PID hash table entries: 4096 (order: 3, 32768 bytes)
[    0.000000] Dentry cache hash table entries: 262144 (order: 9, 2097152 bytes)
[    0.000000] Inode-cache hash table entries: 131072 (order: 8, 1048576 bytes)
[    0.000000] software IO TLB [mem 0xbfe0c000-0xbfe4c000] (0MB) mapped at [ffffffc07ee0c000-ffffffc07ee4bfff]
[    0.000000] Memory: 1964960K/2012160K available (5757K kernel code, 676K rwdata, 3072K rodata, 244K init, 408K bss, 47200K reserved, 0K cma-reserved)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vmalloc : 0xffffff8000000000 - 0xffffffbdbfff0000   (   246 GB)
[    0.000000]     vmemmap : 0xffffffbdc0000000 - 0xffffffbfc0000000   (     8 GB maximum)
[    0.000000]               0xffffffbdc0040000 - 0xffffffbdc2000000   (    31 MB actual)
[    0.000000]     fixed   : 0xffffffbffa7fd000 - 0xffffffbffac00000   (  4108 KB)
[    0.000000]     PCI I/O : 0xffffffbffae00000 - 0xffffffbffbe00000   (    16 MB)
[    0.000000]     modules : 0xffffffbffc000000 - 0xffffffc000000000   (    64 MB)
[    0.000000]     memory  : 0xffffffc000000000 - 0xffffffc07f000000   (  2032 MB)
[    0.000000]       .init : 0xffffffc000921000 - 0xffffffc00095e000   (   244 KB)
[    0.000000]       .text : 0xffffffc000080000 - 0xffffffc000920f94   (  8836 KB)
[    0.000000]       .data : 0xffffffc00096f000 - 0xffffffc000a18200   (   677 KB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Preemptible hierarchical RCU implementation.
[    0.000000] 	CONFIG_RCU_FANOUT set to non-default value of 32
[    0.000000] NR_IRQS:64 nr_irqs:64 0
[    0.000000] Architected cp15 timer(s) running at 24.00MHz (virt).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x588fe9dc0, max_idle_ns: 440795202592 ns
[    0.000005] sched_clock: 56 bits at 24MHz, resolution 41ns, wraps every 4398046511097ns
[    0.000402] Calibrating delay loop (skipped), value calculated using timer frequency.. 48.00 BogoMIPS (lpj=240000)
[... 762 lines cut ...]
[  354.100008] scsi host1: uas_eh_device_reset_handler start
[  354.108969] xhci-hcd xhci-hcd.0.auto: ERROR Transfer event for disabled endpoint or incorrect stream ring
[  354.114489] xhci-hcd xhci-hcd.0.auto: @00000000bc297830 00000000 00000000 1b000000 01058001
[  354.243484] usb 2-1: reset SuperSpeed USB device number 3 using xhci-hcd
[  354.265089] scsi host1: uas_eh_device_reset_handler success
[  354.273512] sd 1:0:0:0: [sdb] 976773168 512-byte logical blocks: (500 GB/466 GiB)
[  354.273540] sd 1:0:0:0: [sdb] 4096-byte physical blocks
[  354.280731] sd 1:0:0:0: [sdb] Write Protect is off
[  354.285243] sd 1:0:0:0: [sdb] Mode Sense: 53 00 00 08
[  354.285637] sd 1:0:0:0: [sdb] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
[  354.290040] xhci-hcd xhci-hcd.0.auto: ERROR Transfer event for disabled endpoint or incorrect stream ring
[  354.298810] xhci-hcd xhci-hcd.0.auto: @00000000bc297ae0 00000000 00000000 1b000000 01078001
[  356.329790]  sdb: sdb1 sdb2 sdb3 sdb4
[  356.332281] sd 1:0:0:0: [sdb] Attached SCSI disk
[  389.551507] [EXFAT] trying to mount...
[  422.352276] [EXFAT] trying to mount...
[  422.525680] [EXFAT] mounted successfully
[  434.332489] [EXFAT] trying to unmount...
[  434.332726] [EXFAT] unmounted successfully
[  447.049701] nf_conntrack: automatic helper assignment is deprecated and it will be removed soon. Use the iptables CT target to attach helpers instead.
[  449.872100] [EXFAT] trying to mount...
[  450.090807] [EXFAT] mounted successfully
[  453.566732] sh (7168): drop_caches: 3
[  455.306392] EXT4-fs (sdb3): recovery complete
[  455.306427] EXT4-fs (sdb3): mounted filesystem with ordered data mode. Opts:

coolsnowwolf, 2022-08-31:
[image]
磁盘有坏道

Lorde627, 2022-09-01:
> [image] 磁盘有坏道

现在的盘的确有一个分区提示损坏了，扫了全盘没有提示有坏道来着。
更换了一个固态硬盘，也是同样的速度。

lxl6125, 2022-09-01:
把bbr关了试试

Lorde627, 2022-09-02:
> 把bbr关了试试

IPQ6000 机型没有关bbr的选项

H3MKL, 2022-09-20:
你好，可以蹲一个编译配置或是固件吗

coolsnowwolf, 2026-09-05:
核查了该报告：问题来自 2022 年 LEDE R22.8.22 的 ipq60xx/MR7350 Plus 固件（Linux 4.4、厂商 wlan 驱动）。日志同时显示 USB 存储在 UAS/xHCI 重置和断连（`usb disconnect`、`I/O error`、`uas_eh_*`），SMB 吞吐下降没有可复现的 Samba/LEDE 包错误或足够的对比测试。当前主线已不再使用该旧 4.4/厂商驱动组合，无法在现行代码中做针对性修复，因此关闭为“不计划处理”。若仍可复现，请提供当前固件版本、`iperf3`/有线与无线对比、`smbd` 配置和完整复现日志。

closed 2026-09-05 by coolsnowwolf