coolsnowwolf/lede #10034: adbyby Plus+的问题，大神看看

rongcenter, 2022-09-01:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

Fri Sep  2 00:52:02 2022 daemon.crit dnsmasq[22984]: bad option at line 1 of /var/etc/dnsmasq-adbyby.d/04-dnsmasq.adblock
Fri Sep  2 00:52:02 2022 daemon.crit dnsmasq[22984]: FAILED to start up
Fri Sep  2 00:52:07 2022 daemon.crit dnsmasq[23006]: bad option at line 1 of /var/etc/dnsmasq-adbyby.d/04-dnsmasq.adblock
Fri Sep  2 00:52:07 2022 daemon.crit dnsmasq[23006]: FAILED to start up

每次启用时，会报这个错误。而且启用后断网。

### (2) 路由器型号和固件版本

x86

### (3) 详细日志

Fri Sep  2 00:52:02 2022 daemon.crit dnsmasq[22984]: bad option at line 1 of /var/etc/dnsmasq-adbyby.d/04-dnsmasq.adblock
Fri Sep  2 00:52:02 2022 daemon.crit dnsmasq[22984]: FAILED to start up
Fri Sep  2 00:52:07 2022 daemon.crit dnsmasq[23006]: bad option at line 1 of /var/etc/dnsmasq-adbyby.d/04-dnsmasq.adblock
Fri Sep  2 00:52:07 2022 daemon.crit dnsmasq[23006]: FAILED to start up

mitsukileung, 2022-09-01:
把adbyby 关了就没事，更新了规则后就会断网

firkerword, 2022-09-02:
规则问题吧。更新之后就这样了  我还为此更新了固件

592732377, 2022-09-03:
+1 x86刚编译的  更新adbyby规则后dnsmasq无法启动。

guoshh1978, 2022-09-03:
是新编译的adbyby-plus生成的配置文件格式和之前的不一样了。

guoshh1978, 2022-09-03:
可以看看新编译的版本启用adbyby-plus以后，里面的04-dnsmasq.adblock应该和下面的是不一样的。我估计把上面那些不合规的内容清理掉，也许dnsmasq就可以正常启动了。
root@OpenWrt:/# cat /var/etc/dnsmasq-adbyby.d/04-dnsmasq.adblock |head -10
address=/001adafg.xyz/0.0.0.0
address=/04jhgyt.com/0.0.0.0
address=/0577.la/0.0.0.0
address=/103.21.91.144/0.0.0.0
address=/1527ego.com/0.0.0.0
address=/15h3na0.xyz/0.0.0.0
address=/17chk.vip/0.0.0.0
address=/1dfsdfg.com/0.0.0.0
address=/1e4y54ve.cn/0.0.0.0
address=/1so8inqb.cn/0.0.0.0

nuo12138, 2022-09-04:
同样的报错
Sun Sep  4 09:35:26 2022 daemon.crit dnsmasq[29195]: bad option at line 1 of /var/etc/dnsmasq-adbyby.d/04-dnsmasq.adblock
Sun Sep  4 09:35:26 2022 daemon.crit dnsmasq[29195]: FAILED to start up
Sun Sep  4 09:35:32 2022 daemon.crit dnsmasq[29236]: bad option at line 1 of /var/etc/dnsmasq-adbyby.d/04-dnsmasq.adblock
Sun Sep  4 09:35:32 2022 daemon.crit dnsmasq[29236]: FAILED to start up
Sun Sep  4 09:35:37 2022 daemon.crit dnsmasq[29256]: bad option at line 1 of /var/etc/dnsmasq-adbyby.d/04-dnsmasq.adblock
Sun Sep  4 09:35:37 2022 daemon.crit dnsmasq[29256]: FAILED to start up
Sun Sep  4 09:35:42 2022 daemon.crit dnsmasq[29266]: bad option at line 1 of /var/etc/dnsmasq-adbyby.d/04-dnsmasq.adblock
Sun Sep  4 09:35:42 2022 daemon.crit dnsmasq[29266]: FAILED to start up
Sun Sep  4 09:35:47 2022 daemon.crit dnsmasq[29296]: bad option at line 1 of /var/etc/dnsmasq-adbyby.d/04-dnsmasq.adblock
Sun Sep  4 09:35:47 2022 daemon.crit dnsmasq[29296]: FAILED to start up
Sun Sep  4 09:35:52 2022 daemon.crit dnsmasq[29322]: bad option at line 1 of /var/etc/dnsmasq-adbyby.d/04-dnsmasq.adblock
Sun Sep  4 09:35:52 2022 daemon.crit dnsmasq[29322]: FAILED to start up
Sun Sep  4 09:35:52 2022 daemon.info procd: Instance dnsmasq::cfg01411c s in a crash loop 6 crashes, 0 seconds since last crash

xydche, 2022-09-04:
虚拟机恢复旧版openwrt快照，关自动更新，升新系统，临时解决了，至少能去广告了

MIAOBUSI, 2022-09-04:
手贱点了规则更新同样报错Sun Sep  4 18:49:11 2022 daemon.crit dnsmasq[18033]: bad option at line 1 of /var/etc/dnsmasq-adbyby.d/04-dnsmasq.adblock
Sun Sep  4 18:49:11 2022 daemon.crit dnsmasq[18033]: FAILED to start up
Sun Sep  4 18:49:16 2022 daemon.crit dnsmasq[18083]: bad option at line 1 of /var/etc/dnsmasq-adbyby.d/04-dnsmasq.adblock
Sun Sep  4 18:49:16 2022 daemon.crit dnsmasq[18083]: FAILED to start up
Sun Sep  4 18:49:21 2022 daemon.crit dnsmasq[18105]: bad option at line 1 of /var/etc/dnsmasq-adbyby.d/04-dnsmasq.adblock
Sun Sep  4 18:49:21 2022 daemon.crit dnsmasq[18105]: FAILED to start up
Sun Sep  4 18:49:26 2022 daemon.crit dnsmasq[18153]: bad option at line 1 of /var/etc/dnsmasq-adbyby.d/04-dnsmasq.adblock
Sun Sep  4 18:49:26 2022 daemon.crit dnsmasq[18153]: FAILED to start up
Sun Sep  4 18:49:31 2022 daemon.crit dnsmasq[18180]: bad option at line 1 of /var/etc/dnsmasq-adbyby.d/04-dnsmasq.adblock
Sun Sep  4 18:49:31 2022 daemon.crit dnsmasq[18180]: FAILED to start up
Sun Sep  4 18:49:36 2022 daemon.crit dnsmasq[18238]: bad option at line 1 of /var/etc/dnsmasq-adbyby.d/04-dnsmasq.adblock
Sun Sep  4 18:49:36 2022 daemon.crit dnsmasq[18238]: FAILED to start up
Sun Sep  4 18:49:36 2022 daemon.info procd: Instance dnsmasq::cfg01411c s in a crash loop 6 crashes, 0 seconds since last crash
Sun Sep  4 18:50:43 2022 cron.err crond[20827]: crond (busybox 1.35.0) started, log level 8
Sun Sep  4 18:51:02 2022 daemon.err odhcp6c[23054]: Failed to send SOLICIT message to ff02::1:2 (Permission denied)
Sun Sep  4 18:53:07 2022 daemon.err odhcp6c[23054]: Failed to send SOLICIT message to ff02::1:2 (Permission denied)
Sun Sep  4 18:54:51 2022 cron.err crond[24206]: crond (busybox 1.35.0) started, log level 8
Sun Sep  4 18:54:51 2022 daemon.err uhttpd[4565]: udhcpc: started, v1.35.0
Sun Sep  4 18:54:51 2022 daemon.err uhttpd[4565]: udhcpc: broadcasting discover
Sun Sep  4 18:54:54 2022 daemon.err uhttpd[4565]: udhcpc: no lease, failing
Sun Sep  4 18:55:06 2022 cron.err crond[24639]: crond (busybox 1.35.0) started, log level 8
Sun Sep  4 18:55:11 2022 daemon.crit dnsmasq[25091]: bad option at line 1 of /var/etc/dnsmasq-adbyby.d/04-dnsmasq.adblock
Sun Sep  4 18:55:11 2022 daemon.crit dnsmasq[25091]: FAILED to start up
Sun Sep  4 18:55:13 2022 daemon.err odhcp6c[23054]: Failed to send SOLICIT message to ff02::1:2 (Permission denied)
Sun Sep  4 18:55:16 2022 daemon.crit dnsmasq[25150]: bad option at line 1 of /var/etc/dnsmasq-adbyby.d/04-dnsmasq.adblock
Sun Sep  4 18:55:16 2022 daemon.crit dnsmasq[25150]: FAILED to start up
Sun Sep  4 18:55:21 2022 daemon.crit dnsmasq[25182]: bad option at line 1 of /var/etc/dnsmasq-adbyby.d/04-dnsmasq.adblock
Sun Sep  4 18:55:21 2022 daemon.crit dnsmasq[25182]: FAILED to start up
Sun Sep  4 18:55:26 2022 cron.err crond[25321]: crond (busybox 1.35.0) started, log level 8

guoshh1978, 2022-09-05:
今天git pull发现adbyby-plus配置文件有更新，重新编译更新了一下最新的版本，这个问题解决了。

closed 2022-09-05 by rongcenter

592732377, 2022-09-07:
更新到最新版本后大家能正常更新规则吗？我的一直卡在  “正在更新”上，规则数量也是0.

MIAOBUSI, 2022-09-07:
> 更新到最新版本后大家能正常更新规则吗？我的一直卡在 “正在更新”上，规则数量也是0.

先点一下清空所有订阅规则库，然后再点更新

592732377, 2022-09-07:
> > 更新到最新版本后大家能正常更新规则吗？我的一直卡在 “正在更新”上，规则数量也是0.
> 
> 先点一下清空所有订阅规则库，然后再点更新

Adblock Plus规则还是不行，但是第三方的可以了，感谢！可能是个人偶发问题。

uosX, 2022-09-20:
Wed Sep 21 00:12:22 2022 daemon.crit dnsmasq[19674]: bad option at line 1 of /var/etc/dnsmasq-adbyby.d/04-dnsmasq.adblock
Wed Sep 21 00:12:22 2022 daemon.crit dnsmasq[19674]: FAILED to start up
Wed Sep 21 00:12:22 2022 cron.err crond[19715]: crond (busybox 1.35.0) started, log level 8

x86的软路由也会这样子；
补充了以下2条订阅规则
https://easylist-downloads.adblockplus.org/easylistchina+easylist.txt
https://easylist-downloads.adblockplus.org/easylistchina.txt

关闭Turbo ACC网络加速的，DNS缓存选项后，问题依旧；
Wed Sep 21 00:15:25 2022 daemon.crit dnsmasq[23302]: bad option at line 1 of /var/etc/dnsmasq-adbyby.d/04-dnsmasq.adblock
Wed Sep 21 00:15:25 2022 daemon.crit dnsmasq[23302]: FAILED to start up
Wed Sep 21 00:15:26 2022 kern.warn kernel: [168501.324993] test 65534
Wed Sep 21 00:15:27 2022 kern.warn kernel: [168502.377737] test 65534
Wed Sep 21 00:15:27 2022 kern.warn kernel: [168503.433491] test 65534
Wed Sep 21 00:15:28 2022 kern.warn kernel: [168503.753960] test 65534
Wed Sep 21 00:15:29 2022 kern.warn kernel: [168504.498719] test 65534
Wed Sep 21 00:15:29 2022 kern.warn kernel: [168505.550004] test 65534
Wed Sep 21 00:15:30 2022 kern.warn kernel: [168505.836201] test 512
Wed Sep 21 00:15:30 2022 daemon.crit dnsmasq[23325]: bad option at line 1 of /var/etc/dnsmasq-adbyby.d/04-dnsmasq.adblock
Wed Sep 21 00:15:30 2022 daemon.crit dnsmasq[23325]: FAILED to start up
Wed Sep 21 00:15:31 2022 kern.warn kernel: [168506.604748] test 65534
Wed Sep 21 00:15:32 2022 kern.warn kernel: [168507.658410] test 65534
Wed Sep 21 00:15:33 2022 kern.warn kernel: [168508.713028] test 65534
Wed Sep 21 00:15:34 2022 kern.warn kernel: [168509.768541] test 65534
Wed Sep 21 00:15:35 2022 daemon.crit dnsmasq[23346]: bad option at line 1 of /var/etc/dnsmasq-adbyby.d/04-dnsmasq.adblock
Wed Sep 21 00:15:35 2022 daemon.crit dnsmasq[23346]: FAILED to start up
Wed Sep 21 00:15:35 2022 kern.warn kernel: [168510.819120] test 65534
Wed Sep 21 00:15:36 2022 kern.warn kernel: [168511.868564] test 65534
Wed Sep 21 00:15:37 2022 kern.warn kernel: [168512.922035] test 65534
Wed Sep 21 00:15:37 2022 kern.warn kernel: [168513.818559] test 65534
Wed Sep 21 00:15:38 2022 kern.warn kernel: [168513.977830] test 65534
Wed Sep 21 00:15:39 2022 kern.warn kernel: [168515.029287] test 65534
Wed Sep 21 00:15:40 2022 daemon.crit dnsmasq[23378]: bad option at line 1 of /var/etc/dnsmasq-adbyby.d/04-dnsmasq.adblock
Wed Sep 21 00:15:40 2022 daemon.crit dnsmasq[23378]: FAILED to start up
Wed Sep 21 00:15:40 2022 daemon.info procd: Instance dnsmasq::cfg01411c s in a crash loop 6 crashes, 0 seconds since last crash
Wed Sep 21 00:15:40 2022 kern.warn kernel: [168516.084328] test 65534
Wed Sep 21 00:15:41 2022 kern.warn kernel: [168517.137872] test 65534
Wed Sep 21 00:15:41 2022 kern.warn kernel: [168517.241848] test 512
Wed Sep 21 00:15:43 2022 kern.warn kernel: [168518.189266] test 65534

xydche, 2022-09-22:
这个bug已经修复了啊

xydche, 2022-09-22:
> 这个问题很严重，直接导致把网络干挂了，我都几个地方都直接挂了，导致lan口连局域网都获取不到，坑了

这个已经修复了啊

q30china, 2023-02-14:
下的固件luci-app-adbyby-plus 2.0-75 ，刚装好一切正常，更新规则后出现问题。看了下，确实是下载的规则库的格式，dnsmasq启动时，加载失败。 想问下，如果不想重新回到以前的固件，而且想要保持更新规则的情况下，该如何处理？

q30china, 2023-02-17:
下载了Debian11.06, 根据lede老大的源，git最新代码，编译，只更新包 luci-app-adbyby-plus更新到2.0-76, /var/etc/dnsmasq-adbyby.d/04-dnsmasq.adblock 文件格式改为‘ Adress.........’， 就恢复到以前了。第一次接触这玩意，就是为了想在AppleTv上无广告的看youtube，误入此地。

WeihanLi, 2024-08-09:
> /var/etc/dnsmasq-adbyby.d/04-dnsmasq.adblock 文件格式改为‘ Adress.........’， 就恢复到以前了

@q30china 文件格式怎么修改？