coolsnowwolf/lede #10036: 还是X86编译问题。

baodp, 2022-09-01:
Collected errors:
 * check_data_file_clashes: Package luci-theme-argon-mod wants to install file /home/bdp/lede/build_dir/target-x86_64_musl/root-x86/usr/lib/lua/luci/view/themes/argon/footer.htm
	But that file is already provided by package  * luci-theme-argon
 * check_data_file_clashes: Package luci-theme-argon-mod wants to install file /home/bdp/lede/build_dir/target-x86_64_musl/root-x86/usr/lib/lua/luci/view/themes/argon/header.htm
	But that file is already provided by package  * luci-theme-argon
 * check_data_file_clashes: Package luci-theme-argon-mod wants to install file /home/bdp/lede/build_dir/target-x86_64_musl/root-x86/www/luci-static/argon/favicon.ico
	But that file is already provided by package  * luci-theme-argon
 * check_data_file_clashes: Package luci-theme-argon-mod wants to install file /home/bdp/lede/build_dir/target-x86_64_musl/root-x86/www/luci-static/argon/js/jquery.min.js
	But that file is already provided by package  * luci-theme-argon
 * check_data_file_clashes: Package luci-theme-argon-mod wants to install file /home/bdp/lede/build_dir/target-x86_64_musl/root-x86/www/luci-static/argon/js/script.js
	But that file is already provided by package  * luci-theme-argon
 * opkg_install_cmd: Cannot install package luci-theme-argon-mod.
make[2]: *** [package/Makefile:70: package/install] Error 255
make[2]: Leaving directory '/home/bdp/lede'
make[1]: *** [package/Makefile:111: /home/bdp/lede/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/bdp/lede'
make: *** [/home/bdp/lede/include/toplevel.mk:230：world] 错误 2

baodp, 2022-09-01:
好像是这个argon出问题的。argon主题不点选就能编译过去

lxl6125, 2022-09-01:
luci-theme-argon 和 luci-theme-argon-mod 只能二选一

baodp, 2022-09-02:
好的。谢谢 哪个好？？

closed 2022-09-02 by aiamadeus