coolsnowwolf/lede #10045: 5.19内核问题，运行一段后崩溃

GreatMichaelLee, 2022-09-02:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：5.19内核运行PVE虚拟机+openwrt,运行一段时间后内核崩溃死机，需要重启openwrt对应的kvm恢复。其他虚拟机运行不影响

### (2) 路由器型号和固件版本

A：PVE7.2.7+康耐信N5105四口I225+OpenWrt R22.9.1 / LuCI branch (git-22.243.50955-10aa838)

### (3) 详细日志

A：
```
Fri Sep  2 22:54:26 2022 daemon.notice miniupnpd[13475]: Unknown udp packet received from 192.168.2.39:51413
Fri Sep  2 22:54:48 2022 daemon.notice miniupnpd[13475]: Unknown udp packet received from 192.168.2.39:51413
Fri Sep  2 22:55:26 2022 daemon.notice miniupnpd[13475]: Unknown udp packet received from 192.168.2.39:51413
Fri Sep  2 23:36:11 2022 daemon.notice miniupnpd[13475]: Unknown udp packet received from 192.168.2.39:51413
Fri Sep  2 23:36:36 2022 daemon.notice miniupnpd[13475]: Unknown udp packet received from 192.168.2.39:51413
Fri Sep  2 23:36:58 2022 daemon.notice miniupnpd[13475]: Unknown udp packet received from 192.168.2.39:51413
Fri Sep  2 23:41:01 2022 user.info mwan3track[13958]: Check (ping) failed for target "223.5.5.5" on interface vwan4 (pppoe-vwan4)
Fri Sep  2 23:41:01 2022 user.info mwan3track[4523]: Check (ping) failed for target "223.5.5.5" on interface vwan2 (pppoe-vwan2)
Fri Sep  2 23:41:03 2022 daemon.info netdata[11660]: Initializing file /var/cache/netdata/net_fifo.eth2/main.db.
Fri Sep  2 23:41:03 2022 daemon.info netdata[11660]: Initializing file /var/cache/netdata/net_fifo.eth2/receive.db.
Fri Sep  2 23:41:03 2022 daemon.info netdata[11660]: Initializing file /var/cache/netdata/net_fifo.eth2/transmit.db.
Fri Sep  2 23:41:09 2022 user.info mwan3track[4523]: Check (ping) failed for target "223.5.5.5" on interface vwan2 (pppoe-vwan2)
Fri Sep  2 23:41:10 2022 user.info mwan3track[13958]: Check (ping) failed for target "223.5.5.5" on interface vwan4 (pppoe-vwan4)
Fri Sep  2 23:41:17 2022 user.info mwan3track[4523]: Check (ping) failed for target "223.5.5.5" on interface vwan2 (pppoe-vwan2)
Fri Sep  2 23:41:19 2022 user.info mwan3track[13958]: Check (ping) failed for target "223.5.5.5" on interface vwan4 (pppoe-vwan4)
Fri Sep  2 23:41:25 2022 user.info mwan3track[4523]: Check (ping) failed for target "223.5.5.5" on interface vwan2 (pppoe-vwan2)
Fri Sep  2 23:41:28 2022 user.info mwan3track[13958]: Check (ping) failed for target "223.5.5.5" on interface vwan4 (pppoe-vwan4)
Fri Sep  2 23:41:29 2022 user.info mwan3track[1419]: Check (ping) failed for target "223.5.5.5" on interface vwan1 (pppoe-vwan1)
Fri Sep  2 23:41:30 2022 user.info mwan3track[9034]: Check (ping) failed for target "223.5.5.5" on interface vwan3 (pppoe-vwan3)
Fri Sep  2 23:41:32 2022 user.info mwan3track[1419]: Check (ping) failed for target "114.114.114.114" on interface vwan1 (pppoe-vwan1)
Fri Sep  2 23:41:33 2022 user.info mwan3track[9034]: Check (ping) failed for target "114.114.114.114" on interface vwan3 (pppoe-vwan3)
Fri Sep  2 23:41:37 2022 user.info mwan3track[13958]: Check (ping) failed for target "223.5.5.5" on interface vwan4 (pppoe-vwan4)
Fri Sep  2 23:41:38 2022 daemon.info n2n-edge[11616]: 02/Sep/2022 23:41:38 [edge_utils.c:1568] WARNING: supernode not responding, now trying [x.x.x.x:10086]
Fri Sep  2 23:41:40 2022 user.info mwan3track[1419]: Check (ping) failed for target "223.5.5.5" on interface vwan1 (pppoe-vwan1)
Fri Sep  2 23:41:41 2022 user.info mwan3track[9034]: Check (ping) failed for target "223.5.5.5" on interface vwan3 (pppoe-vwan3)
Fri Sep  2 23:41:43 2022 user.info mwan3track[1419]: Check (ping) failed for target "114.114.114.114" on interface vwan1 (pppoe-vwan1)
Fri Sep  2 23:41:44 2022 user.info mwan3track[9034]: Check (ping) failed for target "114.114.114.114" on interface vwan3 (pppoe-vwan3)
Fri Sep  2 23:41:46 2022 user.info mwan3track[13958]: Check (ping) failed for target "223.5.5.5" on interface vwan4 (pppoe-vwan4)

Fri Sep  2 23:41:46 2022 kern.err kernel: [46988.771741] rcu: INFO: rcu_preempt self-detected stall on CPU
Fri Sep  2 23:41:46 2022 kern.err kernel: [46988.772539] rcu:   2-....: (2100 ticks this GP) idle=df7/1/0x4000000000000000 softirq=3911142/3911144 fqs=935
Fri Sep  2 23:41:46 2022 kern.warn kernel: [46988.773529]       (t=2101 jiffies g=4750129 q=7342 ncpus=4)
Fri Sep  2 23:41:46 2022 kern.warn kernel: [46988.774140] NMI backtrace for cpu 2
Fri Sep  2 23:41:46 2022 kern.warn kernel: [46988.774606] CPU: 2 PID: 20536 Comm: ipset Not tainted 5.19.6 #0
Fri Sep  2 23:41:46 2022 kern.warn kernel: [46988.775298] Hardware name: QEMU Standard PC (Q35 + ICH9, 2009), BIOS 0.0.0 02/06/2015
Fri Sep  2 23:41:46 2022 kern.warn kernel: [46988.776110] Call Trace:
Fri Sep  2 23:41:46 2022 kern.warn kernel: [46988.776496]  <IRQ>
Fri Sep  2 23:41:46 2022 kern.warn kernel: [46988.776842]  du....lvl+0x4d/0x67
Fri Sep  2 23:41:46 2022 kern.warn kernel: [46988.777292]  du...k+0x14/0x1a
Fri Sep  2 23:41:46 2022 kern.warn kernel: [46988.777711]  m.._b.k..e.7+0x32/0x79
Fri Sep  2 23:41:46 2022 kern.warn kernel: [46988.778163]  ? l%i.c._.plu..+0x80/0x80
Fri Sep  2 23:41:46 2022 kern.warn kernel: [46988.778634]  nm..+g..m..b.k..e+0x119/0x130
Fri Sep  2 23:41:46 2022 kern.warn kernel: [46988.779176]  .._.+g..m..b.k..e+0x18/0x20
Fri Sep  2 23:41:46 2022 kern.warn kernel: [46988.779666]  cu_dum....ks+0xcc/0x104
Fri Sep  2 23:41:46 2022 kern.warn kernel: [46988.780124]  cu..e.c...q^C04+0x1af/0x7ab
Fri Sep  2 23:41:46 2022 kern.warn kernel: [46988.780599]  ? .e.c.+0xd/0x20
Fri Sep  2 23:41:46 2022 kern.warn kernel: [46988.781005]  ? ...:y.e...^U..+0x66/0xa0
Fri Sep  2 23:41:46 2022 kern.warn kernel: [46988.781469]  .d.p.c....s+0x52/0x90
Fri Sep  2 23:41:46 2022 kern.warn kernel: [46988.781931]  .ck..e.^G.+0xcf/0xf0
[... 52 lines cut ...]
Fri Sep  2 23:42:06 2022 user.info mwan3track[9034]: Check (ping) failed for target "114.114.114.114" on interface vwan3 (pppoe-vwan3)
Fri Sep  2 23:42:07 2022 daemon.info n2n-edge[11616]: 02/Sep/2022 23:42:07 [edge_utils.c:1568] WARNING: supernode not responding, now trying [175.178.170.158:10086]
Fri Sep  2 23:42:09 2022 daemon.notice miniupnpd[13475]: remove port mapping 33218 UDP because it has expired
Fri Sep  2 23:42:13 2022 user.info mwan3track[13958]: Check (ping) failed for target "223.5.5.5" on interface vwan4 (pppoe-vwan4)
Fri Sep  2 23:42:18 2022 user.notice vpnprogress: disconnected
Fri Sep  2 23:42:18 2022 kern.err kernel: [47020.380138] rcu: INFO: rcu_preempt detected expedited stalls on CPUs/tasks: { 2-... } 3 jiffies s: 1277 root: 0x4/.
Fri Sep  2 23:42:18 2022 kern.err kernel: [47020.381411] rcu: blocking rcu_node structures (internal RCU debug):
Fri Sep  2 23:42:18 2022 kern.info kernel: [47020.382253] Task dump for CPU 2:
Fri Sep  2 23:42:18 2022 kern.info kernel: [47020.382815] task:ipset           state:R  running task     stack:    0 pid:20536 ppid: 20471 flags:0x00000008
Fri Sep  2 23:42:18 2022 kern.info kernel: [47020.383976] Call Trace:
Fri Sep  2 23:42:18 2022 kern.info kernel: [47020.384495]  <TASK>
Fri Sep  2 23:42:18 2022 kern.info kernel: [47020.384987]  ? x^H.s.     ^P+0x27/0x30
Fri Sep  2 23:42:18 2022 kern.info kernel: [47020.385582]  ? do.sc..^H+0x43/0xc0
Fri Sep  2 23:42:18 2022 kern.info kernel: [47020.386141]  ...SYSCALL_6.aft.hwf..+0x63/0xcd
Fri Sep  2 23:42:18 2022 kern.info kernel: [47020.386810] RIP: 0033:0x7f1ada5e328e
Fri Sep  2 23:42:18 2022 kern.info kernel: [47020.387473] RSP: 002b:00007ffe6c3362a8 EFLAGS: 00000246 ORIG_RAX: 000000000000002c
Fri Sep  2 23:42:18 2022 kern.info kernel: [47020.388479] RAX: ffffffffffffffda RBX: 00007f1ada602ac8 RCX: 00007f1ada5e328e
Fri Sep  2 23:42:18 2022 kern.info kernel: [47020.389413] RDX: 0000000000000038 RSI: 00007f1ada53d528 RDI: 0000000000000003
Fri Sep  2 23:42:18 2022 kern.info kernel: [47020.390346] RBP: 000000000000002c R08: 00007f1ada558000 R09: 000000000000000c
Fri Sep  2 23:42:18 2022 kern.info kernel: [47020.391289] R10: 0000000000000000 R11: 0000000000000246 R12: 0000000000001000
Fri Sep  2 23:42:18 2022 kern.info kernel: [47020.392217] R13: 00007f1ada58c1c0 R14: 0000000000000000 R15: 0000000000000000
Fri Sep  2 23:42:18 2022 kern.info kernel: [47020.393115]  </TASK>
```

后面就是连续狂打印 Call Trace直至死机了。

coolsnowwolf, 2022-09-04:
内核 ipset 的问题，多个 ipset 条目同时写入的时候冲突了，应该加个锁之类

GreatMichaelLee, 2022-09-07:
> 内核 ipset 的问题，多个 ipset 条目同时写入的时候冲突了，应该加个锁之类

好的，麻烦后面哪个版本fix了，大佬顺手把这个关了。

closed 2022-09-19 by coolsnowwolf