coolsnowwolf/lede #10046: 关于使用Ubuntu 20.04.4 LTS配置编译环境缺少依赖以及dl编译报错。

sljssboli, 2022-09-03:
## 1.关于使用Ubuntu 20.04.4 LTS配置编译环境缺少依赖以及dl编译报错。在win10linux子系统下wsl操作。

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述
如题
### (1) 具体问题

A：如题

### (2) 路由器型号和固件版本

A：为树莓派4b编译固件。

### (3) 详细日志

A：
```
Note, selecting 'genisoimage' instead of 'mkisofs'
sudo apt install -y genisoimage
Reading package lists... Done
Building dependency tree
Reading state information... Done
genisoimage is already the newest version (9:1.1.11-3.1ubuntu1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
```
B: 
```
make download -j8
 ERROR: tools/upx failed to build.
```

WYC-2020, 2022-09-03:
编译什么iso，别人也给你提示了安转mkisofs，不会就默认别瞎几把选

sljssboli, 2022-09-03:
> 编译什么iso，别人也给你提示了安转mkisofs，不会就默认别瞎几把选

`Note, selecting 'genisoimage' instead of 'mkisofs'。`
`注意，选择“genisoimage”而不是“mkisofs”。`

sljssboli, 2022-09-03:
> > 编译什么iso，别人也给你提示了安转mkisofs，不会就默认别瞎几把选
> 
> `Note, selecting 'genisoimage' instead of 'mkisofs'。` `注意，选择“genisoimage”而不是“mkisofs”。`

```
sudo apt install -y mkisofs
[sudo] password for yhuk:
Reading package lists... Done
Building dependency tree
Reading state information... Done
Note, selecting 'genisoimage' instead of 'mkisofs'
genisoimage is already the newest version (9:1.1.11-3.1ubuntu1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
```
你好。我应该没有看错。

sljssboli, 2022-09-03:
```
ERROR: package/feeds/packages/htop failed to build.
ERROR: package/feeds/packages/pdnsd-alt failed to build.
ERROR: package/feeds/packages/vlmcsd failed to build.
```
我准备直接重置掉wsl。从头开始试试。

sljssboli, 2022-09-05:
```
72532
drwxr-xr-x 1 yhuk yhuk     4096 Sep  5 09:44 ./
drwxr-xr-x 1 yhuk yhuk     4096 Sep  5 02:14 ../
-rw-r--r-- 1 yhuk yhuk      101 Sep  5 09:25 config.buildinfo
-rw-r--r-- 1 yhuk yhuk      257 Sep  5 09:25 feeds.buildinfo
-rw-r--r-- 1 yhuk yhuk 19675487 Sep  5 09:43 openwrt-bcm27xx-bcm2711-rpi-4-ext4-factory.img.gz
-rw-r--r-- 1 yhuk yhuk 19676332 Sep  5 09:43 openwrt-bcm27xx-bcm2711-rpi-4-ext4-sysupgrade.img.gz
-rw-r--r-- 1 yhuk yhuk 16472771 Sep  5 09:43 openwrt-bcm27xx-bcm2711-rpi-4-squashfs-factory.img.gz
-rw-r--r-- 1 yhuk yhuk 16473615 Sep  5 09:43 openwrt-bcm27xx-bcm2711-rpi-4-squashfs-sysupgrade.img.gz
-rw-r--r-- 1 yhuk yhuk     6606 Sep  5 09:43 openwrt-bcm27xx-bcm2711-rpi-4.manifest
drwxr-xr-x 1 yhuk yhuk     4096 Sep  5 09:44 packages/
-rw-r--r-- 1 yhuk yhuk      832 Sep  5 09:44 sha256sums
-rw-r--r-- 1 yhuk yhuk       16 Sep  5 09:25 version.buildinfo
```
事实证明，还是要在全局环境内执行编译。这样，就啥也不会报错了。

closed 2022-09-05 by sljssboli