coolsnowwolf/lede #10050: MT7621有线网络问题

wuzhuang123, 2022-09-03:
机型HC5962
又来了。前几天更新MT7621后测了几天网络有问题，表现是：
1.一般的网页还是可以打开，下载别的文件也是正常，但是asp的OA系统打不开，显示连接超时，钉钉传输文件速度极慢导致失败
[image]
2.putty登录远程主机后会在半分钟内断开
[image]
3.2.4G无线会每2分钟断开一次，5G暂没发现
日志空荡荡，啥都没有打印

vsvkhs, 2022-09-03:
我遇到了网页加载巨慢，7621的机器5962和dir882刷了最新编译的固件都会这样，刷回两个星期前编译的固件则网页正常秒开

wuzhuang123, 2022-09-03:
> 

以前的也有问题，表现是1G网口会间隔断开成百兆，经常是编译一半putty报断开，我感觉还是六月以前没有更新驱动前稳定。

britneyjeanspears, 2022-09-03:
> > 
> 
> 以前的也有问题，表现是1G网口会间隔断开成百兆，经常是编译一半putty报断开，我感觉还是六月以前没有更新驱动前稳定。

k2p表示更新后很稳定。

yuzhe1129, 2022-09-05:
我前天编译的R22.9.1 R3G B70  都是很卡，测试才几M  10M宽带不如，我还以为是我编译的问题

wuzhuang123, 2022-09-05:
> 我前天编译的R22.9.1 R3G B70 都是很卡，测试才几M 10M宽带不如，我还以为是我编译的问题

putty只要打开menuconfig就很快断开，带css的网页打开很慢，比如argon主题的其他设备。理解不了。

commit 39c57aca (2022-09-05): Revert "ramips: mt7621-dts: mux phy0/4 to gmac1"

This reverts commit a28334d8b8dfa58dccd6fce369f5ee87e5b5a703.
Closed: #10050, #10053
closed 2022-09-05 by aiamadeus